betaflight/betaflight #10680: Autoconnect not working (needs delay to until connect works)

redhonu, 2021-04-10:
### Is your feature request related to a problem? Please describe.

The com port is detected, before I can actually use the "connect" button, this also means that auto connect does not work.

### Describe the solution you'd like

In the setting menu have an option to add a delay to autoconnect: like 0.5 or 1 second.

### Describe alternatives you've considered

If autoconnect failed to make a connection, automatically retry after 0.5 or 1 second.

haslinghuis, 2021-04-10:
I thinks it's expected behavior. Auto-connect works on plugging in FC and after (save and) reboot.

Starting configurator does not automatically connect an attached FC. This is line with disconnect which does not use auto-connect.

redhonu, 2021-04-11:
Maybe I was unclear in the the feature request. The issue happens when betaflight is already running on the computer, and I plug in my FC to USB.

haslinghuis, 2021-04-11:
This issue / feature request should probably be moved to configurator.

But I'm still unsure what's happening?
Do you get any errors?
Does configurator try to connect before the port is recognized?
Maybe you could make a video.
Auto-connect does not happen before the port is recognized. 

Which configurator version and firmware version are we talking about on which OS ?
A lot has changed regarding port recognition and handling. Did you try a nightly build? https://github.com/betaflight/betaflight-configurator-nightlies

**Please beware it's for testing purposes only as it's not a public release.**