betaflight/betaflight #10681: Building on OSX 10.15.7 fails.

hydra, 2021-04-10:
Fresh install osx OSX 10.15.7.

```
make arm_sdk_install
Applications/Xcode.app/Contents/Developer/usr/bin/make -j ./obj/betaflight_4.3.0_STM32F405_025ee87a7.hex
%% (optimised) ./src/main/startup/system_stm32f4xx.c 
%% startup_stm32f40xx.s 
%% (speed optimised) ./src/main/drivers/accgyro/accgyro_mpu.c 
%% (optimised) ./src/main/drivers/adc_stm32f4xx.c 
%% (optimised) ./src/main/drivers/bus_i2c_stm32f10x.c 
arm-none-eabi-gcc: fatal error: cannot execute 'cc1': execvp: No such file or directory
```

Path:

```
PATH=/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/opt/X11/bin
```

Manual download and installation of latest GCC toolchain from here, https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm/downloads, and running this command works:

```
make ARM_SDK_PREFIX=/Applications/ARM/bin/arm-none-eabi-
...
% (size optimised) ./src/main/drivers/serial_usb_vcp.c 
Linking STM32F405 
In function 'crsfWriteString',
    inlined from 'crsfWriteChar' at ./src/main/io/displayport_crsf.c:104:12:
./src/main/io/displayport_crsf.c:93:27: warning: 'memcmp' specified size between 4294967041 and 4294967295 exceeds maximum object size 2147483647 [-Wstringop-overflow=]
   93 |     crsfScreen.updated |= memcmp(rowStart, s, truncLen);
      |                           ^
Memory region         Used Size  Region Size  %age Used
           FLASH:       10139 B        10 KB     99.01%
FLASH_CUSTOM_DEFAULTS:           8 B         6 KB      0.13%
    FLASH_CONFIG:          0 GB        16 KB      0.00%
          FLASH1:      527812 B       976 KB     52.81%
FLASH_CUSTOM_DEFAULTS_EXTENDED:          0 GB        16 KB      0.00%
   SYSTEM_MEMORY:          0 GB        29 KB      0.00%
             RAM:       98968 B       128 KB     75.51%
             CCM:       16748 B        64 KB     25.56%
     BACKUP_SRAM:          0 GB         4 KB      0.00%
       MEMORY_B1:          0 GB         0 GB
   text	   data	    bss	    dec	    hex	filename
 530903	   7056	 108272	 646231	  9dc57	./obj/main/betaflight_STM32F405.elf
Creating HEX ./obj/betaflight_4.3.0_STM32F405_025ee87a7.hex 
```

Reverting 80329a61a4a1d71c0b520795d2a2cf2c871b32eb also results in working builds.

haslinghuis, 2021-04-10:
Further info:

- Latest ARM SDK does not work, see #10415
- On OSX `10.15.5` everything works as expected.
- Clang version is `12.0.0` in both OSX `10.15.5` and `10.15.7`
- Path and configuration verified on both OSX versions.

mikeller, 2021-04-18:
@hydra: If you delete the `downloads/` and `tools/` folders in your clone of the repository, what output does running

```
make arm_sdk_install
```

give you?

Also, what are the contents of your `make/local.mk` (if it exists)?

hydra, 2021-04-18:
@mikeller 

```
hydra@Dominics-MBP betaflight % make arm_sdk_install  
mkdir -p tools
mkdir -p downloads
Warning: Illegal date format for -z, --time-cond (and not a file name). 
Warning: Disabling time condition. See curl_getdate(3) for valid date syntax.
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   243  100   243    0     0    394      0 --:--:-- --:--:-- --:--:--   393
100  136M  100  136M    0     0   631k      0  0:03:41  0:03:41 --:--:--  772k
hydra@Dominics-MBP betaflight % stat make/local/mk
stat: make/local/mk: stat: No such file or directory
```

hydra, 2021-04-18:
However, after re-installing the sdk it appears to work now!

I checked the new and old files and there are some differences, maybe some transient error that I didn't notice before:

```
hydra@Dominics-MBP betaflight % diff -r tools/gcc-arm-none-eabi-9-2020-q2-update ../betaflight-cruft/tools/gcc-arm-none-eabi-9-2020-q2-update        
Only in tools/gcc-arm-none-eabi-9-2020-q2-update: foo
Only in tools/gcc-arm-none-eabi-9-2020-q2-update/lib/gcc/arm-none-eabi/9.3.1: arm
Only in tools/gcc-arm-none-eabi-9-2020-q2-update/lib/gcc/arm-none-eabi/9.3.1: cc1
Only in tools/gcc-arm-none-eabi-9-2020-q2-update/lib/gcc/arm-none-eabi/9.3.1: cc1plus
Only in tools/gcc-arm-none-eabi-9-2020-q2-update/lib/gcc/arm-none-eabi/9.3.1: collect2
Only in tools/gcc-arm-none-eabi-9-2020-q2-update/lib/gcc/arm-none-eabi/9.3.1: crtbegin.o
Only in tools/gcc-arm-none-eabi-9-2020-q2-update/lib/gcc/arm-none-eabi/9.3.1: crtend.o
Only in tools/gcc-arm-none-eabi-9-2020-q2-update/lib/gcc/arm-none-eabi/9.3.1: crtfastmath.o
Only in tools/gcc-arm-none-eabi-9-2020-q2-update/lib/gcc/arm-none-eabi/9.3.1: crti.o
Only in tools/gcc-arm-none-eabi-9-2020-q2-update/lib/gcc/arm-none-eabi/9.3.1: crtn.o
Only in tools/gcc-arm-none-eabi-9-2020-q2-update/lib/gcc/arm-none-eabi/9.3.1: include
Only in tools/gcc-arm-none-eabi-9-2020-q2-update/lib/gcc/arm-none-eabi/9.3.1: include-fixed
Only in tools/gcc-arm-none-eabi-9-2020-q2-update/lib/gcc/arm-none-eabi/9.3.1: install-tools
Only in tools/gcc-arm-none-eabi-9-2020-q2-update/lib/gcc/arm-none-eabi/9.3.1: libgcc.a
Only in tools/gcc-arm-none-eabi-9-2020-q2-update/lib/gcc/arm-none-eabi/9.3.1: libgcov.a
Only in tools/gcc-arm-none-eabi-9-2020-q2-update/lib/gcc/arm-none-eabi/9.3.1: liblto_plugin.0.so
Only in tools/gcc-arm-none-eabi-9-2020-q2-update/lib/gcc/arm-none-eabi/9.3.1: liblto_plugin.so
Only in tools/gcc-arm-none-eabi-9-2020-q2-update/lib/gcc/arm-none-eabi/9.3.1: lto-wrapper
Binary files tools/gcc-arm-none-eabi-9-2020-q2-update/lib/gcc/arm-none-eabi/9.3.1/lto1 and ../betaflight-cruft/tools/gcc-arm-none-eabi-9-2020-q2-update/lib/gcc/arm-none-eabi/9.3.1/lto1 differ
Only in tools/gcc-arm-none-eabi-9-2020-q2-update/lib/gcc/arm-none-eabi/9.3.1: plugin
Only in tools/gcc-arm-none-eabi-9-2020-q2-update/lib/gcc/arm-none-eabi/9.3.1: thumb
Only in tools/gcc-arm-none-eabi-9-2020-q2-update/lib: libcc1.0.so
Only in tools/gcc-arm-none-eabi-9-2020-q2-update/lib: libcc1.so
Only in tools/gcc-arm-none-eabi-9-2020-q2-update: share
```

As it's working now, i'll close this.  Thanks for making me re-download it.

hydra, 2021-04-18:
actually, i guess there is one improvement that could be made - transient errors should be retried.

when it fails it does this:

```
hydra@Dominics-MBP betaflight % make arm_sdk_install
mkdir -p tools
mkdir -p downloads
Warning: Illegal date format for -z, --time-cond (and not a file name). 
Warning: Disabling time condition. See curl_getdate(3) for valid date syntax.
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   243  100   243    0     0    228      0  0:00:01  0:00:01 --:--:--   228
 38  136M   38 52.5M    0     0   634k      0  0:03:39  0:01:24  0:02:15  771k
curl: (56) OpenSSL SSL_read: Connection reset by peer, errno 54
make: *** [downloads/gcc-arm-none-eabi-9-2020-q2-update-mac.tar.bz2] Error 56
```

If you retry it does this:
```
hydra@Dominics-MBP betaflight % make arm_sdk_install
gcc-arm-none-eabi-9-2020-q2-update/arm-none-eabi/lib/thumb/v7-r+fp.sp/hard/libstdc++_nano.a: truncated bzip2 input
tar: Error exit delayed from previous errors.
make: *** [tools/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-gcc-9.3.1] Error 1
```

haslinghuis, 2021-04-18:
Never had this issue. Please check your connection.
```
Network: CURL error 56: TCP connection reset by peer
```
https://superuser.com/questions/1463563/different-curl-errors-happen-occasionally

mikeller, 2021-04-19:
@hydra: The tooling script is 'clever' as in it keeps the downloaded tarball of the toolchain, and on second execution, if the tools directory itself does not exist, skips over the download step and directly unpacks the tarball. It looks like in your latest try, the download of the tarball was interrupted, leaving a corrupted tarball behind. And in the second attempt unpacking of the corrupted archive then fails.
This is expected behaviour.

hydra, 2021-04-19:
Perhaps it's possible to check the exit status of the curl command and delete or resume in case of failures?  Resuming the transfer would be ideal to avoid data charges.

haslinghuis, 2021-04-20:
Could do a `curl --connect-timeout 10 --retry 5 ...`

closed 2021-05-28 by github-actions[bot]