betaflight/betaflight #10683: mA in OSD changes when a USB cable is inserted

ellingo, 2021-04-11:
### Describe the bug
mA in OSD changes when a USB cable is inserted

### To Reproduce
Connect a battery and notice mA draw in OSD
Then plug in a USB cable and see the mA in the OSD change quite dramatically.
Plug in another USB cable and you may see yet another value displayed in the OSD.
The other end of the USB cable may be connected to a computer or be unconnected!

### Expected behavior
I would like this to be fixed

### Flight controller configuration

```
# diff

# version
# Betaflight / STM32F405 (S405) 4.2.8 Feb 15 2021 / 12:08:36 (101738d8e) MSP API: 1.43
# config: manufacturer_id: HGLR, board_name: HGLRCF405, version: 3a35e73b, date: 2019-09-30T05:46:12Z

# start the command batch
batch start

board_name HGLRCF405
manufacturer_id HGLR

# name: REKON5

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature GPS
feature TELEMETRY

# map
map TAER1234

# serial
serial 0 32 115200 57600 0 115200
serial 1 8192 115200 57600 0 115200
serial 2 2 115200 9600 0 115200
serial 5 64 115200 57600 0 115200

# aux
aux 0 0 0 1900 2100 0 0
aux 1 1 1 900 1100 0 0
aux 2 13 2 1900 2100 0 0
aux 3 28 1 1900 2100 0 0
aux 4 35 1 1400 1600 0 0
aux 5 36 2 1900 2100 0 0
aux 6 47 1 1400 1600 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E CUSTOM  5705 5685 5665    0 5885 5905    0    0
vtxtable band 4 FATSHARK F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 4
vtxtable powervalues 25 100 200 400
vtxtable powerlabels 25 100 200 350

# master
set acc_trim_pitch = 2
set acc_calibration = -11,0,-51,1
set mag_hardware = NONE
set rssi_channel = 16
set serialrx_provider = SBUS
set min_throttle = 1070
set dshot_bidir = ON
set use_unsynced_pwm = OFF
set motor_pwm_protocol = DSHOT300
set motor_pwm_rate = 480
set vbat_min_cell_voltage = 350
set vbat_warning_cell_voltage = 360
set ibata_scale = 398
set ibata_offset = -3086
set gps_provider = UBLOX
set gps_ublox_use_galileo = ON
set pid_process_denom = 2
set osd_vbat_pos = 6422
set osd_rssi_pos = 6305
set osd_flymode_pos = 6508
set osd_throttle_pos = 6337
set osd_ah_pos = 6254
set osd_current_pos = 6485
set osd_mah_drawn_pos = 343
set osd_gps_lon_pos = 2448
set osd_gps_lat_pos = 2434
set osd_gps_sats_pos = 6401
set osd_home_dir_pos = 6318
set osd_home_dist_pos = 6466
set osd_altitude_pos = 6434
set osd_power_pos = 6519
set osd_warnings_pos = 6473
set osd_avg_cell_voltage_pos = 6454
set osd_disarmed_pos = 6442
set osd_esc_rpm_pos = 161
set osd_esc_rpm_freq_pos = 183
set debug_mode = GYRO_SCALED
set vtx_band = 5
set vtx_channel = 3
set vtx_power = 4
set vtx_low_power_disarm = ON
set vtx_freq = 5732
set vcd_video_system = PAL
set gyro_1_align_yaw = 1800
set name = REKON5

profile 0

# profile 0
set p_pitch = 69
set d_pitch = 44
set f_pitch = 143
set p_roll = 63
set d_roll = 40
set f_roll = 135
set p_yaw = 68
set f_yaw = 135
set d_min_roll = 0
set d_min_pitch = 0

rateprofile 0

# end the command batch
batch end
```
```
PASTE THE OUTPUT OF 'resource show all' HERE
```

### Setup / Versions
 
 HGLRC FD435 mini stack
 
 - Other components:
 
 - How are the different components wired up:

The stack has an large offset error on the current sensor signal

ellingo, 2021-04-12:
I will check the 3.3V regulator today as its output voltage is used as VREF for the AD converter (or so I believe)

ellingo, 2021-04-12:
There are two 3.3V regulators. The one connected to the processor and to the 3V3 pad is 3.290V +- 0.001V no matter what I do with the USB cable. The problem persists though.

ellingo, 2021-04-12:
I am stretching what is possible to do with the current sensor. I make two measurements to be able to calculate the exact Scale and Offset values to use. One low current and one high current measurement. The problem is, of course, the low current measurement.
I will try to find the best way to achieve my goal by experimenting on a number of quads.
I close this case until I have learned more.

closed 2021-04-12 by ellingo

klutvott123, 2021-04-12:
@ellingo The current pin can be sensitive to noise/crosstalk. Try a 100nF capacitor from current signal to ground. I bet you'll be able to set the offset and scale back to default.

ellingo, 2021-04-12:
That really, really makes a lot of sense. Will do and thx.

ellingo, 2021-04-13:
@klutvott123 Brilliant. Soldered in a 100n capacitor. Did it on the cable connecting ESC and FC. Just pulled back the insulation on the yellow and black wires with my nail first. 
The best fit is using two measurements are now: Scale=513 and Offset=-117. A really good start to the day.
mvh
e