betaflight/betaflight #10685: GNSS NEO-M9N support

Sebbb, 2021-04-11:
Hello,

currently, NEO-M9N GNSS modules are not supported. You get a position, but only see a satellite count of 0. I assume, it only needs to be initiated correctly as the U-Blox protocol is supported by Betaflight.

### Is your feature request related to a problem? Please describe.
No

### Describe the solution you'd like
Support for NEO-M9N modules with the U-BLOX protocol

### Describe alternatives you've considered
Currently it's supported with NMEA only

Sebbb, 2021-05-16:
Nobody interested in this? Should not be a complicated change as ublox support is already there (?)

nullx8, 2021-06-15:
same here, it has the fastest fix of them all, (below 4sec, cold start) Betaflight shows the location but sat: 0

pkruger, 2021-07-18:
The Matek site says to set it to NMEA protocol, why not just do that?
"Betaflight doesn’t support NEO-M9N over UBLOX protocol,  Pls select NMEA protocol & Auto Baud."

Guili123, 2021-07-22:
> The Matek site says to set it to NMEA protocol, why not just do that?
> "Betaflight doesn’t support NEO-M9N over UBLOX protocol, Pls select NMEA protocol & Auto Baud."

I have exact this Neo M9N-5883 Modul. I can get a fix with ublox protocol, but no sat count information`s, not surre what
else is missing also.
When I switch to nmea+auto baud , I verified with ublox configurator that nmea protocol is activated, BF don`t "see" the GPS-Modul
anymore, like no GPS Modul attached.

Sebbb, 2021-07-24:
I'm aware that it is supported with NMEA, as I stated in the original feature request:

"Currently it's supported with NMEA only"

But I think, supporting UBLOX would be only a small adjustment and brings e.g. higher position update rates.

haslinghuis, 2021-07-27:
I like the `AP_PERIPH` approach

FlameFlameFlame, 2023-01-13:
Does this issue affect M10 series too?

ctzsnooze, 2023-08-14:
Please try 4.5 Zulu which now has much improved support for M9 and M10 GPS modules.

haslinghuis, 2023-08-14:
Fixed in #10921
Much improved in #12799

closed 2023-08-14 by haslinghuis