betaflight/betaflight #10688: RTH procedure on Failsafe / GPS Rescue

Philpagan, 2021-04-12:
### Is your feature request related to a problem? Please describe.

When initiating a RTH procedure on Failsafe, the kwad first starts aiming to the home point then flyies up to the initial altitude.
Today, I've lost my kwad (it falled down into the water). Yes, I was flying quite low and it was risky, I agree. However, when RX Lost occurred, my kwad started pointing to the home point but falled a bit and touched the water surface before getting time to fly up to the RTH altitude.

### Describe the solution you'd like

When a failsafe occurs in GPS rescue mode, could it be considered first to ascend (even at half or less the initial altitude) then to aim to the home point ? Maybe it doesn't make sense in all cases but getting more altitude would help in recovering the TX-RX connection in most cases.
Thank you for reading my suggestion.

### Describe alternatives you've considered

https://www.youtube.com/watch?v=1IPAZpvuv1s

rickmanship, 2021-04-12:
Looks to me like you just forgot to set the throttle to hold on the failsafe tab. So then throttle went to 0 during the first part of the failsafe making you touch the water before gps rescue could kick in. In your video you can see it start to descend when the throttle was cut. Too bad you don't have sound since then it would be really easy to tell.

And personally I think the way it works is correct. It would be confusing to have it climb to altitude and then turn towards home. I would be really nervous watching that through the goggles since it would look like it is flying away.

Philpagan, 2021-04-13:
Hi @rickmanship. My kwad is set up with a Crossfire Rx with LQ/RSSI over the telemetry (no AUX channel for failsafe in the failsafe tab). But you're right, in the Rx the failsafe is configured to Cut, as advised for flight controllers. Referring to the TBS manual : 
> Cut-This is the setting that you need to use if you got a flight controller in your aircraft, that handels a failsafe routine like RTh for you. If the link breaks up, the receiver will output no signal, which will be detected by your fc.

I could set the failsafe to "failsafe position" but it should be used for DJI-NAZA like controllers and not for our usual FC boards. Not sure of the behaviour with a FC.
However, I agree that seeing the kwad climbing while still leading to the same direction can be confusing but you get the RTH notification in your goggles and the ascent could be restricted to few meters before the kwad starts turning towards the home point !?

tstibor, 2021-04-13:
@Philpagan: One could also argue vice versa, and hence the current BF implementation seems to be the best choice.

 Fly at e.g. 30meters below a tree branch or power supply line, then a fail safe occurs. The kwad starts to climb N meters and gets stuck in the tree branch or power supply line. So it is the same problem just "mirrored".

Philpagan, 2021-04-13:
@tstibor Correct, unfortunately all situations cannot be covered ! ... but in any case, after a short while, the kwad climbs to the Initial Altitude when GPS Rescue is enabled. When flying in woods or such context, Drop on failsafe should be likely a better option.

rickmanship, 2021-04-13:
> Hi @rickmanship. My kwad is set up with a Crossfire Rx with LQ/RSSI over the telemetry (no AUX channel for failsafe in the failsafe tab). But you're right, in the Rx the failsafe is configured to [...]

@Philpagan I don't think you understand. I'm not talking about your radio settings. You definitely want to set your radio/receiver to "cut". I'm talking about the Betaflight failsafe settings. You need to set the throttle to "hold". If you had done this I don't think you would have hit the water.

[image]

Philpagan, 2021-04-13:
@rickmanship Oh yes, I hadn't got your point (even though it was clearly explained) , I'm sorry. And yes, Throttle was set to Auto in my configuration (and so went to Low on Stage 1) ! Thanks a lot for the clarification and the input, I'll definitely set Throttle to Hold in my new setup.

closed 2021-04-13 by Philpagan