betaflight/betaflight #10689: can someone teach me how to install the firmware on a Nucleo stm32H743zi2 board??? im Stuck

delchrys, 2021-04-13:
### Describe the bug
can someone teach me how to install the firmware on a Nucleo stm32H743zi2 board??? im Stuck

### To Reproduce

### Expected behavior
flashing seems to work with st link or with the newer cube programmer, but betaflight configurator cant connect.

### Flight controller configuration
Nucleo stm32h743zi2

### Setup / Versions
 
 - Flight controller: 
 
 - Other components:
 
 - How are the different components wired up:

delchrys, 2021-04-15:
Can someone help me out, been trying a lot of options but don't know how this works...
I'm pretty good with arduino but the stm32 is totally different. Just like to know how I can upload the firmware and how to connect it to the configurator.
Do I need to change setting first or can I just upload the hex file from the releases tab?

klutvott123, 2021-04-15:
You need to flash the nucleo board with the correct target using st-link. Then you should be able to connect with betaflight configurator. This isn't really the right place for this. Try the betaflight slack, fb or other channels for help with stuff like this.

RySkream, 2021-04-18:
This isn't a bug, you should close this out. I would ask on IntoFPV or one of the fpv facebook groups

joewak, 2021-05-17:
[image]

jflyper, 2021-06-18:
@delchrys Still having problem? Come to the Betaflight slack and drop me a DM.

closed 2021-07-28 by github-actions[bot]