betaflight/betaflight #10691: Sensor Input (Compass ... SDA/SDL) - XF Nano Rx/BN-880

bugsysiegals, 2021-04-15:
### Is your feature request related to a problem? Please describe.

I've 3 UARTS available and am using UART1 for DJI Air Unit, UART3 for XF Nano Rx, and now UART6 for BN-880.  That said, I've no pads for SDA/SCL on my FC; however, the Nano Rx does.  I've wired those wires to the Nano Rx and can see GPS, Hdg, Alt, and Sat Telemetry sensors appear in my TX16S Tx.  

### Describe the solution you'd like

I want to see the compass/heading indicator in the OSD.  Since it's being passed to the Tx as Telemetry, perhaps "Compass" could be added to Sensor Input?

### Describe alternatives you've considered

The only alternative is to buy a new FC as I've no extra UART I can convert to IC2.

sedlacekdavid, 2021-04-18:
DJI osd compass doesn't work, it simply won't pop up if I enable it. At least that's my observation having a fully working magnetometer on my quad.

bugsysiegals, 2021-04-19:
> DJI osd compass doesn't work, it simply won't pop up if I enable it. At least that's my observation having a fully working magnetometer on my quad.

It may not work currently, but could be opened later on, or could be hacked to be placed in the "name" field just like how somebody did so we can see warnings.

sedlacekdavid, 2021-04-19:
@bugsysiegals yes agree, but based on what I read on the internet BF devs are against this hack and kind of pushed a cleaner but not fully finished API to DJI devs who found it not really complete and usable. And thats where the DJI/BF warnings ended.

bugsysiegals, 2021-04-20:
I thought the compass would be necessary for the RTH directional arrow and have found it works without compass!  This was important for me as I crashed a quad because signal got weak, I didn't know RTH direction, and so I purposefully set it down before it flew to far away.  Now I can see the RTH direction and trigger GPS Rescue so I should be good.  That said, this could still be helpful for anybody short on UARTs and needing that compass direction but is low priority.

mikeller, 2021-05-20:
@bugsysiegals:

> That said, I was able to connect the SDA/SCL wires to my XF Nano Rx which added GPS, Hdg, Alt, and Sat sensors on my TX16S Tx. Since the XF Nano Rx is able to receive the BN-880 data and send it to the TX16S, would it be possible for BetaFlight to also receive this data for the OSD if I had "Compass" option on the Sensor Input for my XF Nano Rx UART?

I don't think this will be possible, as the Crossfire hardware that you are using is not sending the GPS information on UART that is connected to the flight controller for the RC data - or at least there is nothing about it in the specification.
If TBS changed their Crossfire firmware to send GPS data 'upstream' we could look into using it in Betaflight.

@sedlacekdavid:

> Same situation in OSD Warnings. Guys kind of pushed a cleaner but not fully finished API to DJI devs who found it not really complete and usable. And thats where the DJI/BF warnings ended.

Please stop spreading untruths. The MSP displayport interface in Betaflight has been complete and available for a few years now, and is currently used by a number of OSD devices. It was DJI's decision to go for the poorly designed solution that we have now, instead of using a tried and proven solution.

closed 2021-05-28 by github-actions[bot]