betaflight/betaflight #10703: Mamab F722 with irc Ghost protocol no stickmovement in BF4.3

cap01fpv, 2021-04-19:
### Describe the bug

I am useing a mamba F722 FC with the irc ghost system. Installed it and all works well with the srl2 protocol (telemetry, stickmovements, ...), but when I set the protocol to ghst and rebind the modul to the receiver, I get telemetry datas shown on my taranis x9d+ but I do not get the stickmovement in BF4.3 or BF 4.2. I tried soldering it to UART 2-3-4-5 all of then are working with the SRL2 protocol but non of them with GHST (with auto mode and manual at binding and always first tried with GHST protocol) and I tried each setup 2-3 times.

### To Reproduce

betaflight configurator 10.8 nightly build or 10.7 with cli comands
FC target is: betaflight_4.3.0_STM32F7X2-2430-025ee87a7.hex
set serialrx_provider = GHST

### Expected behavior

FC should work as with the GHST protocol as it dose with the SRL2 protocol.

### Flight controller configuration

```
# diff all

# version
# Betaflight / STM32F7X2 (S7X2) 4.3.0 Mar 29 2021 / 10:53:39 (025ee87a7) MSP API: 1.44
# config: manufacturer_id: DIAT, board_name: MAMBAF722, version: cc7b58ed, date: 2021-02-14T01:14:37Z

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name MAMBAF722
manufacturer_id DIAT
mcu_id 0028002b3237510e31313935
signature 

# name: QL5V2 F7 Xing

# serial
serial 0 0 115200 57600 0 115200
serial 2 64 115200 57600 0 115200
serial 5 2048 115200 57600 0 115200

# beeper
beeper -GYRO_CALIBRATED
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CRASH_FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL

# beacon
beacon RX_LOST
beacon RX_SET

# aux
aux 0 0 0 900 1025 0 0
aux 1 1 1 1925 2100 0 0
aux 2 13 1 1400 1600 0 0
aux 3 35 2 1900 2100 0 0
aux 4 49 2 900 1100 0 0

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BAND_A   A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BAND_B   B FACTORY 5733 5752 5771 5790 5999 5828 5847 5866
vtxtable band 3 BAND_E   E FACTORY 5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 LOWRACE  L FACTORY 5362 5399 5436 5473 5510 5547 5584 5621
vtxtable powerlevels 4
vtxtable powervalues 0 1 2 3
vtxtable powerlabels 25 200 500 800

# master
set gyro_lowpass2_hz = 325
set dyn_notch_max_hz = 350
set dyn_lpf_gyro_min_hz = 260
set dyn_lpf_gyro_max_hz = 650
set acc_calibration = 4,20,-43,1
set mag_hardware = NONE
set baro_hardware = NONE
set rssi_channel = 8
set serialrx_provider = SRXL2
set serialrx_halfduplex = ON
set blackbox_sample_rate = 1/2
set dshot_bidir = ON
set ibata_scale = 255
set yaw_motors_reversed = ON
set small_angle = 180
set simplified_gyro_filter = ON
set simplified_gyro_filter_multiplier = 130
set osd_warn_batt_not_full = OFF
set osd_vbat_pos = 12322
set osd_rssi_pos = 14402
set osd_link_quality_pos = 14434
set osd_rssi_dbm_pos = 98
set osd_tim_2_pos = 14391
set osd_warnings_pos = 14537
set osd_avg_cell_voltage_pos = 2082
set osd_flip_arrow_pos = 2158
set osd_stat_max_spd = OFF
set osd_stat_bb_no = OFF
set osd_stat_max_fft = ON
set debug_mode = GYRO_SCALED
set vtx_band = 5
set vtx_channel = 8
set vtx_power = 4
set vtx_low_power_disarm = UNTIL_FIRST_ARM
set vtx_freq = 5917
set gyro_1_align_yaw = 1800
set gyro_rpm_notch_harmonics = 2
set name = QL5V2 F7 Xing

profile 0

# profile 0
set dyn_lpf_dterm_min_hz = 91
set dyn_lpf_dterm_max_hz = 221
set dterm_lowpass2_hz = 195
set pidsum_limit = 1000
set pidsum_limit_yaw = 1000
set p_pitch = 75
set d_pitch = 57
set f_pitch = 171
set p_roll = 69
set d_roll = 52
set f_roll = 162
set p_yaw = 74
set f_yaw = 162
set d_min_roll = 0
set d_min_pitch = 0
set ff_interpolate_sp = AVERAGED_2
set simplified_pids_mode = RPY
set simplified_pd_ratio = 110
set simplified_pd_gain = 150
set simplified_dmin_ratio = 200
set simplified_ff_gain = 180
set simplified_dterm_filter = ON
set simplified_dterm_filter_multiplier = 130

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set rates_type = ACTUAL
set roll_rc_rate = 10
set pitch_rc_rate = 10
set yaw_rc_rate = 10
set roll_expo = 60
set pitch_expo = 60
set yaw_expo = 60
set roll_srate = 53
set pitch_srate = 53
set yaw_srate = 53
set tpa_breakpoint = 1750

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
# 
```
```
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: SERIAL_TX 4
A01: SERIAL_RX 4
A02: FREE
A03: FREE
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: MOTOR 3
A09: MOTOR 4
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FLASH_CS
B00: PINIO 1
B01: FREE
B02: BEEPER
B03: FREE
B04: FREE
B05: SPI_MOSI 3
B06: FREE
B07: FREE
B08: CAMERA_CONTROL
B09: FREE
B10: SERIAL_TX 3
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_BATT
C02: FREE
C03: ADC_CURR
C04: GYRO_EXTI
C05: FREE
C06: SERIAL_TX 6
C07: FREE
C08: MOTOR 1
C09: MOTOR 2
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: FREE
C13: FREE
C14: LED 2
C15: LED 1
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4:
    CH3: CAMERA_CONTROL
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1: DSHOT_BITBANG 3
    CH2: DSHOT_BITBANG 1
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 3
DMA2 Stream 3: DSHOT_BITBANG 1
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

# 
```

### Setup / Versions
 
 - Flight controller: Mamba F722 bought in 2018/2019 from a german vendor
  RX: irc ghost-2G4 atto; VTX: rush tank mini; 
 - Other components: runcam swift 2
 
 - How are the different components wired up:
[image]

I already contacted irc and went through the set up process with tehm a few times and I also requested diatone with the question if they actual know about the problem, answere no. The areas where you can see solder and no cable are not longer in use with any add on components

goodnerdboy, 2021-04-20:
Most probably related with #10666  impacting versions 2426 and later.
Seems fine in 2425, it seems a pr that is not merged yet might solve it.

cap01fpv, 2021-05-22:
Anything new on troubleshooting or a newer version then the 2425 which is working fine with the setup?

daleckystepan, 2021-06-19:
I am using BF 4.3.0 from 22/5 with GHOST protocol. Telemetry is limited just to voltage but works (If you have flickering telemetry disable baro/mag/GPS). Try newer build and let us know.

cap01fpv, 2021-06-20:
Thank you for your message, everything works as you wrote, cool.

cap01fpv, 2021-06-20:
I will try newer build and give you a feedback.

cap01fpv, 2021-06-20:
With the latest Build #2473 (13.06.2021 06:23:11) the ghost protocol it is also working with voltage, when I disable baro/mag/GPS.

daleckystepan, 2021-06-20:
Great, you can follow https://github.com/betaflight/betaflight/pull/10801 . I consider this as resolved - it is working and usable for flight. Telemetry issue and other glitches will be resolved during few weeks I hope.

closed 2021-06-20 by cap01fpv