betaflight/betaflight #10709: Betaflight can only store 20 settings on the modes tab??

klmn-roland, 2021-04-23:
**When I try to save more than 20 entries, only the first 20 will be saved.**
You may ask why would I want more than 20, there aren't even 20 modes in BF.
Well, what if I want to group multiple modes into 1 aux channel?
I would be using a lot of OR conditions. I have 8 entries just for my ARM mode. 
**Please double the limit, or make it even more if you can. (100 max would be great)**

If you don't know what I'm talking about (which I highly doubt), have a look: [https://youtu.be/Ut7Hpmmy_Us](https://youtu.be/Ut7Hpmmy_Us)

### Is your feature request related to a problem? Please describe.
**When I try to save more than 20 entries, only the first 20 will be saved.**

### Describe the solution you'd like
**Please double the limit, or make it even more if you can. (100 max would be great)**

### Describe alternatives you've considered
there are no alternatives

xsnoopy, 2021-04-30:
Did you try to set the modes in the CLI?

klmn-roland, 2021-04-30:
@xsnoopy If you type 'aux' in CLI, you will see that BF can only handle 20 modes. aux 0-19.

etracer65, 2021-04-30:
This is unlikely to be changed as 20 mode ranges is more than enough for basically every user. Increasing this count uses more memory and adds processing load so the current setting is a balance of need vs. cost. And it's likely you don't need as many as you think as you should be able to optimize your settings.

However if you want to increase this for your own custom needs, then change:

https://github.com/betaflight/betaflight/blob/2a5e457603a94efa0a18df2eb733ccc0ef0b1c1c/src/main/fc/rc_modes.h#L91

and compile your own firmware. Note that the Configurator will probably be broken so you won't want to use the Modes tab and should setup via CLI. Also note that this is **NOT TESTED** and no guarantees that the firmware will still work and not have flight performance issues. Try at your own risk.