betaflight/betaflight #10710: DJI OSD warnings

sedlacekdavid, 2021-04-23:
### Is your feature request related to a problem? Please describe.

Yes the feature is related to a problem - I can't see BF OSD warnings in my DJI fpv system at the moment.

### Describe the solution you'd like

I'd like to have a switch/cli option to turn on OSD warnings specifically for DJI FPV system. Currently there's no simple way how to check what's going when having troubles with a quad.

### Describe alternatives you've considered

Alternative solutions? As far as I know Emuflight does solve this by hacking quad OSD name. But BF is TOP and I'll just stay flying with it no matter what.

Cheers, enjoy the weather, David.

Guili123, 2021-04-23:
Not optimal but a good workaround:
https://www.youtube.com/watch?v=iUZOzvl_t4s&ab_channel=JoshuaBardwell

etracer65, 2021-05-06:
Betaflight has already implemented changes in next release that will make the warnings easy for DJI to retrieve and display in their OSD. See #10540. We will not be "hacking" the warnings into another field.

closed 2021-06-14 by github-actions[bot]