betaflight/betaflight #10721: Error message on betaflight when I connect my flight controller

ggrosclaude, 2021-04-30:
### Describe the bug

I
[image]
 just bought a Chimera7 6S LR BNF drone w/ DJI Air Unit from iflight and I have a problem with betaflight controller and betaflight configurator. When I connect my flight controller to Betaflight, my drone recognizes it. So far, no problems. It's when I go to the CLI and ask to see the version of my flight controller that I have a problem. This message appears: 

##ERROR: version:  NO CONFIG FOUND# # board: manufacturer_id: IFRC, board_name: IFLIGHT_F722_TWING

Apparently my flight controller has no version when it's just out of the box. I have already tried to implement the Firmware Factory Dumps that Iflight makes available. But it doesn't change anything. Especially since I can't even get into boot loader mode. 

Does anyone know how to help me? I would be very grateful. 

I am using a macOs Big Sur (Version 11.2.3)

### Flight controller configuration

```
# diff
###WARNING: NO CUSTOM DEFAULTS FOUND###

# version
# Betaflight / STM32F7X2 (S7X2) 4.2.0 Jun 14 2020 / 03:05:26 (8f2d21460) MSP API: 1.43
###ERROR: diff: NO CONFIG FOUND###
# start the command batch
batch start

board_name IFLIGHT_F722_TWING
manufacturer_id IFRC

# name: Chimera 7

# resources
resource BEEPER 1 C15
resource MOTOR 1 C08
resource MOTOR 2 C06
resource MOTOR 3 C09
resource MOTOR 4 C07
resource MOTOR 5 B06
resource MOTOR 6 B07
resource MOTOR 7 B01
resource MOTOR 8 B00
resource PPM 1 A03
resource LED_STRIP 1 A01
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 C10
resource SERIAL_TX 5 C12
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A03
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 C11
resource SERIAL_RX 5 D02
resource I2C_SCL 2 B10
resource I2C_SDA 2 B11
resource LED 1 C04
resource SPI_SCK 1 A05
resource SPI_SCK 2 B13
resource SPI_SCK 3 B03
resource SPI_MISO 1 A06
resource SPI_MISO 2 B14
resource SPI_MISO 3 B04
resource SPI_MOSI 1 A07
resource SPI_MOSI 2 B15
resource SPI_MOSI 3 B05
resource CAMERA_CONTROL 1 A00
resource ADC_BATT 1 C01
resource ADC_RSSI 1 C00
resource ADC_CURR 1 C02
resource PINIO 1 C13
resource PINIO 2 C14
resource PINIO 3 B08
resource FLASH_CS 1 B09
resource OSD_CS 1 B12
resource GYRO_EXTI 1 A08
resource GYRO_EXTI 2 B02
resource GYRO_CS 1 A15
resource GYRO_CS 2 C03

# timer
timer A00 AF2
# pin A00: TIM5 CH1 (AF2)
timer A03 AF3
# pin A03: TIM9 CH2 (AF3)
timer C08 AF3
# pin C08: TIM8 CH3 (AF3)
timer C06 AF3
# pin C06: TIM8 CH1 (AF3)
timer C09 AF3
# pin C09: TIM8 CH4 (AF3)
timer C07 AF3
# pin C07: TIM8 CH2 (AF3)
timer B06 AF2
# pin B06: TIM4 CH1 (AF2)
timer B07 AF2
# pin B07: TIM4 CH2 (AF2)
timer B01 AF2
# pin B01: TIM3 CH4 (AF2)
timer B00 AF2
# pin B00: TIM3 CH3 (AF2)
timer A01 AF1
# pin A01: TIM2 CH2 (AF1)

# dma
dma ADC 3 0
# ADC 3: DMA2 Stream 0 Channel 2
dma pin A00 0
# pin A00: DMA1 Stream 2 Channel 6
dma pin C08 1
# pin C08: DMA2 Stream 4 Channel 7
dma pin C06 0
# pin C06: DMA2 Stream 2 Channel 0
dma pin C09 0
# pin C09: DMA2 Stream 7 Channel 7
dma pin C07 1
# pin C07: DMA2 Stream 3 Channel 7
dma pin B06 0
# pin B06: DMA1 Stream 0 Channel 2
dma pin B07 0
# pin B07: DMA1 Stream 3 Channel 2
dma pin B01 0
# pin B01: DMA1 Stream 2 Channel 5
dma pin B00 0
# pin B00: DMA1 Stream 7 Channel 5
dma pin A01 0
# pin A01: DMA1 Stream 6 Channel 3

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature GPS
feature TELEMETRY
feature LED_STRIP
feature OSD

# beacon
beacon RX_LOST
beacon RX_SET

# serial
serial 0 64 115200 57600 0 115200
serial 1 1 115200 57600 0 115200
serial 3 2 115200 9600 0 115200

# led
led 0 0,0::L:1
led 1 1,0::L:0
led 2 2,0::L:0
led 3 3,0::L:0
led 4 4,0::L:0
led 5 5,0::L:0
led 6 6,0::L:0
led 7 0,1::L:0
led 8 1,1::L:0
led 9 2,1::L:0
led 10 3,1::L:0
led 11 4,1::L:0
led 12 5,1::L:0
led 13 6,1::L:0

# aux
aux 0 0 0 1675 2075 0 0
aux 1 1 1 900 2100 0 0

# master
set gyro_lowpass2_hz = 375
set gyro_to_use = BOTH
set dyn_notch_width_percent = 0
set dyn_notch_q = 250
set dyn_notch_min_hz = 70
set dyn_notch_max_hz = 350
set dyn_lpf_gyro_min_hz = 300
set dyn_lpf_gyro_max_hz = 750
set acc_calibration = 70,-17,-16,1
set mag_bustype = I2C
set mag_i2c_device = 2
set mag_hardware = NONE
set baro_bustype = I2C
set baro_i2c_device = 2
set rc_smoothing_derivative_type = PT1
set serialrx_provider = CRSF
set adc_device = 3
set blackbox_p_ratio = 16
set blackbox_device = SPIFLASH
set dshot_idle_value = 800
set dshot_burst = ON
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set failsafe_off_delay = 15
set failsafe_throttle = 1600
set current_meter = ADC
set battery_meter = ADC
set vbat_scale = 111
set ibata_scale = 100
set beeper_inversion = ON
set beeper_od = OFF
set beeper_dshot_beacon_tone = 2
set yaw_motors_reversed = ON
set small_angle = 180
set gps_provider = UBLOX
set gps_auto_baud = ON
set yaw_deadband = 5
set osd_vbat_pos = 455
set osd_rssi_pos = 14680
set osd_link_quality_pos = 481
set osd_rssi_dbm_pos = 345
set osd_tim_1_pos = 295
set osd_tim_2_pos = 322
set osd_flymode_pos = 6573
set osd_throttle_pos = 441
set osd_vtx_channel_pos = 355
set osd_current_pos = 2400
set osd_mah_drawn_pos = 2432
set osd_craft_name_pos = 14799
set osd_display_name_pos = 490
set osd_gps_speed_pos = 2455
set osd_gps_lon_pos = 2304
set osd_gps_lat_pos = 2336
set osd_gps_sats_pos = 2423
set osd_home_dir_pos = 2415
set osd_home_dist_pos = 2542
set osd_flight_dist_pos = 213
set osd_altitude_pos = 2485
set osd_pidrate_profile_pos = 407
set osd_warnings_pos = 14731
set osd_avg_cell_voltage_pos = 14656
set osd_disarmed_pos = 14752
set debug_mode = GYRO_SCALED
set max7456_spi_bus = 2
set dashboard_i2c_bus = 2
set pinio_config = 129,1,1,1
set pinio_box = 0,255,255,255
set flash_spi_bus = 3
set gyro_1_bustype = SPI
set gyro_1_spibus = 1
set gyro_2_spibus = 1
set gyro_2_sensor_align = CW90
set gyro_2_align_yaw = 900
set gyro_rpm_notch_q = 800
set name = Chimera 7

profile 0

# profile 0
set dyn_lpf_dterm_min_hz = 84
set dyn_lpf_dterm_max_hz = 204
set dyn_lpf_dterm_curve_expo = 6
set dterm_lowpass2_hz = 180
set vbat_sag_compensation = 100
set anti_gravity_gain = 6000
set feedforward_transition = 40
set iterm_relax_cutoff = 10
set yaw_lowpass_hz = 70
set throttle_boost_cutoff = 10
set p_pitch = 66
set i_pitch = 99
set d_pitch = 59
set f_pitch = 125
set p_roll = 61
set i_roll = 94
set d_roll = 55
set f_roll = 119
set p_yaw = 65
set i_yaw = 99
set f_yaw = 119
set d_min_roll = 36
set d_min_pitch = 40
set ff_interpolate_sp = AVERAGED_3
set ff_spike_limit = 55
set ff_smooth_factor = 40

rateprofile 0

# end the command batch
batch end

```
```
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: CAMERA_CONTROL
A01: LED_STRIP
A02: SERIAL_TX 2
A03: SERIAL_RX 2
A04: FREE
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: GYRO_EXTI
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: GYRO_CS 1
B00: FREE
B01: FREE
B02: GYRO_EXTI
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: FREE
B07: FREE
B08: PINIO 3
B09: FLASH_CS
B10: I2C_SCL 2
B11: I2C_SDA 2
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_BATT
C02: ADC_CURR
C03: GYRO_CS 2
C04: LED 1
C05: FREE
C06: MOTOR 2
C07: MOTOR 4
C08: MOTOR 1
C09: MOTOR 3
C10: SERIAL_TX 4
C11: SERIAL_RX 4
C12: FREE
C13: PINIO 1
C14: PINIO 2
C15: BEEPER
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2:
    CH2: LED_STRIP
TIM3: FREE
TIM4: FREE
TIM5:
    CH1: CAMERA_CONTROL
TIM6: FREE
TIM7: FREE
TIM8:
    CH1: DSHOT_BITBANG 3
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: LED_STRIP
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 3
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

### Setup / Versions
 
 - Flight controller: SucceX-D F7 V2.1 TwinG Flight Controller (HD Version)
 
 - Other components: 
MCU: STM32 F722
Gyro：MPU6000
Baro: DPS310 
Blackbox：32MB
Betaflight OSD: YES
BEC: 5V 2.5A / 9V 2.5A
Current sensor scale:  100
Inverter for SBUS input (UART2-RX)
Smartaudio & Tramp VTX protocol supported
Beeper pad : Yes
RSSI input pad: No
UARTs：5
Motor outputs: 8
8 pin 4in1 ESC port (GND, VBAT, M1, M2, M3, M4, CUR, R5)
6 pin DJI Air Unit port (R1, GND, R2, T2, GND, 9V)
BLHeli32 Telemetry：RX5
Input voltage：recommended 2-6S, MAX 8S
Firmware target: IFLIGHT_F722_TWING
USB type-C

McGiverGim, 2021-05-02:
This error appears because after flashing, and connecting to Betaflight, it asks to apply custom defaults, and they selected NO. If you reflash it again and accepts custom defaults the first time you connect, this error will disappear.

ggrosclaude, 2021-05-03:
> This error appears because after flashing, and connecting to Betaflight, it asks to apply custom defaults, and they selected NO. If you reflash it again and accepts custom defaults the first time you connect, this error will disappear.

Thanks for your answer @McGiverGim ! But how can I flash my flight controller if I can't get into DFU/boot loader mode? I am using a mac.

McGiverGim, 2021-05-03:
I think remember that there is a go to dfu button in the first tab when you connect to Betaflight.

ggrosclaude, 2021-05-03:
> I think remember that there is a go to dfu button in the first tab when you connect to Betaflight.

Yes indeed, there is that button. But that doesn't work either.  Do you think this is because of the flight controller? Thanks for your help.

McGiverGim, 2021-05-03:
There is a boot button in the FC. If you connect the FC pressing this button the FC will be in DFU mode for sure. If not, there is some big problem with the FC.

ggrosclaude, 2021-05-03:
Okay, so yes indeed a problem with my flight controller. Because I already tried to press the button. Thanks for your help.

McGiverGim, 2021-05-03:
Are you sure it does not go into DFU mode? Is very strange that. Try several times, disconnect all from the FC (GPS, RX, etc.) if you can.

ggrosclaude, 2021-05-03:
Yes, quite sure. I'm afraid it's because my drone got hot when I tried to test my air unit. I will try to unplug the RX, GPS and try again.

McGiverGim, 2021-05-03:
I close this issue because it seems clear that this is not a develop issue. If you need more help I recommend that you use the official Slack chat or Facebook group: https://github.com/betaflight/betaflight#support-and-developers-channel

closed 2021-05-03 by McGiverGim

ggrosclaude, 2021-05-03:
Yes, okay, sorry for opening the discussion in the wrong place. But thanks for your help! 

Best regards!