betaflight/betaflight #10729: Serial RX + IBUS Telemetry on the same UART

jojolll, 2021-05-06:
### Describe the bug
It is not possible to activate Serial RX and IBUS TELEMETRY on the same serial port. In my case, this is necessary to operate the IA6B receiver (with the 14 channels firmware on the FS-I6).

This feature was available on Cleanflight 2.2, and seems to me to be unavailable since Cleanflight 2.4. (I am not sure how the versions of CF and BF relate to each other, but I can say that in BF version 4.0.6 it is impossible to enable IBUS + SerialRX on my SPF3. Not tested on 3.X versions, but from the feedback it was functional).

Given the way I fixed the problem on my personal firmware, I think BF versions 4.2 are all affected as well. 
The fact that I'm on an F3 should not change anything since I did the necessary activations/deactivations in "common_pre.h".

Given the way I fixed the problem on my personal firmware, I think BF versions 4.2 are all affected as well.

### To Reproduce
Flash BF 4.0.6 (and maybe any 4.X.X version) on SPF3 (and maybe any other FC) 
Goto "ports"  in BF configurator
Activate Serial RX and IBUS TELEMETRY on the same UART
Save and reboot.

--> the config is reinit

### Expected behavior
The config is saved

### Flight controller configuration
I'm on my corrected firmware so I don't think it matters.
```
PASTE THE OUTPUT OF 'diff' HERE
```
```
PASTE THE OUTPUT OF 'resource show all' HERE
```

### Setup / Versions
 - Flight controller: SPF3
 - Other components:  IA6B RX, GPS UBLOX, BLHELI ESC DSHOT300
 - How are the different components wired up: IBUS RX -> UART3 TX ; IBUS TX UART3 RX

Since I have been flying CF2.2 for a long time and have always operated that way, I tried to find out why the activation of IBUS telemetry is not working under the latest version of BF for my card.

Reason 1: IBUS telemetry disabled on F3
-> so I created my own firmware with a lot of useless functions disabled for my case
At this moment it still doesn't work.

Reason 2: In the src/io/serial.h file, the line : 
`#define TELEMETRY_SHAREABLE_PORT_FUNCTIONS_MASK (FUNCTION_TELEMETRY_FRSKY_HUB | FUNCTION_TELEMETRY_LTM | FUNCTION_TELEMETRY_MAVLINK)`
seemed to block the operation I was looking for. So I added `| FUNCTION_TELEMETRY_IBUS` and that solved my problem.

The question is certainly why IBUS was not included in this list. According to my tests (for the moment very little in BF 4.0.6), this configuration does not cause any particular problem. Everything works as I expect, and even better than with CF2.2 because the IBUS telemetry had to be reworked.

etracer65, 2021-05-06:
F3 flight controllers are no longer supported on Betaflight. The last supported version was 4.0.6. If it doesn't work there then there will be no support or updates.

However I just tested on the current release and it works fine. Be sure to first **ONLY** enable Serial Rx on the Ports tab and save/reboot. Then go to the Configuration tab and select serial receiver and IBUS, enable the Telemetry feature and save/reboot. Then you can go back to the Ports tab and select IBUS telemetry on the UART and save/reboot.

We can't provide any support for any custom versions you compile yourself.

jojolll, 2021-05-07:
I thought that the problem was certainly still present on the latest versions, I'm glad to know that this is not the case (I will soon have a newer card, so I'm reassured) :)  

Regarding my custom version, I managed to activate everything I needed anyway, even if it's not supported anymore on the F3 

Thank you for your answer.

closed 2021-05-07 by jojolll