betaflight/betaflight #10732: OSD Image/Font saves, but on reboot over-rides.

swg-fpv, 2021-05-09:
### Describe the bug

Whenever you add a new font/startup logo and click "load font" it will save perfectly, but then will reboot and over-write the changes made and set it to default. Very frustrating basically. 
### To Reproduce

Create a custom logo, paste that into the logo upload area in osd, click load font (or similar) and then on reboot you'll see no changes are made. Repeat these steps but this time unplug before the flight controller is fully rebooted. This time a logo will appear on the next power on. 
### Expected behavior

The logo/font to appear without having to do a hot-unplug.
### Flight controller configuration

```
PASTE THE OUTPUT OF 'diff' HERE
```
# 

# diff all

# version
# Betaflight / STM32F745 (S745) 4.2.9 Apr 27 2021 / 19:34:04 (e097f4ab7) MSP API: 1.43
# config: manufacturer_id: FLWO, board_name: FLYWOOF745, version: fda05330, date: 2020-11-21T10:29:52Z

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name FLYWOOF745
manufacturer_id FLWO
mcu_id 004200373138511238343530
signature 

# name: Fouride

# feature
feature -RX_PARALLEL_PWM
feature -AIRMODE
feature GPS
feature ESC_SENSOR

# serial
serial 2 64 115200 57600 0 115200
serial 3 8192 115200 57600 0 115200
serial 5 2 115200 115200 0 115200
serial 6 1024 115200 57600 0 115200

# led
led 0 7,6::CTO:1
led 1 8,6::CTO:1
led 2 9,6::CTO:1
led 3 10,6::CTO:1

# mode_color
mode_color 7 0 8

# aux
aux 0 0 0 1900 2100 0 0
aux 1 1 1 1450 2050 0 0
aux 2 46 1 1875 2100 0 0
aux 3 13 3 1900 2100 0 0
aux 4 35 2 1875 2100 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BAND_A   A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BAND_B   B CUSTOM  5733 5752 5771 5790 5999 5828 5847 5866
vtxtable band 3 BAND_E   E CUSTOM  5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 5
vtxtable powervalues 25 50 100 200 600
vtxtable powerlabels 25 50 100 200 MAX

# rxfail
rxfail 5 s 2000

# master
set gyro_overflow_detect = OFF
set dyn_notch_width_percent = 0
set dyn_notch_q = 250
set dyn_notch_max_hz = 350
set dyn_lpf_gyro_min_hz = 100
set dyn_lpf_gyro_max_hz = 400
set acc_calibration = 62,55,20,1
set baro_hardware = AUTO
set fpv_mix_degrees = 30
set serialrx_provider = CRSF
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set failsafe_procedure = GPS-RESCUE
set align_board_yaw = 45
set vbat_min_cell_voltage = 230
set vbat_warning_cell_voltage = 270
set small_angle = 180
set gps_provider = UBLOX
set gps_rescue_ground_speed = 900
set gps_rescue_sanity_checks = RESCUE_SANITY_FS_ONLY
set gps_rescue_min_sats = 7
set gps_rescue_allow_arming_without_fix = ON
set thrust_linear = 25
set osd_vbat_pos = 6614
set osd_rssi_pos = 6562
set osd_link_quality_pos = 6594
set osd_tim_1_pos = 243
set osd_tim_2_pos = 6210
set osd_mah_drawn_pos = 6583
set osd_gps_speed_pos = 6178
set osd_gps_lon_pos = 65
set osd_gps_lat_pos = 33
set osd_gps_sats_pos = 6200
set osd_home_dir_pos = 102
set osd_home_dist_pos = 97
set osd_flight_dist_pos = 120
set osd_efficiency_pos = 469
set osd_stat_max_dist = ON
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set osd_stat_flight_dist = ON
set debug_mode = GYRO_SCALED
set vtx_band = 1
set vtx_channel = 1
set vtx_power = 4
set vtx_freq = 5865
set gyro_1_align_yaw = 2700
set gyro_rpm_notch_harmonics = 2
set name = Fouride

profile 0

# profile 0
set dyn_lpf_dterm_min_hz = 80
set dyn_lpf_dterm_max_hz = 200
set dyn_lpf_dterm_curve_expo = 8
set dterm_lowpass2_hz = 200
set anti_gravity_gain = 5000
set feedforward_transition = 25
set iterm_relax_cutoff = 11
set i_pitch = 65
set d_pitch = 35
set f_pitch = 73
set p_roll = 41
set i_roll = 61
set d_roll = 33
set f_roll = 70
set i_yaw = 65
set f_yaw = 75
set d_min_roll = 0
set d_min_pitch = 0

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set roll_srate = 75
set pitch_srate = 72
set yaw_srate = 71
set tpa_rate = 30

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
# 
```
PASTE THE OUTPUT OF 'resource show all' HERE
```
A00: SERIAL_TX 4
A01: FREE
A02: LED 1
A03: FREE
A04: FLASH_CS
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FREE
B00: MOTOR 1
B01: MOTOR 2
B02: FREE
B03: FREE
B04: FREE
B05: FREE
B06: I2C_SCL 1
B07: I2C_SDA 1
B08: FREE
B09: FREE
B10: SERIAL_TX 3
B11: SERIAL_RX 3
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: PINIO 1
C01: FREE
C02: ADC_CURR
C03: ADC_BATT
C04: FREE
C05: FREE
C06: SERIAL_TX 6
C07: SERIAL_RX 6
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: FREE
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: LED_STRIP
D13: FREE
D14: FREE
D15: BEEPER
E00: FREE
E01: GYRO_EXTI
E02: SPI_SCK 4
E03: FREE
E04: GYRO_CS 1
E05: SPI_MISO 4
E06: SPI_MOSI 4
E07: SERIAL_RX 7
E08: FREE
E09: MOTOR 3
E10: FREE
E11: MOTOR 4
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4:
    CH1: LED_STRIP
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1: DSHOT_BITBANG 2
    CH2: DSHOT_BITBANG 5
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: LED_STRIP
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 2
DMA2 Stream 3: DSHOT_BITBANG 5
DMA2 Stream 4: ADC
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

### Setup / Versions
 <!-- Specify your flight controller model (what type is it, where was it bought from, ...) -->
 - Flight controller: Flywoo GN745 (40a, f745 aio) 
 <!-- Specify other components attached to the flight controller (RX, VTX, brand / model for all of them, firmware version where applicable...) --> Xf nano RX, m80 gps, (foxeer razer micro + flywoo hm600 vtx, motors)
 - Other components: As above
 <!-- Details about how all is wired --> Very well; tx-rx, rx-tx on gps, vtx to correct uart, cam... not relevant overall.
 - How are the different components wired up: As per manual; again, not relevant in this scenario.

<!-- Add any other context about the problem that you think might be relevant here. --> This has happened on all flight controllers I have on every FW version. This being an F745 shows that the problem affects even the best chips around.

McGiverGim, 2021-05-09:
Have you a battery connected? If not try with battery connected (props off!!!)

swg-fpv, 2021-05-09:
Oh that works thanks; although I would still stress that it's an
unnecessary hassle. I have no clue how hard it is to get this kind of stuff
to work (you people are God-sends being active to reply etc) but it's
always worth mentioning :)
Kind regards
SCott

McGiverGim, 2021-05-09:
The problem is that some FCs don't power the OSD chip without battery. It's not a problem of Betaflight, is a problem of the FC

swg-fpv, 2021-05-09:
Ah I told you I was rubbish with knowing how it works :D Thanks for
explaining! That must be it :)

haslinghuis, 2021-05-09:
@swg-fpv as you found the solution please close the issue.

closed 2021-05-10 by McGiverGim