betaflight/betaflight #10734: CLI Mode on Switch

terry-dactle, 2021-05-10:
### Is your feature request related to a problem? Please describe.
Situation: I dont want to use Programmable RGB leds as it is unneeded and saves complexities but I want to be able to turn on/off the leds so I connected 5V RGB Leds to my Buzzer Pins using the colors I want. The DShot beacon is used as a buzzer and my leds also turn on with Beeper Mode.

Problem: I want to be able to turn on the led's via the external buzzer pin without the Dshot Beacon so that I can land at night without waking everyone up.

### Describe the solution you'd like
I think it would be nice to be able to toggle the Dshot Beacon ON/OFF via a switch on the mode menu

### Describe alternatives you've considered
What I was thinking would be even better would be the ability to send a one shot string to the CLI via the modes for complex or out of the ordinary situations.

In my situation I could bind my 3 position switch to have the console sent  
"set beeper_dshot_beacon_tone=4
save"
on one position and
"set beeper_dshot_beacon_tone=0
save"
on another

Nothing else as of yet, but would like to thank all the devs for their amazing work, we all appreciate you greatly.

etracer65, 2021-05-10:
I think you're confused. The DSHOT beacon is using motors acting as a beeper. They can't "beep" while you're flying. It's not controlling an external beeper. The BEEPER setting RX_SET and related BEEPER mode are what controls the physical beeper device connected to your flight controller connected to the buzzer pads. But chances are that transistor used to drive the buzzer on the flight controller is not capable of supplying the current needed to power your LEDs. So connecting directly to the buzzer pads may damage the flight controller. You might want to look at using PINIO (see the wiki) along with an external MOSFET to drive the LEDs.

This is not the correct place to ask support or configuration questions. This is meant for reporting firmware bugs or new features. See here instead: https://github.com/betaflight/betaflight#support-and-developers-channel

etracer65, 2021-05-10:
And if you were using addressable RGB LEDs then you could simply use the `LEDLOW` mode to turn them on/off with a switch.

closed 2021-05-10 by etracer65