betaflight/betaflight #10736: Connect Betaflight SITL to mavproxy talking MAVLink

aniongithub, 2021-05-12:
### Describe the bug
I'm trying to get Betaflight-SITL to listen to/send MAVLink messages over port 5761. I used the instructions on [this](https://github.com/betaflight/betaflight/blob/master/src/main/target/SITL/README.md) page to build the SITL target and then tried to connect to it using `mavproxy` and a [custom receiver](https://github.com/aniongithub/betaflight-SITL-MAVLink/blob/main/mavlink_listener.py) that uses `pymavlink`. Neither of the approaches are able to do anything beyond establishing a connection to port 5761, not even receive a heartbeat.

### To Reproduce
- Clone [this](https://github.com/aniongithub/betaflight-SITL-MAVLink) repository that has a complete and verifiable example and submodules.
- Build the SITL target (on the host if you have all the dependencies), or use my convenience script [`build-run-betaflight-SITL.sh`](https://github.com/aniongithub/betaflight-SITL-MAVLink/blob/main/build-run-betaflight-SITL.sh) in a terminal window. Note that this needs docker installed. The script will build and run the produced `betaflight_SITL.elf` binary, which listens on `127.0.0.1:5761`
- Run `mavproxy.py` and connect to `127.0.0.1:5761` or use my convenience script [`run-mavproxy.sh`](https://github.com/aniongithub/betaflight-SITL-MAVLink/blob/main/run-mavproxy.sh) that needs docker. Run this command in a separate terminal.

### Expected behavior
We're able to receive heartbeat from `betaflight_SITL.elf` and send commands to it using `mavproxy`.

### Flight controller configuration
```
n/a (SITL)
```
```
n/a (SITL)
```
### Setup / Versions
[Latest commit](https://github.com/betaflight/betaflight/tree/2a5e457603a94efa0a18df2eb733ccc0ef0b1c1c) from master
 - Other components: `betaflight_SITL.elf` listening on port 5761 for connections, `mavproxy.py` trying to communicate via `127.0.0.1:5761`

- How are the different components wired up:
Direct network connection via `127.0.0.1:5761`

A similar issue [here](https://gitter.im/betaflight/betaflight?at=5b0343d1677d792bd1a4cd9c) on gitter, but there are no details on what configuration is needed or any more information.

Glad to provide more details if needed. Thanks in advance!

haslinghuis, 2021-05-12:
Connection with configurator 10.8 is working with SITL 4.3 build - tested just yesterday. Did not try mavlink.
Do I have to just run mavlink_listener.py ? Are there any arguments needed? I'm on Fedora 32 with python 3.8.9

aniongithub, 2021-05-12:
I run `mavlink_listener.py --master udp:127.0.0.1:5761`

haslinghuis, 2021-05-12:
Should be tcp instead of upd

No luck replicating your issue:
```
[mark@localhost Documents]$ python mavlink.py --device tcp:127.0.0.1:5761
Traceback (most recent call last):
  File "mavlink.py", line 6, in <module>
    from pymavlink import mavutil
ModuleNotFoundError: No module named 'pymavlink'
```

aniongithub, 2021-05-12:
Please use a docker run command, similar to how [`run-mavproxy.sh`](https://github.com/aniongithub/betaflight-SITL-MAVLink/blob/main/run-mavproxy.sh) does that runs in a docker container with all dependencies installed. It appears you don’t have pymavlink on your system.

I’ll add this to the repo shortly.

haslinghuis, 2021-05-12:
Won't need docker. Just a virtual environment

aniongithub, 2021-05-13:
> ```
> [mark@localhost ~]$ docker build -t betaflight-sitl .
> unable to prepare context: unable to evaluate symlinks in Dockerfile path: lstat /home/mark/Dockerfile: no such file or directory
> [mark@localhost ~]$ docker run --rm -it --privileged --network=host -v $PWD:/betaflight-sitl-mavlink -t betaflight-sitl mavlink_listener.py --device udp:127.0.0.1:5761
> docker: Cannot connect to the Docker daemon at unix:///var/run/docker.sock. Is the docker daemon running?.
> See 'docker run --help'.
> [mark@localhost ~]$ 
> ```

Just checking in to find out if you needed any more information? 

The above console dump looks like you weren't in the project folder (`lstat /home/mark/Dockerfile: no such file or directory`) and that the also, docker socket seems unreachable (`Cannot connect to the Docker daemon at unix:///var/run/docker.sock`).

haslinghuis, 2021-05-13:
```
Sending build context to Docker daemon  83.46kB
Step 1/13 : FROM ubuntu:18.04
18.04: Pulling from library/ubuntu
01bf7da0a88c: Pull complete 
f3b4a5f15c7a: Pull complete 
57ffbe87baa1: Pull complete 
Digest: sha256:538529c9d229fb55f50e6746b119e899775205d62c0fc1b7e679b30d02ecb6e8
Status: Downloaded newer image for ubuntu:18.04
 ---> 4eb8f7c43909
Step 2/13 : ARG DEBIAN_FRONTEND=noninteractive
 ---> Running in c41b062ce634
Removing intermediate container c41b062ce634
 ---> e832dd90599b
Step 3/13 : RUN apt-get -y update &&    apt-get -y install         build-essential         gdb         software-properties-common         git         ccache         python3-dev         python3-pip         curl         bzip2         libblocksruntime-dev
 ---> Running in adaf0d20f081
OCI runtime create failed: this version of runc doesn't work on cgroups v2: unknown
[mark@localhost betaflight-SITL-MAVLink]$
```

aniongithub, 2021-05-13:
Upon googling the error message, this seems to be a [general issue](https://github.com/docker/for-linux/issues/841) with docker and specific versions of [Linux kernels](https://bugzilla.redhat.com/show_bug.cgi?id=1507149)

The fix is outlined [here](https://github.com/docker/for-linux/issues/841#issuecomment-582644329)
- You need to upgrade your kernel (seems to affect RHEL based distros, see the linked kernel bug)
- You may also be able to add `cgroup.memory=nokmem` to `/etc/default/grub`, `update-grub` and reboot

As you can see in the GIF below, this works fine if your docker setup is fine (I'm on Ubuntu) and has nothing to do with the issue I reported above.

[image]

haslinghuis, 2021-05-13:
Still no luck, even updated kernel drivers
```
[mark@localhost betaflight-SITL-MAVLink]$ sudo docker build -t betaflight-sitl .
[sudo] password for mark: 
Sending build context to Docker daemon  83.46kB
Step 1/13 : FROM ubuntu:18.04
 ---> 4eb8f7c43909
Step 2/13 : ARG DEBIAN_FRONTEND=noninteractive
 ---> Using cache
 ---> e832dd90599b
Step 3/13 : RUN apt-get -y update &&    apt-get -y install         build-essential         gdb         software-properties-common         git         ccache         python3-dev         python3-pip         curl         bzip2         libblocksruntime-dev
 ---> Running in 54dd7863f36f
OCI runtime create failed: this version of runc doesn't work on cgroups v2: unknown

[mark@localhost betaflight-SITL-MAVLink]$ cat /proc/cmdline
BOOT_IMAGE=(hd0,gpt2)/vmlinuz-5.11.19-100.fc32.x86_64 root=/dev/mapper/fedora_localhost--live-root ro cgroup.memory=nokmem resume=/dev/mapper/fedora_localhost--live-swap rd.lvm.lv=fedora_localhost-live/root rd.lvm.lv=fedora_localhost-live/swap rhgb quiet

[mark@localhost betaflight-SITL-MAVLink]$ uname -r
5.11.19-100.fc32.x86_64
```

aniongithub, 2021-05-13:
No worries. Thanks for trying, although I think we're digressing from the point of the issue here. The RHEL docker issue seems unrelated to my current problem. Given a working Docker installation, I was hoping to make this easier to test but it seems we're going the opposite way :sweat_smile: 

To get back on track, what can I do to help you run `mavlink_listener.py`? You can run it directly on your host (or a venv) after installing [`pymavlink`](https://github.com/ArduPilot/pymavlink/#installation). There should be no other dependencies to run that passive listener python script.

Were you able to build the Betaflight SITL target and start it listening on that port?

This [gitter.im](https://gitter.im/betaflight/betaflight?at=5b08b03752e35117cde11524) post from @mikeller seems to indicate that there's some way to "configure port 5761 (shown as UART1) for Mavlink, or this won't work".
[image]

What needs to be configured for this to work?

haslinghuis, 2021-05-13:
Would guess set telemetry output to Mavlink in the ports tab on uart1 for port 5761.

`Podman build -t betaflight-sitl .` worked, but running the next command doesn't. So let's forget docker.
```
[mark@localhost betaflight-SITL-MAVLink]$ python -m venv sitl
[mark@localhost betaflight-SITL-MAVLink]$ ./sitl/bin/activate
bash: ./sitl/bin/activate: Permission denied
[mark@localhost betaflight-SITL-MAVLink]$ source sitl/bin/activate
(sitl) [mark@localhost betaflight-SITL-MAVLink]$ 
```

aniongithub, 2021-05-13:
> Would guess set telemetry output to Mavlink in the ports tab on uart1 for port 5761.

Could you elaborate on this? How do I set it? Do I use the configurator? [`src/main/target/SITL/target.h`](https://github.com/betaflight/betaflight/blob/2a5e457603a94efa0a18df2eb733ccc0ef0b1c1c/src/main/target/SITL/target.h#L114) explicitly has a `#undef USE_TELEMETRY_MAVLINK` so I don't understand how this could be a runtime configuration?
 
> `Podman build -t betaflight-sitl .` worked, but running the next command doesn't. Something with parsing the $PWD:
> 
> ```
> [mark@localhost betaflight-SITL-MAVLink]$ podman run --rm -it --privileged --network=host -v
> $PWD:/betaflight-sitl-mavlink -t betaflight-sitl mavlink_listener.py --device udp:127.0.0.1:5761
> Error: executable file `mavlink_listener.py` not found in $PATH: No such file or directory: OCI not found
> ```

This is great, note you'll need to adjust the path as I have in the utility script [`run_mavlink-listener.sh`](https://github.com/aniongithub/betaflight-SITL-MAVLink/blob/8415e87ffaa3b972060a80142ef2e85eb6d9201c/run-mavlink_listener.sh#L4). The command should be

`podman run --rm -it --privileged --network=host -v $PWD:/betaflight-sitl-mavlink -t betaflight-sitl python3 "/betaflight-sitl-mavlink/mavlink_listener.py" --device udp:127.0.0.1:5761`

because you are in / but `$PWD` is mounted to `/betaflight-sitl-mavlink`.

> Running python:
> 
> ```
> [mark@localhost betaflight-SITL-MAVLink]$ python -m venv sitl
> [mark@localhost betaflight-SITL-MAVLink]$ source sitl/bin/activate
> (sitl) [mark@localhost betaflight-SITL-MAVLink]$ pip install pymavlink
> ```
> 
> Pip install errors out, do you have a requirements file for pip? Using Python 3.8.9
> Trying to follow instructions on https://github.com/ArduPilot/pymavlink/#installation but seems Fedora is not compatible (does not have those packages to install, that's a big problem with ROS too)

The updated `podman` command above _should_ work (given Fedora package issues that would probably be easiest), but in case it doesn't, please try the attached `requirements.txt`. Note, you need to `export DISABLE_MAVNATIVE=True` before using `pip3` to install dependencies. I have **not** tested this with python2 or pip.

Thank you for patiently working with me to get this going, I appreciate it! :smile: 

[requirements.txt](https://github.com/betaflight/betaflight/files/6475006/requirements.txt)

haslinghuis, 2021-05-13:
The updated podman command works without error, still able to connect using betaflight configurator 10.8 on firmware sitl running 4.3
> Could you elaborate on this? How do I set it? Do I use the configurator? src/main/target/SITL/target.h explicitly has a #undef 
> USE_TELEMETRY_MAVLINK so I don't understand how this could be a runtime configuration?
Yes this would setting this (in configurator) useless. But still should check it.

aniongithub, 2021-05-13:
Great! 

As you can see in [`mavlink_listener.py` line 88](https://github.com/aniongithub/betaflight-SITL-MAVLink/blob/8415e87ffaa3b972060a80142ef2e85eb6d9201c/mavlink_listener.py#L88), at this time the listener is waiting on a heartbeat that it never receives.

`master.wait_heartbeat()`

never returns and so the subsequent `print("got heartbeat")` on [line 90](https://github.com/aniongithub/betaflight-SITL-MAVLink/blob/8415e87ffaa3b972060a80142ef2e85eb6d9201c/mavlink_listener.py#L90) is never executed. If I skip that line, nothing is ever received by my listener and all requests are ignored. Presumably because the link isn't talking MAVLink. So my question is,

### How do I connect Betaflight SITL to mavproxy (or a custom mavlink client) talking MAVLink over port 5761?

haslinghuis, 2021-05-13:
https://research.ece.ncsu.edu/wireless/MadeInWALAN/CentMeshWikiBackup/tracwiki/raw-attachment/wiki/HardWare/MobileNode/Programming/Tutorial_drone-with-mavutilusage.pdf

haslinghuis, 2021-05-13:
@mikeller any more insight on this?

madratman, 2021-05-13:
Hi @haslinghuis, can one send control commands to betaflight via mavlink, or it only supports reading of telemetry?
https://github.com/betaflight/betaflight/blob/87955d97341ba5b05959846278d1ee78adb5d871/src/main/telemetry/mavlink.c looking at this file, it seems it's only the latter?

haslinghuis, 2021-05-13:
Afaik you are right

aniongithub, 2021-05-17:
> Did test myself and python is not consuming any heartbeat - not sure if it the heartbeat is there to begin with.
> 
> Just a silly question - code is TL;DR - should it be armed first to start the heartbeat? Does it need a crossfire receiver in Mavlink mode, etc, but yes it SITL not HITL, maybe need Gazebo or MavSDK to get it going. Should compare with Ardupilot / Pixhawk and maybe Inav implementation of Mavlink.
> 
> [image]
> 
> https://github.com/betaflight/betaflight/blob/2a5e457603a94efa0a18df2eb733ccc0ef0b1c1c/src/main/telemetry/mavlink.c#L512-L536
> 
> Some reference: https://research.ece.ncsu.edu/wireless/MadeInWALAN/CentMeshWikiBackup/tracwiki/raw-attachment/wiki/HardWare/MobileNode/Programming/Tutorial_drone-with-mavutilusage.pdf
> 
> Betaflight is excluded from being Mavlink capable: https://www.team-blacksheep.com/tbs-crossfire-mavlink-over-wifi.

I had some time over the weekend to go over some of those resources and try a number of different commands/methods to read telemetry. 

I think the issue as @madratman pointed out is that Betaflight doesn't support receiving commands over MAVLink.  This unfortunately also means I don't receive any telemetry with even a simple read loop like

```
# Get some information !
# From: https://www.ardusub.com/developers/pymavlink.html
while True:
try:
	print(master.recv_match().to_dict())
except:
	pass

time.sleep(0.1)	

```

because SITL is an inert device and without me sending it commands, has nothing to report?

closed 2021-06-25 by github-actions[bot]