betaflight/betaflight #10737: AirMode does not work propertly without Bidirectional DShot

CarlsonFPV, 2021-05-13:
### Describe the bug
If "Bidirectional DShot"  is On and AirMOde is Off indicator shows ACRO but in fact its Angle (I am new in hobby and can not tell exactly witch one) or the same but not ACRO. To have real ACRO with "Bidirectional DShot", AirMOde should be ON. 

### To Reproduce
"Bidirectional DShot"  = On and AirMOde = Off

### Expected behavior
I expect to have full control if I use ACRO. It should not depend on tuning settings.

### Flight controller configuration
All default. Min settings to fly.

### Setup / Versions
iFlight F722
HifionRC F722

etracer65, 2021-05-13:
Your assumptions are incorrect. Airmode has nothing whatsoever to do with bidirectional DSHOT. Airmode is not a flight mode - it's an enhancement to the PID controller and mixer to improve flight performance. Airmode works with all flight modes (acro, angle, horizon, etc.). The flight mode display in the OSD will show:

- `ANGL` for angle mode
- `HOR` for horizon mode
- `ATRN` for acro trainer
- `ACRO` for acro mode **without** airmode enabled
- `AIR` for acro **with** airmode enabled

CarlsonFPV, 2021-05-13:
@etracer65  I wrote steps to reproduce. Have you check it?

etracer65, 2021-05-13:
There is nothing to check. You have a misunderstanding of the flight modes. No further help can be provided until you provide the requested information as stated when you opened the issue.

> ### Flight controller configuration
> Create a diff and post it here in a code block. Put (three backticks) at the start and end of the diff block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/)
> Use resource show all to create a resource allocation list and post it here in a code block. Put (three backticks) at the start and end of the output block.
> ```
> PASTE THE OUTPUT OF 'diff' HERE
> ```
> ```
> PASTE THE OUTPUT OF 'resource show all' HERE
> ```
>

CarlsonFPV, 2021-05-13:
@etracer65 I do not need help. I tried to help BetaFlight. I wrote steps to reproduce and wrote the result. If you think that mentioned options and result are correct you may close this ticket. But if you have not check steps to reproduce I do not see any reason to discuss anything. Please check steps to reproduce and let me know the result.

etracer65, 2021-05-13:
If you don't provide the request information we can't assist you. I suspect that you're confusing the `ACRO TRAINER` flight mode with full acro flight. But without the requested information it's impossible to confirm. I suggest you see the wiki page for Acro Trainer.

CarlsonFPV, 2021-05-13:
@etracer65 I mentioned that all settings are default and this tickets is in Error scope.  You told me theory I told you about practice. Do you understand the difference?

closed 2021-05-13 by etracer65