betaflight/betaflight #10738: blackbox: Only 16 MB of 256 MB can be used to store flight data

renqiang06, 2021-05-13:
### Describe the bug
Blackbox: Only 16 MB of 256 MB in onboard dataflash  can be used to store flight data.

### To Reproduce

### Expected behavior
Onboard dataflash show the free space has 16.0 MB. But when i activated mass storage device (MSC) mode, it shows 256 MB left(filled by padding.txt). 

[image]
[image]

### Flight controller configuration

### Setup / Versions
 - Flight controller: Board: IFRC/IFLIGHT_F745_AIO(STM32F745)

etracer65, 2021-05-13:
This is correct. You only have 16MB of flash (check your flight controller specs from the manufacturer). The device mounts as a 256MB drive so that it will be recognized as a FAT32 device and not FAT16. The virtual "padding.txt" file is added to make it appear to be a 256MB drive to Windows.

renqiang06, 2021-05-13:
> This is correct. You only have 16MB of flash (check your flight controller specs from the manufacturer). The device mounts as a 256MB drive so that it will be recognized as a FAT32 device and not FAT16. The virtual "padding.txt" file is added to make it appear to be a 256MB drive to Windows.

got it. Thx

closed 2021-05-13 by renqiang06