betaflight/betaflight #10741: GPS data comes delayed (about 6 sec in my case)

SunjunKim, 2021-05-15:
### Describe the bug

On a flight, GPS data was delayed compared to barometer data (which was correct as of the recorded video). I compared the trend in GPS altitude with barometer altitude, and found the GPS altitude was lagging behind the barometer about 6 secs. 

This delay led my quad towards a crash when it failsafed, because the FC was reading the wrong GPS altitude from 6 secs ago when it gets into GPS rescue mode (usually starts with a rapid raising), causing a controlled descent to the ground when it started to cruise back to the home position.

I link my blackbox file here:
https://www.dropbox.com/s/xzl61kd5f5xb91d/LOG00056_gps_data_delayed.BFL?dl=0
* CSV exported: https://www.dropbox.com/s/1bnofkqn0qzo23s/LOG00056.01.csv?dl=0
* GPS exported: https://www.dropbox.com/s/ymhn9ipsombm78b/LOG00056.01.gps.csv?dl=0
* A graph comparing altitudes from barometer and GPS, check the delayed pattern.
[image]

### To Reproduce

I haven't found out how to replace this behavior. This is an anecdotal incident so far. 

### Expected behavior

The barometer and GPS altitudes should be aligned in time. I confirmed this behavior in other log files on the same-day flights. 

### Flight controller configuration

```
# 

# diff all

# version
# Betaflight / STM32F745 (S745) 4.2.8 Feb 15 2021 / 12:10:00 (101738d8e) MSP API: 1.43
# config: manufacturer_id: HBRO, board_name: KAKUTEF7, version: cc8e118f, date: 2020-03-04T02:37:20Z

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name KAKUTEF7
manufacturer_id HBRO
mcu_id 0025001e3039510338353231
signature 

# name: Holy119

# resources
resource CAMERA_CONTROL 1 B03

# timer
timer B03 AF1
# pin B03: TIM2 CH2 (AF1)

# dma
dma pin B03 0
# pin B03: DMA1 Stream 6 Channel 3

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature GPS
feature TELEMETRY
feature ESC_SENSOR

# beeper
beeper -BAT_LOW

# beacon
beacon RX_LOST
beacon RX_SET

[... 404 lines cut ...]
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

# 
```

### Setup / Versions

Build pic: https://i.imgur.com/dHW7EdG.jpeg
-------------------
Frame: XBee LB1 6"
FC: Holybro Kakute F7 v1.5
ESC: Holybro Tekko 32 F3 Metal
Motor: HobbyBrothers 2507 1750kv
GPS Unit: Betian BN-220T
Prop: DalProp 6x4.5x3
VTX: AKK Race Ranger
Radio: TBS Tracer
-------------------

 
 - How are the different components wired up: 
   - TBS Tracer to UART6 TX/RX
   - GPS to UART4 TX/RX
   - VTX SmartAudio to UART1

etracer65, 2021-05-15:
GPS altitude has limited accuracy and will often be significantly incorrect depending on the quality of the GPS satellite fix. You can choose how altitude is determined by adjusting the `position_alt_source` setting in the CLI.

Also GPS Rescue is intentionally designed to descend into the ground and crash when reaching the home position. It does not have the capability of auto-landing. GPS Rescue is only meant to get the quad back towards home so that **the pilot can regain control as soon as possible**. Do not allow it to fly all the way to the home position.

SunjunKim, 2021-05-16:
> GPS altitude has limited accuracy and will often be significantly incorrect depending on the quality of the GPS satellite fix. You can choose how altitude is determined by adjusting the [...]

1. I’m pretty sure it’s not a GPS accuracy issue. The GPS altitude is tracked (kinda) properly, but delayed compared to the quad’s real state.
2. It was not anywhere near the home position, actually about 2km away from the home. That descent behavior was triggered by the delayed GPS readings.

But thanks for pointing me position_alt_source CLI setting. I’ll try.

etracer65, 2021-05-16:
There is no mechanism to "delay" the GPS data. Betaflight simply reads what the GPS module supplies at that moment. So if you feel there's some accuracy problems then possibly try a different higher quality GPS module. The cheap cloned BN-220 units are not known for their quality or reliability.

And the "descent behavior" you refer to in GPS Rescue is not based on altitude, it's based on proximity to the home position. So if you believe you were 2km out and this happened then you have bigger problem with the position accuracy of your GPS module.

Also, how do you know that the barometer data was accurate as your baseline? For example have you properly isolated the sensor from airflow and light effects? The baro modules are extremely sensitive to changes in airflow (like from the props) and also are affected by light sources (like the sun). Both of these will significantly skew the altitude readings.

SunjunKim, 2021-05-16:
> There is no mechanism to "delay" the GPS data. Betaflight simply reads what the GPS module supplies at that moment. So if you feel there's some accuracy problems then possibly try a different higher quality GPS module. The cheap cloned BN-220 units are not known for their quality or reliability.

Yes I’m thinking about that also, and plan to change my GPS module. This delay may come from the GPS unit itself, and nothing wrong with BetaFlight.

> And the "descent behavior" you refer to in GPS Rescue is not based on altitude, it's based on proximity to the home position. So if you believe you were 2km out and this happened then you have bigger problem with the position accuracy of your GPS module.

Normally you’re right, However I believe that during that particular event, the FC was trying to maintain the set altitude during the GPS rescue (which was set “maximum” in my rescue setting, it would be around 110-120m based on the max. altitude + 15m during the flight). Because the GPS altitude was continuously increasing (because of the aforementioned delay), FC decided to descend the quad continuously to match the set altitude. The GPS LAT/LONG positions were tracked nominally without much error but the delay.

Most of all, you can see the delayed pattern in the graph I uploaded, which is clear. 

> Also, how do you know that the barometer data was accurate as your baseline? For example have you properly isolated the sensor from airflow and light effects? The baro modules are extremely [...]

I played a DVR (+HD recording from GoPro) video on BetaFlight BlackBox log viewer, and the barometer reading was showing the expected trends compared to the prop sound (which indicates the throttle for climbing/descending) and overall footage. The barometer is poor in accuracy which I agree, but it is very sensitive to the relative altitude changes which I believe worked OK in the incident.

Anyway at this point, I’m thinking it’s probably a bug from the GPS unit. I’ll check the related codes and make sure there’re no buffering mechanism in the BetaFlight code, and finally change my GPS unit. (Do you have any recommended module?)

McGiverGim, 2021-05-16:
If it helps I only can say that maybe the GPS altitude is slowly, but for sure is not 6 seconds later. I use it a lot and never noticed this kind of magnitude. Maybe one or two seconds, no more.

MrMap, 2021-05-16:
The GPS unit has an internal filter whos characteristics is determined by the Dynamic Platform Model in use.
The best fitting Dynamic Platform Model can be selected by the user (commands sent to the GPS module).
If an unsuitable model is used, e.g. Pedestrian, the model will be lagging heavily (position, speed and course changes will be delayed) compared to how a fast moving and fast accelerating quad moves. 

iNAV has a cli command to set the desired model, and so has ArduPilot/ ArduCopter.
Does Beta Flight provide this capacity to communicate the choice of Dynamic Platform Model to the GPS unit? 
I am new to BF and searched for it but no joy.

A google search will provide further info on this topic for anyone looking for deeper info.
Here is one link:
https://wiki.paparazziuav.org/wiki/Module/GPS_UBlox_UCenter
And I bet the uBlox data sheets covers it as well.

I am not saying this is cause of the problem you described, but it came to my mind.
I have just begun using a new RTF quad and my initial flights has given me reason to suspect GPS data update lagging too - but I have to get some more flights to observe if there seems to be an issue.
/Tom

McGiverGim, 2021-05-16:
It was configured for airborne some time ago (if you select auto config, of course): https://github.com/betaflight/betaflight/pull/3110

Some new PR has modified this code a little but the big part remains unchanged.

MrMap, 2021-05-17:
Ok, thank you for linking to that issue record.
Do you know if Airborne 4G was the final choice for the gps_auto_config?
Or, Airborne 2G, as implied by the title of the issue ("Changed gps_auto_config to set mode to "airborne <2g")

It just seems odd to me that 2g was implemented while everybody seemed to agree that 4g was preferable...

McGiverGim, 2021-05-17:
By default it uses airborne 4G: 
https://github.com/betaflight/betaflight/blob/dfcd11bd691fdf530e6c684ce1e13129db28ed11/src/main/io/gps.c#L164

McGiverGim, 2021-05-17:
A one thing more: it was made selectable some time ago: https://github.com/betaflight/betaflight/pull/9388

SunjunKim, 2021-05-24:
I tried to reproduce this bug for a while but no success.
Maybe it's just a very rare occasional bug from my GPS unit. I'm planning to replace it.

For you guys, REMEMBER: check your GPS is responding properly as soon as you launch when doing a long-range flight.

closed 2021-05-24 by SunjunKim