betaflight/betaflight #10746: Configuration parameter runaway_takeoff_prevention = FIRST_ARM

joewak, 2021-05-17:
### Is your feature request related to a problem? Please describe.
I'm always frustrated when "Runaway Takeoff Prevention" prevents me from getting back up in the air quickly after crashing during a race.  This often happens when in longer grass or in some situation where the props are partially obstructed.  In the heat of a race I would like to be able to use as much throttle as needed to try to clear the props and get back up in the air with the understanding that damage to the quad is likely.

### Describe the solution you'd like
I would like for the configuration parameter "runaway_takeoff_prevention" to have a new possible value of "FIRST_ARM".  This configuration parameter would have the possible values of OFF, ON, FIRST_ARM.  OFF and ON would behave as they do now.  FIRST_ARM would only allow the feature to activate on the first arm after power up, it would not activate after Crash-Flip.

### Describe alternatives you've considered
I've considered disabling the feature completely but I prefer to have the protection for the first arm. 

I believe this configuration option would allow the feature to be more in line with the original intent of the feature as described here.

[image]

etracer65, 2021-05-17:
Sounds reasonable, although I don't like the "FIRST_ARM" option naming. The feature already only works on the first arming during normal flight conditions. So having an option named like this would imply that it's active for every arming which is not the case. The naming should give some indication that it's preventing the crash-flip reset.

joewak, 2021-05-17:
"Sounds reasonable" -- Honored!

Some possible alternative names:
OFF, ON, RACE
OFF, ON, NO_CRASH_FLIP
OFF, ON, CRASH_FLIP_NO_RESET

I'm going to take a crack at doing a pull request for this feature.  I'm open to any name.

haslinghuis, 2021-05-17:
Please just go ahead, naming is the hardest part :smile: reviewers will guide this on stage.

etracer65, 2021-05-18:
I'm thinking it's better to have a separate setting to control this and leave the existing main on/off. Suggest adding a `runaway_takeoff_prevention_crashflip_reset` on/off that defaults to "ON".

babipsylon, 2024-06-18:
Had this feature been implemented? Does not seem so?

Only found this:

# get runaway_takeoff_prevention
runaway_takeoff_prevention = ON
Allowed values: OFF, ON

# get runaway_takeoff_deactivate_delay
runaway_takeoff_deactivate_delay = 500
Allowed range: 100 - 1000

# get runaway_takeoff_deactivate_throttle_percent
runaway_takeoff_deactivate_throttle_percent = 20
Allowed range: 0 - 100

closed 2026-09-26 by blckmn