betaflight/betaflight #10748: Air mode - depending on throttle

nightflasher, 2021-05-18:
### Is your feature request related to a problem? Please describe.

Air mode is needed, when throttle position is low or next to zero, so you don't need it switched on all the time during flight. With air mode switched on all the time (e.g. on ducted kwads), the kwad over-reacts to any kind of impact (sucking to walls). 
Spinning motors higher for no reason, or just a small bump against small branches. And even landing the kwad can be danger sometimes, when air mode "detects something wrong". Imagine an x-class landing with air mode on an spinning the motors up again! 

### Describe the solution you'd like
Just like Joshua Bardwell in his video (https://youtu.be/tCTI2J0QCwc) said, and described the workaround in openTX. 
And I've added in my comment:
Add another value to the "air mode - feature switch", like the "arming angle - value", when to 'dynamically' turn on/off air mode by throttle position in percentage. 

### Describe alternatives you've considered
For now, I have air mode on an aux switch, which also is my fm switch where only angle has no air mode. So I always have to switch to angle mode, to land "save".

etracer65, 2021-05-18:
> Air mode is needed, when throttle position is low or next to zero, so you don't need it switched on all the time during flight. 

This is not true and a misunderstanding of how airmode works and how it improves flight.

> And even landing the kwad can be danger sometimes, when air mode "detects something wrong".

How would your proposal help this? I assume you are at "low throttle" when trying to land.

nightflasher, 2021-05-19:
Oh i see, I forgot to mention a "general off for air mode" when landing and take off.
And I definitely forgot some more - I was way too tired to get everything correct 🙈
Maybe someone else will take over this topic and add missing informations.

ctzsnooze, 2021-05-22:
Many people have air mode active only on the third position of a three position arming switch.
Quite a lot of whoop and ducted type quad pilots have reported that they 'stick to walls' far more aggressively when airmode is on.  This is kind of expected, since propellers push air backwards, and when enclosed in a duct, they are like little vacuum cleaners.
Airmode just makes responses stronger when one pair of motors is commanded to zero.  Having it off won't prevent the quad sticking to walls, or reacting to impacts.
If you don't plan to flip or roll, just fly with airmode off, and the machine will be a little less aggressive in reactions to walls, but will still stick and will still react to impacts.  
If you don't want to stick to walls, don't hit them :-)
If you want airmode to be automatically off at zero throttle, the quad won't flip or roll properly at zero throttle.  In fact it is most useful at zero throttle, that's when you particularly do want it on :-)
If you want to make a good 'un-stick' method for whoops etc, use a three position arm switch, and enable motor stop.  Put airmode active only in the third position.  
In the third position, at zero throttle you can flip and roll as much as you like, it will be fine, but it will stick strongly to the wall.
In the second position, the motors will stop at zero throttle, and you cannot do zero throttle flips and rolls.  So you have to fly without ever going to zero throttle :-) But this position is good for landing and for un-sticking from the wall.  If you are stuck, move the switch to the second position and cut throttle, the motors will stop and you will fall off, if you are quick you may catch the quad by throttling up before it hits the ground.  Also on landing, move to the second position as you approach landing, lower the quad close and then cut throttle, and the motors stop and it won't bounce around.
There is a danger in using motor stop; it can be that you have forgotten to properly set your arm switch to off, and it is in the position 2, then just bumping throttle will make the quad leap up.  Hence you need to always properly disarm after landing and use a 2-step arming sequence if you use motor stop.

haslinghuis, 2021-05-22:
Did anyone actually see the video. Setup is done on OpenTX. Using throttle and idle up switch with arm switch to set airmode accordingly.

haslinghuis, 2021-05-22:
Did anyone see the video? Like to have some comments on the video from JB as I like the proposal doing this in OpenTX.

TomAlperin, 2021-06-26:
Saw the video and did some tests. Activating air mode based on the throttle held position better than no air mode but still not as good as having air mode on all the time. You get better performance putting air mode on a switch or permanently enabling it. It’s easy enough to set this up on the transmitter and it doesn’t take additional channels to have options for off, on, and based on throttle.

bigspeedfpv, 2021-07-07:
If you *really* wanted this, you could probably set it up as a logical switch through OpenTX (or whatever firmware your radio uses) and add air mode as a mode switch in Betaflight... but as the others have said, this probably isn't going to help the issue

haslinghuis, 2021-07-07:
It's unlikely this would result in a code change on our side as this is just a preference and configuration issue.

closed 2021-07-07 by haslinghuis

tenzmedia, 2022-05-02:
Not sure if this still works since the thread is closed, but I also was thinking about what the benefit is of havin air mode on constantly instead of using it like JB said it in his video? Wouldnt that be a "better" solution for whoops as well as for freestyle quads? I mean if you're flying with throttle at lets say 35%, you dont need air mode anyway right? On a whoop, it results in wall-sucking when hitting something, but since some whoops are able to do some acro manouvers like diving and stuff, it would be great if this is happening dynamicly like we all learned in JBs video. I'm pretty sure my thinking is too simple and there's a reason the devs havent set it up like that. I just try to understand what that reason could be...