betaflight/betaflight #10758: unexpected failsafe occur during flight with OMNINXT7 4.2.9 - 27-04-2021 21:40 and never occured with 4.2.8 - 15-02-2021 13:24

sybu0, 2021-05-23:
unexpected failsafe often occur during flight with OMNINXT7 4.2.9 - 27-04-2021 21:40 and never happened with 4.2.8 - 15-02-2021 13:24.
I went back to 4.2.8 release and the issue did not happen again.

The Flight Controller goes into Failsafe mode without any error message.

RX: integrated in FC R9MM 

### Flight controller configuration:

# diff

# version
# Betaflight / STM32F7X2 (S7X2) 4.2.9 Apr 27 2021 / 19:34:29 (e097f4ab7) MSP API: 1.43
# config: manufacturer_id: AIRB, board_name: OMNINXT7, version: 30af1b1d, date: 2019-10-19T08:28:02Z

# start the command batch
batch start

board_name OMNINXT7
manufacturer_id AIRB

# resources
resource MOTOR 1 B01
resource MOTOR 2 B00
resource MOTOR 3 B04
resource MOTOR 4 B05

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature MOTOR_STOP
feature GPS
feature TELEMETRY
feature DISPLAY
feature ESC_SENSOR

# beacon
beacon RX_LOST
beacon RX_SET

# map
map AERT1234

# serial
serial 1 64 115200 57600 0 115200
serial 3 2048 115200 57600 0 115200
serial 4 1024 115200 57600 0 115200
serial 5 2 115200 57600 0 115200

# aux
aux 0 0 0 900 1975 1 0
aux 1 0 7 900 1450 0 0
aux 2 0 7 900 1925 1 0
aux 3 1 2 1800 2100 0 0
aux 4 2 2 1300 1700 0 0
aux 5 27 5 1150 2100 0 0
aux 6 13 8 1900 2100 0 0

# adjrange
adjrange 0 0 1 900 2100 29 1 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 4
vtxtable powervalues 0 1 2 3
vtxtable powerlabels 25 200 600 800

# master
set acc_trim_pitch = 2
set acc_calibration = -30,-30,-74,1
set align_mag = CW270FLIP
set mag_align_pitch = 1800
set mag_align_yaw = 2700
set mag_calibration = -91,38,-16
set rssi_channel = 16
set fpv_mix_degrees = 20
set serialrx_provider = FPORT
set serialrx_halfduplex = ON
set rx_spi_protocol = FRSKY_X
set blackbox_p_ratio = 16
set dshot_idle_value = 350
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set failsafe_delay = 50
set failsafe_throttle = 1400
set failsafe_switch_mode = STAGE2
set failsafe_procedure = GPS-RESCUE
set align_board_yaw = 180
set bat_capacity = 1600
set vbat_min_cell_voltage = 310
set ibata_scale = 100
set ibata_offset = -2500
set beeper_dshot_beacon_tone = 3
set yaw_motors_reversed = ON
set gps_provider = UBLOX
set gps_ublox_use_galileo = ON
set gps_rescue_angle = 30
set gps_rescue_initial_alt = 40
set gps_rescue_descent_dist = 60
set gps_rescue_ground_speed = 1000
set gps_rescue_throttle_min = 1200
set gps_rescue_throttle_max = 1850
set gps_rescue_ascend_rate = 200
set gps_rescue_descend_rate = 100
set gps_rescue_throttle_hover = 1300
set gps_rescue_min_sats = 10
set pid_process_denom = 2
set osd_warn_batt_not_full = OFF
set osd_warn_rssi = ON
set osd_rssi_alarm = 40
set osd_cap_alarm = 1800
set osd_tim1 = 2563
set osd_tim2 = 2560
set osd_rssi_pos = 14434
set osd_link_quality_pos = 409
set osd_rssi_dbm_pos = 129
set osd_throttle_pos = 375
set osd_vtx_channel_pos = 14720
set osd_ah_pos = 8333
set osd_current_pos = 8534
set osd_gps_speed_pos = 8471
set osd_gps_sats_pos = 14404
set osd_home_dir_pos = 14445
set osd_home_dist_pos = 14465
set osd_flight_dist_pos = 129
set osd_altitude_pos = 14487
set osd_warnings_pos = 14688
set osd_avg_cell_voltage_pos = 14743
set osd_disarmed_pos = 14734
set osd_stat_tim_2 = OFF
set osd_stat_max_dist = ON
set osd_stat_max_alt = ON
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set osd_stat_flight_dist = ON
set osd_profile = 3
set vtx_band = 4
set vtx_channel = 4
set vtx_power = 1
set vtx_freq = 5800
set gyro_2_sensor_align = CW180
set gyro_2_align_yaw = 1800

profile 0

# profile 0
set p_pitch = 56
set i_pitch = 99
set d_pitch = 46
set f_pitch = 105
set p_roll = 51
set i_roll = 94
set d_roll = 43
set f_roll = 99
set p_yaw = 55
set i_yaw = 99
set f_yaw = 99
set d_min_roll = 28
set d_min_pitch = 31

rateprofile 0

# end the command batch
batch end

# 
```
```
PASTE THE OUTPUT OF 'resource show all' HERE
# 

# resource
resource BEEPER 1 C13
resource MOTOR 1 B01
resource MOTOR 2 B00
resource MOTOR 3 B04
resource MOTOR 4 B05
resource PPM 1 B07
resource LED_STRIP 1 A09
resource SERIAL_TX 1 B06
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 A00
resource SERIAL_TX 6 C06
resource SERIAL_RX 1 B07
resource SERIAL_RX 2 A03
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 A01
resource SERIAL_RX 5 D02
resource SERIAL_RX 6 C07
resource I2C_SCL 1 B08
resource I2C_SDA 1 B09
resource LED 1 B02
resource SPI_SCK 1 A05
resource SPI_SCK 2 B13
resource SPI_SCK 3 C10
resource SPI_MISO 1 A06
resource SPI_MISO 2 B14
resource SPI_MISO 3 C11
resource SPI_MOSI 1 A07
resource SPI_MOSI 2 C03
resource SPI_MOSI 3 C12
resource CAMERA_CONTROL 1 B15
resource ADC_BATT 1 C00
resource ADC_RSSI 1 C04
resource ADC_CURR 1 C01
resource ADC_EXT 1 A04
resource BARO_CS 1 A10
resource FLASH_CS 1 C14
resource OSD_CS 1 A15
resource GYRO_CS 1 B12
resource GYRO_CS 2 A08

# 

Blackbox Status just when the problem occured:

Flight Mode Flags | 67 | ARM\|ANGLE\|FAILSAFE
PID Feedforward [pitch] | 0 | 0.0% | State Flags | 3 | GPS_FIX_HOME\|GPS_FIX
PID Feedforward [yaw] | 0 | 0.0% | Failsafe Phase | 0 | IDLE
RC Command [roll] | -103 | 1397 us | RX Signal Received | 1 | &nbsp;
RC Command [pitch] | -23 | 1477 us | RX Flight Ch. Valid | 1

<br class="Apple-interchange-newline">

Blackbox Status just before the problem occured:

Flight Mode Flags | 3 | ARM\|ANGLE
PID Feedforward [pitch] | 0 | 0.0% | State Flags | 3 | GPS_FIX_HOME\|GPS_FIX
PID Feedforward [yaw] | 0 | 0.0% | Failsafe Phase | 0 | IDLE
RC Command [roll] | -104 | 1396 us | RX Signal Received | 1 | &nbsp;
RC Command [pitch] | -23 | 1477 us | RX Flight Ch. Valid | 1

McGiverGim, 2021-05-24:
You can attach with drag&drop the blackbox file. It can be useful.

sybu0, 2021-05-24:
Hello McGiverGim,

You will find the issue at the end of 2/3 part. The bbl file is renamed jpg 

Best regards

[image]

sybu0, 2021-05-24:
It will be easier with a zip file I suppose, sorry...

You will find the issue at the end of 2/3 part. 

[btfl_001.bbl.csv.zip](https://github.com/betaflight/betaflight/files/6530468/btfl_001.bbl.csv.zip)

sybu0, 2021-05-24:
bbl file:
[btfl_all.zip](https://github.com/betaflight/betaflight/files/6530530/btfl_all.zip)

sybu0, 2021-06-14:
As the issue didn't happen again, I suppose it was due to communication issue between esc and flight controller, so I close the issue.
Regards

closed 2021-06-14 by sybu0