betaflight/betaflight #10759: Angel/Horizon Mode: Only for roll or pitch

maxledfreak, 2021-05-24:
### Is your feature request related to a problem? Please describe.
No, not related to a problem. Everything works fine 👍 

### Describe the solution you'd like
Feature request: Angel or Horizon mode is selectable for roll or pitch axis or both. 

### Describe alternatives you've considered
See above.

dkustec, 2021-05-24:
Check #9481.
Not fast enough race mode. This is an acro - level mode hybrid. Level mode is used for the roll axis and acro mode is used for pitch.

This feature will only work in level mode and can be enabled via CLI:

`set level_race_mode = ON`

maxledfreak, 2021-05-24:
Thanks for this info! I will test it. But regarding GPS rescue is not clear for im if it works correct with this setting. Do you have any informations about this ?

closed 2021-07-01 by github-actions[bot]