betaflight/betaflight #10760: Enable Barometer and Magnetometer/Compass hardware via MSP

kafie1980, 2021-05-24:
The MATEKSYS M8Q-CAN can provide GPS/BARO/COMPASS data over MSP : http://www.mateksys.com/?portfolio=m8q-can#tab-id-6

This module is a great add-on for long range quads especially since its an all in one module and the data cant be sent over a single UART via MSP.

It seems that GPS support via MSP has already been added to Betaflight: https://github.com/betaflight/betaflight/pull/6662

GPS over MAP can be setup via the Configurator GUI> Configuration Page> GPS section:

 [image]

Or via the CLI:

[image]

However, Magnetometer and Barometer hardware options do no provide support for MSP:

[image]

[image]

One option for the Barometer (DPS310) and Magnetometer (QMC5883L) will be to connect them via the SCL/SDA I2C pads, however it is not clear if the DPS310 barometer is supported as it is not listed in the barometer hardware listing via the CLI.

jflyper, 2021-06-18:
I think this is a de-sync between BF and BFC. Can you raise this issue at the configurator repo. We would also appreciate if you can supply a suggested change as a PR :)

kafie1980, 2021-06-18:
I am not sure what you mean by a desync between BF and BFC? There is no MSP support for Compass or Barometer in Betaflight.

I am not sure what it will take to merge into BF existing MSP based senor support for GPS+Mag+Baro from existing flight control firmware's:

iNav: https://github.com/iNavFlight/inav/pull/6090

ArduPilot: https://github.com/ArduPilot/ardupilot/pull/15232

kafie1980, 2021-06-22:
@MATEKSYS seems to be active on the Betaflight github submitting providing support for their flight controller boards. Perhaps @MATEKSYS can submit a PR to enable full MSP support for the MATEKSYS M8Q-CAN GPS/Barometer/Compass module.

kafie1980, 2021-07-26:
I have tested the M8Q-CAN connected via MSP and I2C and even the GPS data is not received over MSP.

BetaFlight implementation of MSP will not work with the M8Q-CAN GPS module and it seems that the I2C port is only for the AirSpeed sensor rather than a passthrough for the FC to connect the Magnetometer and Barometer. 

I have tested the M8Q-CAN unit (also updated to the latest firmware) and FC and rechecked all the wiring/configuration to ensure everything is correctly setup.

Eventually I test flashed iNav and everything worked as expected.

robustini, 2022-05-31:
Isn't there an evolution on this request?
INAV and Ardupilot yes and Betaflight no?
Impossible! ;-)

kafie1980, 2022-06-24:
> @kafie1980 I tried today with Betaflight 4.3 and the gps in MSP is not seen, with iNav instead now everything works for me, also the barometer with a new fix for this. Is it possible that with the 4.3 something has been introduced that bothers the GPS?

MSP based GPS/Barometer/Compass is currently not supported in Betaflight and this has been flagged as a feature request.

Perhaps in the future.

trenth00, 2024-02-12:
Would be great to get this feature to enable easy access to high performance Mags such as RM3100 without having to provide explicit device support

alexrjohnston, 2024-04-17:
I would also love to see this happen!

trenth00, 2024-04-18:
#13448

MATEKSYS, 2024-04-18:
We have sent M10Q-5883 and CAN-L431 samples to some BF devs in last Nov.
CAN-L431 node with AP_Periph fw supports GPS/Mag MSP output if connnecting it with M10Q-5883.
No idea how is the progress.

trenth00, 2024-04-18:
> We have sent M10Q-5883 and CAN-L431 samples to some BF devs in last Nov. CAN-L431 node with AP_Periph fw supports GPS/Mag MSP output if connnecting it with M10Q-5883. No idea how is the progress.

There seems to be interest in CAN support with Betaflight.. now that would be really good and something even iNav doesn’t yet have

closed 2026-09-26 by blckmn