betaflight/betaflight #10763: KISSFCV2F7 Target - Crossfire receiver not working

Zuldan, 2021-05-25:
### Describe the bug
Crossfire Receiver not working. Cannot see any stick inputs. 

### To Reproduce
Flash Kiss FC v2 with KISSFCV2F7 BF v4.2.9 target using Kiss GUI Firmware Flasher. 

### Expected behavior
Receiver should receive stick inputs when enabling Serial RX on UART2, setting Serial Receiver Provider to CRFS. 

### Flight controller configuration
```
# Betaflight / KISSFCV2F7 (KISSV2) 4.2.9 Apr 27 2021 / 20:02:49 (e097f4ab7) MSP API: 1.43

# start the command batch
batch start

board_name KISSFCV2F7

# serial
serial 4 2048 115200 57600 0 115200

# master
set acc_calibration = 19,-84,176,1
set serialrx_provider = CRSF
set motor_pwm_protocol = DSHOT600

profile 0

rateprofile 0

# end the command batch
batch end
```
```
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: MOTOR 2
A01: FREE
A02: SERIAL_TX 2
A03: SERIAL_RX 2
A04: FREE
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: LED 1
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: FREE
A12: FREE
A13: SWD
A14: SWD
A15: FREE
B00: FREE
B01: ADC_BATT
B02: FREE
B03: FREE
B04: FREE
B05: FREE
B06: MOTOR 4
B07: FREE
B08: MOTOR 3
B09: FREE
B10: FREE
B11: FREE
B12: GYRO_CS 1
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: FREE
C03: FREE
C04: FREE
C05: FREE
C06: MOTOR 1
C07: FREE
C08: LED 2
C09: BEEPER
C10: SERIAL_TX 4
C11: SERIAL_RX 4
C12: SERIAL_TX 5
C13: FREE
C14: FREE
C15: FREE
D02: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4:
    CH1: MOTOR 4
    CH3: MOTOR 3
TIM5:
    CH1: MOTOR 2
TIM6: FREE
TIM7: FREE
TIM8:
    CH1: MOTOR 1
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: MOTOR 4
DMA1 Stream 1: FREE
DMA1 Stream 2: MOTOR 2
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: MOTOR 3
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: MOTOR 1
DMA2 Stream 3: FREE
DMA2 Stream 4: ADC
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

### Setup / Versions
 - Flight controller: Kiss FC v2. Bought from Flydruino website. 
RX = TBS Crossfire Nano
 - Other components:
 Receiver wired up to UART2
 - How are the different components wired up:
VTX (TBS Unify HV) wired to UART5 (TBS Smart Audio working perfectly)

Receiver works 100% when using Kiss firmware on UART2. Flashing the same quad with Betaflight v4.2.9 (configure the receiver) and it no longer works.
[image]
[image]
[image]

klutvott123, 2022-04-17:
@Zuldan is this still an issue? If so try:
```
set serialrx_inverted = OFF
set serialrx_halfduplex = OFF
```

Zuldan, 2022-04-23:
@klutvott123 apologies for the delay. I got some time to test those settings today on the latest master image 22/04.

Tried the following combinations but the receiver is still not working.

set serialrx_inverted = OFF
set serialrx_halfduplex = OFF

set serialrx_inverted = ON
set serialrx_halfduplex = ON

set serialrx_inverted = OFF
set serialrx_halfduplex = ON

set serialrx_inverted = ON
set serialrx_halfduplex = OFF

I also tried @spatzengr (UAV Tech) suggestion of swapping the RX/TX wires on the FC (makes no sense but he's had to do this during some testing he did). I then tried the combination of settings again above but still no luck :(

[Betaflight Kiss F7 Diff.txt](https://github.com/betaflight/betaflight/files/8545805/Betaflight.Kiss.F7.Diff.txt)

[Betaflight Kiss F7 Dump.txt](https://github.com/betaflight/betaflight/files/8545806/Betaflight.Kiss.F7.Dump.txt)

klutvott123, 2022-04-23:
@Zuldan the diff you posted last shows no serial port configured like in the diff in the first post

Zuldan, 2022-04-23:
The UART2 was configured for serial by default when I flashed it today. I’m guessing that’s why it’s not showing in the diff.

S-T-A, 2022-07-07:
did you get it to work i'm havinf the same issue exactly like you.

Zuldan, 2022-07-07:
I haven’t had time to look at the issue again but UAV Tech recently released this video. Have a look at his wiring at the end. Let us know if you have any luck.

https://youtu.be/0imE8n4gi2s

haslinghuis, 2023-10-27:
Closing as stale.

closed 2023-10-27 by haslinghuis