betaflight/betaflight #10769: Can't change the blackbox logging on MatekF722SE Firmware.

SmartFPV, 2021-05-28:
The MatekF722SE Firmware is used in the HAKRC F722 mini flight controller. The Matek FC has a SD-card slot. So the BB is set to SD-card. 
The HAKRCF722 has a spi flash memory on it. 

### Describe the bug
I tried to set the blackbox logging to spi_flash in the CLI. But it won't recognize the flash chip and/or is set back to no logging after rebooting the FC.

### To Reproduce
Flash the HAKRC F722 FC to the Matek F722 firmware and black box wont work.

### Expected behavior
After changing the settings in the CLI, I expected it to work, but it still showed critical error.

### Flight controller configuration

```
PASTE THE OUTPUT OF 'diff' HERE
```
board_name MATEKF722SE
manufacturer_id MTKS

# resources
resource MOTOR 1 B00
resource MOTOR 2 B01
resource MOTOR 3 B04
resource MOTOR 4 B05

# beacon
beacon RX_LOST
beacon RX_SET

# serial
serial 0 8192 115200 57600 0 115200

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 1 1300 1700 0 0
aux 2 13 2 1700 2100 0 0
aux 3 35 1 900 1300 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 FATSHARK F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 4 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 5 IMD6     I CUSTOM  5362 5399 5436 5473 5510 5547 5584 5621
vtxtable powerlevels 5
vtxtable powervalues 25 100 200 400 1000
vtxtable powerlabels 25 100 200 400 1W

# master
set gyro_lowpass_hz = 0
set gyro_lowpass2_hz = 0
set dyn_notch_width_percent = 6
set dyn_lpf_gyro_min_hz = 0
set acc_calibration = 31,110,51,1
set rc_smoothing_derivative_type = PT1
set rc_smoothing_auto_smoothness = 14
set serialrx_provider = CRSF
set blackbox_p_ratio = 64
set blackbox_device = NONE
set dshot_bidir = ON
set beeper_dshot_beacon_tone = 3
set yaw_motors_reversed = ON
set small_angle = 180
set sdcard_spi_bus = 0
set osd_link_quality_pos = 2103
set osd_rssi_dbm_pos = 2135
set osd_tim_2_pos = 2081
set osd_avg_cell_voltage_pos = 2092
set osd_stat_max_spd = OFF
set osd_stat_max_alt = ON
set osd_stat_bbox = OFF
set osd_stat_max_esc_rpm = ON
set debug_mode = GYRO_SCALED
set vtx_band = 1
set vtx_channel = 8
set vtx_power = 4
set vtx_freq = 5725
set flash_spi_bus = 3
set gyro_1_align_pitch = 1800
set gyro_1_align_yaw = 1800
set gyro_2_align_yaw = 900

profile 0

# profile 0
set dyn_lpf_dterm_min_hz = 110
set dyn_lpf_dterm_max_hz = 270
set dterm_lowpass2_hz = 245
set anti_gravity_gain = 5500
set iterm_relax_cutoff = 22
set throttle_boost = 2
set p_pitch = 81
set i_pitch = 99
set d_pitch = 74
set f_pitch = 84
set p_roll = 74
set i_roll = 94
set d_roll = 68
set f_roll = 79
set p_yaw = 79
set i_yaw = 99
set f_yaw = 79
set d_min_roll = 44
set d_min_pitch = 48

rateprofile 0

# rateprofile 0
set roll_rc_rate = 88
set pitch_rc_rate = 88
set yaw_rc_rate = 80
set roll_expo = 10
set pitch_expo = 10
set yaw_expo = 20
set roll_srate = 77
set pitch_srate = 77
set tpa_rate = 70
set tpa_breakpoint = 1410

# end the command batch
batch end

```
PASTE THE OUTPUT OF 'resource show all' HERE
```
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: SERIAL_TX 2
A03: SERIAL_RX 2
A04: FREE
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: SERIAL_TX 1
A10: FREE
A11: USB
A12: USB
A13: LED 2
A14: LED 1
A15: FREE
B00: MOTOR 1
B01: MOTOR 2
B02: GYRO_CS 1
B03: FREE
B04: MOTOR 3
B05: MOTOR 4
B06: FREE
B07: FREE
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: FREE
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_CURR
C02: ADC_BATT
C03: FREE
C04: GYRO_EXTI
C05: FREE
C06: FREE
C07: FREE
C08: PINIO 1
C09: PINIO 2
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: BEEPER
C14: FREE
C15: PREINIT 2
D00: FREE
D01: FREE
D02: PREINIT 3
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1: DSHOT_BITBANG 2
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 2
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

### Setup / Versions
 <!-- Specify your flight controller model (what type is it, where was it bought from, ...) -->
 - Flight controller:  HAKRC F722 DM Mini
 <!-- Specify other components attached to the flight controller (RX, VTX, brand / model for all of them, firmware version where applicable...) -->
 - Other components: Happymodel EP1, HifionRC Tanso v2, Runcam Swift 3
 <!-- Details about how all is wired -->
 - How are the different components wired up:
RX on UART2, VTX on TX1

<!-- Add any other context about the problem that you think might be relevant here. -->

haslinghuis, 2021-07-09:
Did you raise the question with HAKRC support as I don't have the hardware to support?
Also you left out the version information in the `diff`

SmartFPV, 2021-07-09:
> Did you raise the question with HAKRC support as I don't have the hardware to support?
> Also you left out the version information in the `diff`

No I didnt. I dont think that HAKRC is even a company. They dont have a website, a support, a customer service or even a business regestration. So unfortunally I see no chance to get in contact with them directly.

The version I used is Betaflight 4.2.9. Sorry that I missed that.

haslinghuis, 2021-07-09:
I don't what this means: `D02: PREINIT 3`, this should be `resource SDCARD_CS 1 D02` as it's in MATEKF722SE.config.

Are you sure it's the right target for your board, you could try MATEKF722 or HAKRCF722 as there are different versions of the board. With the Boot button you always return to another flash.

shro70, 2021-08-10:
I'm able to get the blackbox 16m chip working by using hlrg f722 target. The only big issue is no gyro/accel lol.

SergPle, 2021-09-07:
Just use MATEK722MINI and change the line in the configuration file to connect the barometer (replace NONE with BMP280) And enjoy!

BrH-83, 2021-11-24:
> I'm able to get the blackbox 16m chip working by using hlrg f722 target. The only big issue is no gyro/accel lol.

I’m having this issue.  Has anyone resolved this.  My fc will not exit dfu unless I flash the matek f722 std firmware downloaded from their site.  I ask their customer support for help but they wouldn’t oblige.

haslinghuis, 2021-11-24:
Should ask HAKRC about the correct target. GetFPV uses Matek F722 and Banggood Matek F722. Both list version 4.1.
There is no website listed.

@DusKing1 Do you have more info on HAKRC. @mikeller Should we remove it from BF?

freasy, 2022-03-28:
The solution to this is:

`
resource SDCARD_CS 1 none
resource FLASH_CS 1 D02
set blackbox_device = SPIFLASH
set flash_spi_bus = 3
save
`

shro70, 2022-03-31:
Thanks. I use the MATEKF722MINI target as suggested by SergPle

haslinghuis, 2022-04-02:
@freasy let's do it

haslinghuis, 2022-04-15:
Fixed in: https://github.com/betaflight/unified-targets/pull/582

closed 2022-04-15 by haslinghuis