betaflight/betaflight #10773: Support for rp2040

debuti, 2021-05-29:
### Is your feature request related to a problem? Please describe.
No. I think adding a wonderful multicore device such as the rp2040 would greatly push the boundaries

### Describe the solution you'd like
Rp2040 is supported by betaflight

### Describe alternatives you've considered
Current MCUs are quite powerful, but in order to continue evolving the multicore solution is a must. 133mhz driving 2x m0+s for pretty cheap

ItsMeAubey, 2021-05-29:
As far as I understand the RP2040 is literally a quarter as fast as current f722... Getting it to 1:1 coremark parity requires a huge overclock and overvolt. STM32H7 is the future, not this.

jflyper, 2021-06-18:
In addition, M0+ usually lacks FPU, which will be another reason to avoid RP2040 (if it does not have FPU).