betaflight/betaflight #10779: External gyro support

and-sh, 2021-06-01:
Is there somewhere information how to use external imu?

Today I have some questions:
Does SPI imu use int pin?
In case of one imu on spi bus can I leave cs pin always high?
Can I disable second imu search with:
`set gyro_2_spibus = 0`?
Does I2C imu use int pin?
Do I need to set the correct address
`set gyro_1_i2c_address = 104`?

What are the developers' plans for using the both imu
`gyro_to_use = both`?

jflyper, 2021-06-18:
@and-sh 

I believe we do not use INT signal from gyros atm, and you can connect external gyro chip, configure it by gyroX_XXX_bus and other parameters, and select which gyro to use by gyro_to_use.

As an additional note,  I do not recommend using I2C gyros. They are too slow to read, and will most likely to clobber your cycle time and result in poor flight performance.

and-sh, 2021-06-18:
Thank you.

It didn't work with i2c, my firmware does not support i2c gyroscope. Examining the source code showed that it depends on the amount of memory.  I decided that doing my own firmware is perspectiveless.

There are 3 spi on my board (matek 722mini se, stm32f722 lqfp64). All in use:osd, imu, blackbox. I have not found any alternative pins that would go to the edge of the board.

I unsoldered the imu (mpu6000), all additional pins are wired: cs and int.
Thus, now the imu requires 7 wires: gnd, vdd, miso, mosi, sck, cs, int.

I have external imu board with mpu6000. I'll try to solder it next week.
If we manage to reduce the number of wires to 5 it will be good: no int, always cs.

and-sh, 2021-06-18:
One more question: as I understand current list of supported imu is:
`typedef enum {
           GYRO_NONE = 0,
            GYRO_DEFAULT,
            GYRO_MPU6050,
            GYRO_L3G4200D,
            GYRO_MPU3050,
            GYRO_L3GD20,
            GYRO_MPU6000,
            GYRO_MPU6500,
            GYRO_MPU9250,
            GYRO_ICM20601,
            GYRO_ICM20602,
            GYRO_ICM20608G,
            GYRO_ICM20649,
            GYRO_ICM20689,
            GYRO_ICM42605,
            GYRO_BMI160,
            GYRO_BMI270,
            GYRO_LSM6DSO,
            GYRO_FAKE
        } gyroHardware_e;`
Some of these imu are currently unsupported on targets with 512kflash.

Which of these chips are considered promising and will be supported on most firmware's?

and-sh, 2021-06-25:
It works without int. It doesn't want to work without cs. Maybe some peculiarities with initialization, while I left it that way. 

[image]

and-sh, 2021-07-05:
I understand correctly that iсm42605 is currently not supported?
[image]

and-sh, 2021-07-10:
Hooray! icm20602 work well.

closed 2021-07-10 by and-sh