betaflight/betaflight #10782: Controller Diagrams

pidtuner, 2021-06-04:
### Is your feature request related to a problem? Please describe.

I have been looking in the wiki and documentation for controller diagrams [(similar to the ones available for **PX4**)](https://docs.px4.io/master/en/flight_stack/controller_diagrams.html), but could not find any. 

I saw in [an old **issue**](https://github.com/betaflight/betaflight/issues/7112) that user @arquicanedo requested this in the past, but there was nothing at the time. He promised to post his own, but cannot find where he posted his results.

### Describe the solution you'd like

I cannot see how such a flight controller software can be maintained without control diagrams. Specially one so popular as BetaFlight. So could somebody point me to where I can find such information?

### Describe alternatives you've considered

If this information does not exist, could somebody point me out to resources where I can start researching about it?

Also please do not auto close this issue, I think this is a very important feature for this kind of software.

haslinghuis, 2021-06-04:
Please take a look at: https://github.com/betaflight/betaflight/wiki/Signal-Flow-Diagram

pidtuner, 2021-06-05:
Thanks, that one is higher level for control very useful indeed. 
Looking at the original source I found one that has the information I was looking for, even relating the config parameters:

* <https://github.com/mrRobot62/betaflight_processing/blob/master/doc/PID.4i2.pdf>

Very nice job from the author!
Thanks again for pointing out to it.

closed 2021-06-05 by pidtuner