betaflight/betaflight #10784: Blackbox GPS only

grostim, 2021-06-05:
### Is your feature request related to a problem? Please describe.

Hi!
Blackbox is a great feature. However the flash chip limited size makes it bearely usable when you want to log gps only of let's say all flights within a day.
For those interested in GPS logging only, it would be great to have a specific stting for the blackbox:
- blackbox timing limited to let's say 1Hz
- Very limited subset of data saved in each record. It could be limited to let's say: gps coordinates, speed, altitude and compass bearing.

I think this would allow to optimize the few Mb of a flash chip .
### Describe the solution you'd like

### Describe alternatives you've considered

McGiverGim, 2021-06-07:
What you want I think was included here: https://github.com/betaflight/betaflight/pull/9726 so is ready in nightlies.

Only CLI at this moment, search for `blackbox_disable` in CLI and you will find all the methods. At some moment it will be available in configurator too.

haslinghuis, 2021-07-08:
Solved in #9726

closed 2021-07-08 by haslinghuis