betaflight/betaflight #10786: Softserial usage (for telem) seems to disable my servo

dgrat, 2021-06-07:
### Describe the bug
I use a ultra micro servo for camera angle adjustment. 
I configured a softserial out for telemetry (which is still working). 
If I don't configure that softserial port as frsky smartport, the servo works.
As soon as I configure this port as smartport the servo, moves down (1000 PWM state) and cannot be controlled anymore.
I have the feel, that shouldn't be the case, because the servo is connected to Motor 6 (LED pin). 

### Flight controller configuration
Hobbywing XRotor Micro OMNIBUS F4 G2
```
# version
# Betaflight / STM32F405 (S405) 4.2.9 Apr 27 2021 / 19:33:01 (e097f4ab7) MSP API: 1.43
# config: manufacturer_id: AIRB, board_name: OMNIBUSF4SD, version: 3a35e73b, date: 2019-09-30T05:46:12Z

# start the command batch
batch start

board_name OMNIBUSF4SD
manufacturer_id AIRB

# resources
resource MOTOR 5 NONE
resource MOTOR 6 NONE
resource SERVO 1 A08
resource SERIAL_TX 1 NONE
resource SERIAL_TX 11 A09

# servo
servo 0 1000 1800 1500 100 8

# feature
feature SERVO_TILT
feature SOFTSERIAL
feature TELEMETRY

# beacon
beacon RX_LOST
beacon RX_SET

# serial
serial 0 64 115200 57600 0 115200
serial 2 1024 115200 57600 0 115200
serial 5 8192 115200 57600 0 115200
serial 30 32 115200 57600 0 115200
```
```
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: MOTOR 4
A03: MOTOR 3
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: SERVO 1
A09: FREE
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: OSD_CS
B00: MOTOR 1
B01: MOTOR 2
B02: FREE
B03: PREINIT 2
B04: BEEPER
B05: LED 1
B06: FREE
B07: SDCARD_DETECT
B08: FREE
B09: FREE
B10: FREE
B11: FREE
B12: SDCARD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_CURR
C02: ADC_BATT
C03: FREE
C04: GYRO_EXTI
C05: FREE
C06: SERIAL_TX 6
C07: FREE
C08: INVERTER 6
C09: INVERTER 3
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: FREE
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1:
    CH1: SERVO 1
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1: DSHOT_BITBANG 2
    CH3: DSHOT_BITBANG 1
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 2
DMA2 Stream 3: ADC 2
DMA2 Stream 4: DSHOT_BITBANG 1
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

```

haslinghuis, 2021-06-08:
Could you please try with `4.2` or even `4.1` to see if this is recently introduced?

dgrat, 2021-06-08:
4.2 and 4.1 show the same behavior.

dgrat, 2021-06-08:
https://www.rcgroups.com/forums/showthread.php?3092423-Hobbywing-XRotor-Micro-OMNIBUS-F4-G2-Flight-Controller-OSD
Seems to be a hardware issue?! If you use TX1 then the LED Pad stops working, but why?!

dgrat, 2021-06-08:
Also noticed that active softserial on tx6 disturbs uart3.

haslinghuis, 2021-06-08:
[image]
What's strange is that in the diff it's resourced as `SERIAL_TX 11 A09` while in resources it shows as:
```
A09: FREE
A10: SERIAL_RX 1
```

dgrat, 2021-06-09:
I wonder what can theoretically be the reason that always one of the other uarts starts failing if I assign a pin for softserial.

haslinghuis, 2021-06-09:
@mikeller ?

dgrat, 2021-06-11:
@haslinghuis 
Not very sure what you mean.

haslinghuis, 2021-06-11:
@dgrat Was expecting to see `A10: SERIAL_RX 1`

haslinghuis, 2021-06-11:
@dgrat Was expecting to see `A09: SERIAL_RX 1` as commanded with the resource assignment

jflyper, 2021-06-18:
@dgrat 

The original issue you have asked about is caused by a timer conflict:
PA8 (label PWM5 or PWM6) used for servo output shares timer TIM1 with PA9 (label TX1) used for software serial 1 TX. These functions works on different base frequencies, so they can not share the same timer (and the current BF is not made to share a timer between different functions).

[image]

You might have a chance using PWM5 (uses TIM5) for servo if TIM5 is not used for other purposes (which seems true atm).