betaflight/betaflight #10793: Running Linux on Chrome OS, and Configurator 10.7 Won't Recognize My Quad

PinapplePizzaaa, 2021-06-10:
### I am running betaflight 10.7 on my chromebook, using linux, but when I plug in my quad, betaflight doesn't recognize my quad.

### Install 10.7 on a chromebook, using linux, and plug in a quad

### betaflight will not recognize your quad as a port

### 
```
# diff all

# version
# Betaflight / STM32F7X2 (S7X2) 4.1.6 Apr 25 2020 / 05:12:26 (283bda8bf) MSP API: 1.42
# manufacturer_id: DIAT   board_name: MAMBAF722_I2C   custom defaults: YES

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name MAMBAF722_I2C
manufacturer_id DIAT
mcu_id 001f003b345650112030324b
signature 

# name: DIATONE ROMA

# feature
feature -TELEMETRY
feature -DYNAMIC_FILTER

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -RX_SET
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CRASH_FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL

# serial
serial 2 8192 115200 57600 0 115200
serial 4 33 115200 57600 0 115200

# led
led 0 14,12::C:1
led 1 15,12::C:1
led 2 14,13::C:1
led 3 15,13::C:1
led 4 14,14::C:1
led 5 15,14::C:1
led 6 14,15::C:1
led 7 15,15::C:1

# aux
aux 0 0 0 1600 2100 0 0
aux 1 1 1 900 1300 0 0
aux 2 13 2 1700 2100 0 0
aux 3 26 3 1700 2100 0 0
aux 4 35 1 1700 2100 0 0

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E CUSTOM  5705 5685 5665    0 5885 5905    0    0
vtxtable band 4 FATSHARK F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 IMD6     I CUSTOM  5732 5765 5828 5840 5866 5740    0    0
vtxtable powerlevels 5
vtxtable powervalues 25 100 200 400 600
vtxtable powerlabels 25 100 200 400 600

# master
set acc_calibration = -43,-18,-78
set mag_hardware = NONE
set baro_hardware = NONE
set min_check = 1010
set bat_capacity = 1300
set beeper_dshot_beacon_tone = 5
set small_angle = 180
set pid_process_denom = 2
set osd_units = IMPERIAL
set osd_rssi_alarm = 0
set osd_cap_alarm = 13892
set osd_alt_alarm = 8224
set osd_tim1 = 2561
set osd_tim2 = 2562
set osd_vbat_pos = 2408
set osd_rssi_pos = 2081
set osd_flymode_pos = 2104
set osd_crosshairs_pos = 14541
set osd_ah_pos = 2126
set osd_mah_drawn_pos = 2414
set osd_craft_name_pos = 2089
set osd_avg_cell_voltage_pos = 2401
set osd_stat_battery = ON
set osd_stat_total_time = ON
set vtx_band = 5
set vtx_channel = 8
set vtx_power = 4
set vtx_low_power_disarm = ON
set vtx_freq = 5917
set gyro_1_align_yaw = 1800
set name = DIATONE ROMA

profile 0

# profile 0
set p_pitch = 49
set d_pitch = 40
set p_roll = 47
set d_roll = 38
set p_yaw = 32

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
```

```
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: SERIAL_TX 4
A01: SERIAL_RX 4
A02: FREE
A03: FREE
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: MOTOR 3
A09: MOTOR 4
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FLASH_CS
B00: PINIO 1
B01: FREE
B02: BEEPER
B03: FREE
B04: FREE
B05: SPI_MOSI 3
B06: FREE
B07: SERIAL_RX 1
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: SERIAL_TX 3
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_BATT
C02: FREE
C03: ADC_CURR
C04: GYRO_EXTI
C05: FREE
C06: FREE
C07: FREE
C08: MOTOR 1
C09: MOTOR 2
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SERIAL_TX 5
C13: FREE
C14: LED 2
C15: LED 1
D00: FREE
D01: FREE
D02: SERIAL_RX 5
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1:
    CH1: MOTOR 3
    CH2: MOTOR 4
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH3: MOTOR 1
    CH4: MOTOR 2
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC
DMA2 Stream 1: TIMUP 8
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: TIMUP 1
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE```

### FC: Mamba APP F722 MK1
Camera: Runcam Phoenix 2
Antenna: Mamba Ultras
VTX: Mamba TX400 25-400mW VTX
ESC: Mamba F50 4in1
Motors: Mamba TOKA 2207.5 2450KV

haslinghuis, 2021-06-10:
This should be fixed in `10.8`, could check a nightly build for testing purposes only, as it's not a general released. Not sure it will work on Chrome OS. Could you be more specific and how did you get this diff without the recognition of your Quad?

PinapplePizzaaa, 2021-06-10:
I just used the chrome web app, but I'm trying to use the 10.7 tuning stuff

PinapplePizzaaa, 2021-06-10:
I don't think it is a problem with beta flight though, It won't recognize
any USB devices on the Linux portion, so thanks for the reply.

haslinghuis, 2021-06-10:
Chrome web app is not supported anymore, ignoring this not sure if this is your problem as I can't verify without the OS.

PinapplePizzaaa, 2021-06-10:
Yeah on second thought is is my problem lol, sorry about that. The chrome
app still works for me, it's just a bit outdated. I also installed it a
while ago, so maybe I have an older version or something.

bree29, 2021-06-11:
I don't know about Chrome OS, but there's that from the [betaflight-configurator README.md](https://github.com/betaflight/betaflight-configurator/blob/master/README.md) ; maybe you can try the [betaflight-configurator_10.7.0_linux64.zip](https://github.com/betaflight/betaflight-configurator/releases/tag/10.7.0) release.

> #### Linux users
> 
> In most Linux distributions your user won't have access to serial interfaces by default. To add this access right type the following command in a terminal, log out your user and log in again:
> 
> ```
> sudo usermod -aG dialout ${USER}
> ```

closed 2021-07-19 by github-actions[bot]