betaflight/betaflight #10795: Enabling LED breaks UART connection to betaflight

wterreb, 2021-06-13:
### Describe the bug
If you use a serial UART to connect to Betaflight, it stops working as soon as you enable LED strip

### To Reproduce
My micro USB connector is damaged.  So I enabled UART with an FTDI board to connect to Betaflight.
But as soon as you enable LED strip in Betaflight configuration, then you can no longer connect via UART1
Only way to recover with solder flimsy wires to the micro USB and use that to reconnect and disable LED strip agian.
As soon as you disable the the LED strip then you can connect to Betaflight through UART1 once more.
This is very repeatable.

### Expected behavior
Enabling LED strip should not break the connection via UART1

### Flight controller configuration

```
PASTE THE OUTPUT OF 'diff' HERE
# diff

# version
# Betaflight / STM32F7X2 (S7X2) 4.2.9 Apr 27 2021 / 19:34:29 (e097f4ab7) MSP API: 1.43
# config: manufacturer_id: JHEF, board_name: JHEF7DUAL, version: 6fac4231, date: 2019-10-18T22:07:46Z

# start the command batch
batch start

board_name JHEF7DUAL
manufacturer_id JHEF

# name: 7_INCHER

# feature
feature -RX_PARALLEL_PWM
feature -AIRMODE
feature RX_SERIAL
feature GPS

# beeper
beeper -RX_LOST

# beacon
beacon RX_LOST

# serial
serial 0 1 115200 57600 0 115200
serial 2 2048 115200 57600 0 115200
serial 3 64 115200 57600 0 115200
serial 5 2 115200 9600 0 115200

# aux
aux 0 0 0 1875 2100 0 0
aux 1 1 1 900 1100 0 0
aux 2 2 1 1350 1600 0 0
aux 3 46 2 1875 2100 0 0
aux 4 46 3 1300 2100 0 0
aux 5 13 2 1875 2100 0 0
aux 6 13 3 1300 2100 0 0
aux 7 26 0 1900 2100 0 0
aux 8 28 1 1350 2100 0 0
aux 9 41 4 1300 2100 0 0

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665    0 5885 5905    0    0
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 IMD6     I CUSTOM  5732 5765 5828 5840 5866 5740    0    0
vtxtable powerlevels 4
vtxtable powervalues 0 1 2 4
vtxtable powerlabels 25 200 500 800

# master
set acc_calibration = 11,-25,45,1
set serialrx_provider = CRSF
set motor_pwm_protocol = DSHOT600
set failsafe_procedure = GPS-RESCUE
set gps_rescue_allow_arming_without_fix = ON
set osd_tim1 = 1281
set osd_tim2 = 1794
set osd_vbat_pos = 2071
set osd_link_quality_pos = 2049
set osd_rssi_dbm_pos = 2251
set osd_tim_1_pos = 2519
set osd_tim_2_pos = 439
set osd_flymode_pos = 2512
set osd_vtx_channel_pos = 2465
set osd_craft_name_pos = 2503
set osd_gps_lon_pos = 2529
set osd_gps_lat_pos = 2543
set osd_gps_sats_pos = 2497
set osd_home_dir_pos = 2285
set osd_home_dist_pos = 2316
set osd_flight_dist_pos = 2487
set osd_warnings_pos = 2377
set osd_avg_cell_voltage_pos = 2064
set osd_stat_total_time = ON
set vtx_band = 5
set vtx_channel = 1
set vtx_power = 4
set vtx_freq = 5658
set gyro_1_align_yaw = 900
set gyro_2_align_yaw = 900
set name = 7_INCHER

profile 0

rateprofile 0

# rateprofile 0
set roll_rc_rate = 150
set pitch_rc_rate = 150
set roll_expo = 20
set pitch_expo = 20

# end the command batch
batch end

PASTE THE OUTPUT OF 'resource show all' HERE
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: SERIAL_TX 4
A01: SERIAL_RX 4
A02: FREE
A03: FREE
A04: GYRO_CS 2
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: LED 1
B00: MOTOR 1
B01: MOTOR 2
B02: GYRO_CS 1
B03: MOTOR 4
B04: MOTOR 3
B05: SPI_MOSI 3
B06: I2C_SCL 1
B07: I2C_SDA 1
B08: CAMERA_CONTROL
B09: PINIO 2
B10: SERIAL_TX 3
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_CURR
C02: ADC_BATT
C03: FREE
C04: GYRO_EXTI
C05: FREE
C06: SERIAL_TX 6
C07: SERIAL_RX 6
C08: FREE
C09: FREE
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: FREE
C13: FLASH_CS
C14: PINIO 1
C15: BEEPER
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2:
    CH2: MOTOR 4
TIM3:
    CH1: MOTOR 3
    CH3: MOTOR 1
    CH4: MOTOR 2
TIM4:
    CH3: CAMERA_CONTROL
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: TIMUP 3
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: TIMUP 2
DMA2 Stream 0: ADC
DMA2 Stream 1: FREE

### Setup / Versions
 - Flight controller: 
# Betaflight / STM32F7X2 (S7X2) 4.2.9 Apr 27 2021 / 19:34:29 (e097f4ab7) MSP API: 1.43
# config: manufacturer_id: JHEF, board_name: JHEF7DUAL, version: 6fac4231, date: 2019-10-18T22:07:46Z
# board: manufacturer_id: JHEF, board_name: JHEF7DUAL

 - Other components:
GPS, Buzzer, VTX, ExpressLRS receiver

 - How are the different components wired up:
Nothing unusual

jflyper, 2021-06-18:
Hi. It seems that the supplied configuration does not contain any thing related with LED strip.
Can we tell us some more on how you "enable LED strip" ?

wterreb, 2021-06-20:
OH I just go into Betaflight on other features and enable the slide called "LED_STRIP"

[image]

closed 2021-07-28 by github-actions[bot]