betaflight/betaflight #10796: Camera control does not work as it should on target ZEEZF7V2

robustini, 2021-06-14:
### Describe the bug
Hi, i've this issue only with ZEEF7V2 target, same fpv cam and other F7 boards and I have no problem.
PB15 is the default pin used in Betaflight, I enter the cam menus correctly, but the "drop value" command does not work, and others are reversed from how it should work.
With other targets, even custom, everything works perfectly, so there is something anomalous with this fc with this Foxeer Predator.
I use the camera control with many other F7 fc and I have no problem, with the default settings.
So please test it and post these values in the correct arrangement, because by default not all work and some are reversed.

### To Reproduce
Just try with the default settings, even changing them the problem remains. 

### Expected behavior
Even changing them the problem remains.

### Flight controller configuration

```
PASTE THE OUTPUT OF 'diff' HERE
```
```
PASTE THE OUTPUT OF 'resource show all' HERE

# version
# Betaflight / STM32F7X2 (S7X2) 4.2.9 Jun  4 2021 / 17:33:47 (norevision) MSP API: 1.43

# start the command batch
batch start

board_name ZEEZF7V2
manufacturer_id ZEEZ

# name: Edo

# resources
resource BEEPER 1 B02
resource MOTOR 1 B07
resource MOTOR 2 B06
resource MOTOR 3 B04
resource MOTOR 4 B03
resource MOTOR 5 C08
resource MOTOR 6 C07
resource MOTOR 7 C06
resource MOTOR 8 B01
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 NONE
resource PWM 1 NONE
resource PWM 2 NONE
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource SONAR_TRIGGER 1 NONE
resource SONAR_ECHO 1 NONE
resource LED_STRIP 1 B00
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 A00
resource SERIAL_TX 5 C12
resource SERIAL_TX 6 NONE
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
[... 1053 lines cut ...]
set yaw_rc_rate = 100
set roll_expo = 0
set pitch_expo = 0
set yaw_expo = 0
set roll_srate = 70
set pitch_srate = 70
set yaw_srate = 70
set tpa_rate = 65
set tpa_breakpoint = 1350
set tpa_mode = D
set throttle_limit_type = OFF
set throttle_limit_percent = 100
set roll_rate_limit = 1998
set pitch_rate_limit = 1998
set yaw_rate_limit = 1998

### Setup / Versions
 <!-- Specify your flight controller model (what type is it, where was it bought from, ...) -->
 - Flight controller: ZEEZF7V2
 <!-- Specify other components attached to the flight controller (RX, VTX, brand / model for all of them, firmware version where applicable...) -->
 - Other components: rx Crossfire, VTX 600 mW, ESCs 40A
 <!-- Details about how all is wired -->
 - How are the different components wired up: standard wiring, the camera OSD is connected to the PB15 of the FC via a 220 ohm resistor. 

<!-- Add any other context about the problem that you think might be relevant here. -->

haslinghuis, 2021-07-15:
Please provide the `diff` and `resource show all` as request while opening the issue.

After that maybe @Guidus93 could have a look at it.

robustini, 2021-07-16:
> 
> 
> Please remove resource output and provide the `diff` and `resource show all` as requested while opening the issue and edit in the opening message. I've now edited to show how to use the backticks
> Do you have the 20x20 or 30x30 board?
> 
> After that maybe @Guidus93 could have a look at it.

I had already chatted with Guido and he told me that with RunCam he didn't encounter this problem.
Yet this unfortunately happens here with a Foxeer Predator.

robustini, 2021-07-16:
@haslinghuis I entered the requested information, sorry for the layout but it is not easy to understand how your layout designer thinks.

haslinghuis, 2021-07-17:
@robustini It's default github markup, we didn't anything about layout. Updated your post. Perhaps Foxeer doesn't need a resister (or another value to be used) like the Runcam camera does but this is just speculation.

robustini, 2021-07-17:
> 
> 
> @robustini It's default github markup, we didn't anything about layout. Updated your post. Perhaps Foxeer doesn't need a resister (or another value to be used) like the Runcam camera does but this is just speculation.

Tested both with and without resistor, the result does not change.
Installed another F7, so with the same target (STM32F7X2) and zero problems, same parameters.

haslinghuis, 2021-07-17:
Should camera control not have a dma pin assignment in the config file @Guidus93?
```
dma pin B15 0
```

robustini, 2021-07-18:
One thing I have not tried, remap the resource of the leds and use that, only as a comparison.
I would not like the pin dedicated to the camera control of this board to have circuit problems.
But the problem is not that it doesn't work, it works but the selection with the sticks is wrong, as I wrote above, as if the five resistance values that can be changed are not correct for this board.
We tried to change them and we managed to get four controls to work, but the fifth one remains not working.

haslinghuis, 2023-10-27:
[Should be] fixed in https://github.com/betaflight/unified-targets/pull/946

closed 2023-10-27 by haslinghuis