betaflight/betaflight #10797: OFFSETS FROM memory

KBS91WAW, 2021-06-14:
### Is your feature request related to a problem? Please describe.
Drift yaw problem. Flight controller autocalibrates gyro and acc offsets every power on.
Some misalignment causes yaw to drift slowely. 
If we give button option in Calibration program to disable autocalibration on power-on
then we can keep calibration values that were found on testing bench on good ( not shaking) ground.

### Describe the solution you'd like
Option on Configuration GUI that will block autocalibration at power on and takes offsets ( zero values) from
memory.
betaflight-4.2.9\betaflight-4.2.9\src\main\fc\init.c 
functions to block:
accStartCalibration();
gyroStartCalibration(false);

### Describe alternatives you've considered
Make my own code to do it.

haslinghuis, 2021-06-14:
`gyro_cal_on_first_arm = OFF`

KBS91WAW, 2021-06-15:
Dear Betaflight,

Is there any way to avoid YAW incrementing slowly?
I see that value increasing in longer period of time.

Regards,
Chris

haslinghuis, 2021-06-15:
Duplicate of https://github.com/betaflight/betaflight/issues/9052 or https://github.com/betaflight/betaflight/issues/10557

KBS91WAW, 2021-06-15:
Is it possible to block acc calibration at start?

I still observe drifting when controller power-on in movement.

Regards,
Chris

haslinghuis, 2021-06-15:
Some things you can look at:
- Fix noise and vibration issues with blackbox logging
- Add some yaw deadband to deal with jitter
- Calibrate your radio
- Replace FC

Please read the links in the comment above.

KBS91WAW, 2021-06-16:
Dear Sirs,

I managed to make Yaw increment slowly.   I will keep like that and wont
change for now.

Last my question is if it is possible to decrease power consumption or
disable motor control algorithm part.

Simply I use Flight controller for precise angle estimation in different
device.  I do not need other functionalities.
I only read yaw, pitch, roll by MSP protocol.

Regards,
Chris

haslinghuis, 2021-06-16:
Please take a look at:

Mixer = `GIMBAL` (or `CUSTOM`)
```
set motor_pwm_protocol = DISABLED
```

closed 2021-07-24 by github-actions[bot]