betaflight/betaflight #10802: Enable crossfire/tracer MyVTX SmartAudio functions in conjunction with BF

JekleFPV, 2021-06-17:
### Is your feature request related to a problem? Please describe.
No, not really a problem. Apart from occasionally stomping on someone else's video.
The essence of the functionality I'm looking to add is to set up the band/channel/power before powering up my quad so I don't stomp on someone else's video when I'm flying with others.
TBS' MyVTX would enable that, but bypassing the FC would take away some other features that would still like to have, such as band/channel/power on the OSD, putting output power on a switch and/or low power when unarmed.

### Describe the solution you'd like
I guess the easiest way to implement a solution is the ability to route the SmartAudio signal from a crossfire/tracer RX (on SA mode, not MyVTX mode of course) to a UART on the FC, setting the FC VTX band/channel/power (taking into account the low power when disarmed), which the FC SA port then sends to the VTX.
That should be possible right? The only issue I can think of is that the switching from RX to FC to VTX on powerup might take just a little too long to completely prevent stomping on somebody else's video.

### Describe alternatives you've considered
From what Trappy told me, hooking both the FC and crossfire/tracer RX up to the VTXs' SmartAudio port pretty much disregards the MyVTX setting because the FC will continuously send the SA-vtx settings. Not to mention losing the previously mentioned features.
Sorry to be greedy, but I want it all :P

lyndsysimon, 2021-07-29:
I just finished installed a TBS SIXTY9 (combo vtx/rx), and was surprised to discover that Betaflight can’t see teh vtx portion, even though the unit has a bidirectional UART port.

If I understand the OP, what they’re asking for is effectively the same use case as the TBS SIXTY9.

haslinghuis, 2021-07-30:
@lyndsysimon Wow. Very sleek (new) product from TBS.

@JekleFPV like @tbs-trappy said you can't get both as it would fight each other. Using TBS Agent-Lite LUA should be sufficient.

k4cus, 2021-08-25:
But it is not sufficient. And TBS didn't mention that I will loose smartaudio capability when change from cheap VTX to sixty9!! In manual there is only small * that smartaudio is not available on sixty9.  I feel tricked by TBS. They should include smartaudio option in next firmware. I don't want to use myvtx, because I can't assign radio buttons to switch vtx power in flight. And this is feature that I want, as I am saving few wats by changing power of vtx when required.

tbs-fpv, 2021-08-25:
chillax @k4cus, Install v6.09 and get SmartAudio capability.

k4cus, 2021-08-25:
I have got 6.09 and can't find smartaudio option. Where is it exactly?

tbs-fpv, 2021-08-25:
You can configure port 1 or port 4 to act as a smart audio input.

k4cus, 2021-08-25:
OK i have 6.07 but my receiver updated yesterday so I don't know why it is in older version. Will need to check. Thanks for a tip. It should be written in the manual. I find tbs crossfire manual pdf poor in knowledge. Options are not explained fully. I need to watch youtube movies which takes hours of my time instead reading in manual.

tbs-fpv, 2021-08-25:
@k4cus 6.09 was released a few days ago and it's still in BETA version. Manual only documents PUBLIC version of the firmware. Also, just for the record, we didn't "cheat" anyone, we just couldn't imagine why anyone would want SmartAudio in the Sixty9. When customers requested it, we added it.

k4cus, 2021-08-25:
So now You can imagine my use case. If there was option to bind radio switch to change myvtx settings instead of jumping over menus I wouldn't need smartaudio at all. I would use myvtx. I gues you didn't do it on purpose however not mentioning that your flagship vtx doesn't support smartaudio is tricky.

k4cus, 2021-08-25:
Ok I have last question. I confirm that on 6.09 beta firmware smartaudio option has appeared. However port 1 and port 4 are crsf tx both. On page 12 in manual pdf there is a wiring diagram. Upper picture Out1 and Out2 are RX TX that are connected to FC. Which wire on the plug is smartaudio? I have connected audio port to FC and It would be best to set smaraudio on that pin, as this is common configuration.

k4cus, 2021-08-25:
I need to ask, because manual lacks information how Ouput 1-6 on my crossfire configuration corresponds to outputs on the plug (page 12 in manual pdf - Sixty9 (FW 6.04 or later required))

tbs-fpv, 2021-08-25:
SmartAudio Port 4 is not on the main plug, it's a solder pad on the bottom. Manual, page 13.

k4cus, 2021-08-25:
Ok thanks. Well that sucks, I just finished my quad build yesterday and now I need to disassembly it again.
There is no software option in opentx to change myvtx settings via switch inflight?
I think this post will help many people to avoid my 2 day struggle.
Thank you for your quick help, have a nice day :)

tbs-fpv, 2021-08-25:
We have pitmode setting on switch. All other settings can only be manipulated through the LUA scripts, Agent X or Agent M mobile app.

k4cus, 2021-08-25:
I will read about it, if you have some valuable resources I would appreciate links. Maybe it is possible to attache some script or function in opentx to make it switch power. However if I could have pitmode on radio switch, it would be satisfactory.

tbs-fpv, 2021-08-25:
6.09 has "Video Transmitter" menu in your Crossfire / Tracer transmitter settings. There is a Pitmode setting which you have to set to "Switch". The remainder should be self-explanatory. We are kind of hijacking a feature request here ...

k4cus, 2021-08-25:
I set VTX control to TX and then:
Pitmodectrl Switch
Pitmode SW channel, I must set extra channel. I have all 8 channel occupied so I need to set at channel 9? Do I need to change number of crossfire channels from 8 to 12 then?

As I am reading about lua, I can see what I need is function lua script that will change VTX output power on switch change. Maybe someone has written such a script already?

tbs-fpv, 2021-08-25:
You don't have to change the number of channels transmitted over the air to access additional switches. you can also allocate it to e.g. the arm switch.

We currently don't have VTx power change on a switch. That is low on priority as it would simplify cheating at FPV races.

k4cus, 2021-08-25:
I just went to fly now and sixty9 is really cool thing. At 100 mW VTX I have got 1,3 km range. There is one downside however, since I don't care about radio range and my picture quality is much better, than using old VTX which was overheating, I am flying faster and further, so my flight time is reduced by 1 minute and I keep forgetting about battery capacity :)
I think if someone wants to cheat he/she will solder smartaudio cable and will do it anyway. Function script to change channel and txpower would be super cool thing. Then I might forget about smart audio from FC.
Actually pitmode works very well over CRSF, it simply work. With smartaudio I had a trouble that sometimes it did turn off and sometimes it didn't. Now it always work, it can by assigned to arm channel or mixed, that either you arm or turn switch and pitmode is off. 
At this point I understand why you assumed people will forget smartaudio (it simply works and you manage all models in one place in transmitter), but you didn't prepare proper manuals howto do thing your way, so naturally everyone will go for well known smartaudio solution.

k4cus, 2021-09-01:
Ok it seams that what You say is smartaudio output to control vtx. What I want is to control sixty9 vtx over smartaudio from flight controller. Is it somehow possible?

tbs-fpv, 2021-09-01:
Yes. As stated before, wire up a cable between FC's smart audio port and Ch4, then set the Sixty9 output port on ch4 to Smart Audio.

k4cus, 2021-09-01:
OK, but is it smartaudio output or input? I got confused, because on other CF receivers this is smartaudio output. On this one this will be input?

closed 2021-10-09 by github-actions[bot]