betaflight/betaflight #10803: Change `stats_total_flights` couting method

bree29, 2021-06-18:
Hi !

### Is your feature request related to a problem? Please describe.
Today, you have one pack and you're going to cruise with it. 2 minutes, and you crash, you disarm, you arm again and you're going to finish your flight. What happens is you have +1 on stats_total_flights the time you crashed, and +1 when you will land at the end of the pack. However, that was just one, interrupted but only one, flight.

### Describe the solution you'd like
I'd love to be able to do it myself, but I'm only able to do some "code guessing". Correct me if I'm wrong, but this behavior is expected as [the code says for itself line 85](https://github.com/betaflight/betaflight/blob/025ee87a7aca068e3659fd066b8a9afbed123361/src/main/fc/stats.c) in ` betaflight/src/main/fc/stats.c`.
```
void statsOnDisarm(void)
{
    int8_t minArmedTimeS = statsConfig()->stats_min_armed_time_s;
    if (minArmedTimeS >= 0) {
        uint32_t dtS = (millis() - arm_millis) / 1000;
        if (dtS >= (uint8_t)minArmedTimeS) {
            statsConfigMutable()->stats_total_flights += 1;    // arm / flight counter
            statsConfigMutable()->stats_total_time_s += dtS;
            statsConfigMutable()->stats_total_dist_m += (DISTANCE_FLOWN_CM - arm_distance_cm) / 100;

            saveRequired = true;
        }

        if (saveRequired) {

    {.......<=

    }
}
```
A solution to avoid this "+1" count on stats_total_flights on each disarm that satisfies `minArmedTimeS`, defined by `statsConfig()->stats_min_armed_time_s`, it would be possible to implement a bolean var that would be "0" when the FC boot the first time (or reboot), and be set to "1" after you increment the flight count. Then, if you arm again and disarm again few minutes later, it will check that 0/1 variable status before to increment the flight count.

### Describe alternatives you've considered
Alternative would be to add yet another variable `stats_total_cycles` or `stats_total_packs` but it would uselessly populates the CLI variables, on my opinion.

rickmanship, 2021-06-18:
Why do you think your definition of 'total flights' is more correct then the way it currently works? You seem to be more interested in 'total packs' which is something different then the current function. You wanting something different doesn't mean that the current function isn't correct and useful to people. Or alternately you could not crash...

bree29, 2021-06-18:
> Why do you think your definition of 'total flights' is more correct then the way it currently works? You seem to be more interested in 'total packs' which is something different then the current [...]

Hello, I'm sorry if I seemed to be say the current function is somewhat dysfunctional/useless ; it wasn't what I wanted to transmit (sorry, not an english guy). I thought I have been explicit enought in my feature request to say I do think the current behavior is the expected behavior. I merely stated another way to load it as a feature request (and not _bug report_).

After that, that's on the dev team to see if/how/when the feature is to be considered. Sorry if I seemed rude in my feature request.

And yeah, I first wanted to say in "alternatives" "do not crash, do always record DVR 1 time and not multiples, or not not at all ;)

haslinghuis, 2021-06-19:
Please change the title like `[FR] Add stats_total_packs/cycles` for requesting addition of this new case to the code instead of replacing field tested code behavior.

closed 2026-09-26 by blckmn