betaflight/betaflight #10809: Bessel low pass filter

LSchwerdt, 2021-06-21:
tldr: The second order Bessel low pass filter might be a good alternative to the PT3, PT2 and Butterworth (Biquad) filter.

Inspired by the recent discussions about the addition of PT2 and PT3 low pass filters, I searched for other filter types, that might be better in some sense. Sadly, there is no free lunch and improvments in one area lead to worse performance in another. Nonetheless, the Bessel filter seems to be a good compromise. The Bessel filter is analog by nature, so I used the bilinear transformation of the analog version.

The desing criterion for the Bessel filter is a constant delay in the pass band. It has a Q factor of 1/sqrt(3) ≈ 0.577 and is therefore between the PT2 and Butterworth filters in all aspects. (Q PT2: 0.5, Q Butterworth: 1/sqrt(2) ≈ 0.707)

### Plots
The plots to compare the different filters are generated usingthis Matlab script:
[filterComparisonMatlab.zip](https://github.com/betaflight/betaflight/files/6681961/filterComparisonMatlab.zip)

[image]
[image]
[image]
[image]

### Discussion
Comparing the filters, you can rank them according to multiple figures of merit:

Criterion | Ranking: best to worst
--- | ---
Amplitudes near cutoff (steep slope) | Bu2 > **Be2** > PT3 > PT2 > PT1
Phase lag (delay) | PT1 < PT2 < **Be2** < Bu2 < PT3
Amplitudes for f-> inf | PT3 < Bu2 = **Be2** = PT2 < PT1
Step response overshoot | PT1 = PT2 = PT3 < **Be2** < Bu2

If the very small amount of overshoot can be tolerated, the Bessel filter is superior to the PT3. Due to the small amount of overshoot it might be less beneficial for the setpoint filtering and more useful for the gyro and d-term, but I think this needs to be tested for a definitive conclusion. Specifically, how much overshoot is tolerable for 1. Setpoint filtering, dynamic filters and static filters? I think the reasoning "Butterworth has too much overshoot and rining -> lets go to the extreme to avoid these" might go too far and there might be different optimal tradeoffs for the different applications.

### Implementation
Since the Bessel filter is a regular second order filter, i should be possible to implement it using the biquad code with other coefficients. Maybe it would even make sense to have just a general biquad filter with a user-selectable Q.

The main question is, wether the additional implementation effort and having "too many" options for the user is woth it, considering we already have the PT2 and Butterworth filters.

KarateBrot, 2021-06-21:
Sounds interesting :) I googled the biquad coefficients for the Bessel LPF but it seems like there is no analytic solution. If you could provide us with the numerical solution for the coefficients, implementing and testing this is pretty straight forward.

LSchwerdt, 2021-06-21:
I just wanted to make sure there is at least some interest before I take the time to find/calculate the coefficients. But it seems we have to thank the original author(s) of the biquad filter, because the calculation is already in the code:
The function `biquadFilterUpdate` (line 180) in https://github.com/betaflight/betaflight/blob/master/src/main/common/filter.c accepts a Q value and calculates the coefficients. The only thing left to do is to change the Quality factor (defined in line 32).

KarateBrot, 2021-06-21:
So we can just use the coefficients of the Butterworth LPF? Just want to make sure because the Q factor is the least of problems. If it has the same coefficients, the implementation is merely 3 lines of code.

ctzsnooze, 2021-06-22:
The 'special' thing about the PT3 is the slow initial rise in amplitude.  For RC smoothing that's particularly important. By not responding quickly to a step increase, it really 'straightens out' the sharp steps up.  And at higher values gives a lovely smoothness to otherwise abrupt inputs.  I would not see any value in any other filter topology in that application.
For general gyro or D filtering, overshoot is very undesirable.  
We currently use series PT1's on gyro and D, with one of the pair being dynamically shifted to higher cutoffs as throttle increases.  For gyro this shifting was historically intended, in the pre-RPM filter, to 'shape' the input noise so that the dynamic notch would focus on the fundamental motor peak.  Probably it serves no purpose anymore.  
For D, optimising delay is the main thing, and we need relatively strong filtering on arming, so having a PT1 that starts low and is dynamically lifted is useful.  We then shift from first order to second order as we move up in throttle, to deal with the extra noise higher up.  We can adjust the two cutoffs with some precision.  Computationally two PT1's is not very difficult, even if one is dynamic, and it works well.
A biquad second order filter has been historically used to filter gyro for Dmin, that's the only place currently where that kind of filter is still used, and I have a PR to convert that to PT2.  After that, we won't be using the biquad filter anywhere in the code.
Whether it is helpful to change our gyro or D filtering to a single second order filter, rather than two series first order filters, especially on D, hasn't been tested.  
A clean pass-band for D was the primary reason for using the biquad but it just didn't work out well at all.  
If this filter did provide a clean pass-band without ringing or exaggerations, when in the final coded form, it could be useful.
I think we'd have to model the actual code implementation, step for step, in order to assess whether there was a real benefi.  We'd also have to look at the code complexity, especially if we were to recalculate the coefficients in a dynamic setting.

LSchwerdt, 2021-06-22:
@KarateBrot: Yes. The code is for the general form of a biquad filter, and should work for any Q. And it wont add any computations compared to the current Butterworth biquad.

@ctzsnooze: Thank you for the in depth explanation of the reasoning behind the current default filter setup.
And you are right, for the setpoint filtering (RC smoothing) a PT3 absolutely makes sense. Stepped inputs are expected and the PT3 smoothes them to yield a continuous second derivative - the analog version at least.

For the default gyro/D filtering a PT2 is a good choice. But maybe for people who deactivate one or both gyro PT1s, a D term filter with a steeper cutoff is worth the extra delay. Maybe the PT2 is (close enough to) the best choice for everyone. But I have not seen the possibility of having Q values between 0.5 and 0.707 mentioned in the discussions, so I wanted to bring it up.

Why is having overshoot of the step response such a problem for the gyro/D term filter? I thought if the overshoot is small enough it should hide in the noisy signal. I would guess the main problem of the Butterworth filter might be the larger phase lag near the cutoff, which could lead to a smaller phase margin and less damping of these frequencies by the closed loop. But that is hard to assess without building a model.

LSchwerdt, 2021-06-22:
I did some more digging and according to [Low-Pass Risetime Filters for Time Domain Applications](https://kh6htv.files.wordpress.com/2015/11/an-07a-risetime-filters.pdf) the gaussian filter should be the ideal filter for our application. But beacuse it is non causal, one can only aproximate it.

On one hand, the Bessel filter is given as an example of an iir approximation to the gaussian filter. But on the other hand, the gaussian filter is the fastest filter without any overshoot, which matches the PT2. So I guess the optimum must be somewhere in the range between (and including) the PT2 and Bessel filter.

Quick-Flash, 2021-07-13:
The L-Optimum filter is very interesting to me. https://www.mathscinotes.com/2011/06/the-papoulis-filter-aka-optimum-l-filter/
[image]
I'd love to help get something like this implemented in code.

LSchwerdt, 2021-07-14:
Although the frequency domain characteristics look good, I suspect the L-filter might have too much delay, given that the Butterworth filter was abandoned as the default in favor of filters with worse frequency domain performance but lower delay.
Can you provide any information on the phase response or group delay of the L-filter? 

Going by the the tables in your link, the second order L-filter is the same as the second order [Butterworth](https://en.wikipedia.org/wiki/Butterworth_filter#Normalized_Butterworth_polynomials). Am I mistaken, or do you think we need higher order filters?

Quick-Flash, 2021-07-15:
> Although the frequency domain characteristics look good, I suspect the L-filter might have too much delay, given that the Butterworth filter was abandoned as the default in favor of filters with [...]

I'm no expert on the L-Filter and there sadly isn't a ton of info out on it. I do think however that the filter may have value. 

One thing that could be possible with higher order filters is the ability to raise the cutoff and diminish the effects of the extra delay that they provide. Eleptical filters also look interesting besides the fact that they have ripple. Which this L filter provides similar filtering without the ripples. I'd say it's worth testing.

KarateBrot, 2021-07-23:
This is what Wikipedia says about digital Bessel filters:

> As the important characteristic of a Bessel filter is its maximally-flat group delay, and not the amplitude response, it is inappropriate to use the bilinear transform to convert the analog Bessel filter into a digital form (since this preserves the amplitude response but not the group delay).
>
>The digital equivalent is the Thiran filter, also an all-pole low-pass filter with maximally-flat group delay,[12][13] which can also be transformed into an allpass filter, to implement fractional delays.[14][15] 

Does that mean changing the Q value in the biquad update function doesn't exactly represent a Bessel filter in discrete time?

bw1129, 2021-07-28:
Here's a little real world data from 2 different quads. Both compared on Dterm prefiltered (dotted traces) vs dterm filtered (solid traces. Bessel is bright red and PT2-3 brownish (PT2 and 3 comparisons with corrected cutoffs). Bessel showed more attenuation than PT2 but about the same or slightly less than a PT3. 

[image]
[image]

LSchwerdt, 2021-07-28:
> @LSchwerdt This is what [Wikipedia](https://en.wikipedia.org/wiki/Bessel_filter#Digital) says about digital Bessel filters:
> [...]

As far as I understand it, the bilinear transformation does not accurately preserve the group delay, wich is the design criterion of the Bessel filter. But according to [this post on dsp.stackexchange](https://dsp.stackexchange.com/questions/8319/simulating-analog-filter-using-convolution-or-converting-with-fft/8420#8420) the errors are small when the cutoff frequency is small compared to the sampling frequency. We probably use mostly cutoffs less than 200Hz at sampling frequencies of 4kHz or 8kHz, so we should not see any problems for this application. But to be save, the same checks should probably applied that are currently used for the butterworth filter to prevent using very high cutoff frequencies. And we dont know what filter is the best (for a particular craft + tune), so maybe the slight difference to the ideal Bessel filter might actually be be good. ;)

@bw1129 Thank you for testing the filter. The results look promising. Especially when comparing the Bessel filter to the PT3. To be fair to the PT2 though, it has less delay than the Bessel filter so it is "allowed" to have less attenuation. I think the hard question is, if switching to a Bessel filter is better than lowering the frequency of a PT2, as both options increase the amount of filtering while introducing more delay.

SupaflyFPV, 2021-07-31:
@LSchwerdt I think if you reduce cutoff on the PT2 to match the overall attenuation of the Bessel, then I think you may have more phase delay in the passband key frequencies which are involved in PID control...(if I understand this correctly)?

I tried a comparison here where @KarateBrot  told me the equivalent freq was 85hz PT2 to Bessel 120Hz, and the PT2 example felt on the sticks a lot more 'filtered' which I suspect that feeling is a result of phase shift in frequencies affecting the pid controller.
[Logs.zip](https://github.com/betaflight/betaflight/files/6911463/Logs.zip)

closed 2026-09-26 by blckmn