betaflight/betaflight #10815: Dynamic output limit.

levelfifty, 2021-06-29:
### Is your feature request related to a problem? Please describe.
Output limit is currently static.   If using 6S on 4S motors with a 66% output limit, at 4.2vpc the motors see an effective 16.8V, which is intended.  But at 3.7vpc the motors see an effective voltage of 14.8, when up to 22.2V is available.   This increases the current requirement to maintain the same power to the motor, increasing losses and lowering efficiency.  

### Describe the solution you'd like
Dynamic output limit.   Add a field to either input your desired max voltage (16.8v, etc) to the motors or desired battery class. (4s,3s, etc).   Referencing the already available voltage data, apply an output limit that maintains the desired output.   Example:  if max voltage selected is 16.8V, a 6S battery at full 25.2V would start with an output limit of 66%.   At 22.2V, the output limit would increase to 76%.   This will leave the effective voltage the motors see at 16.8v throughout the lipo’s life so the motors perform with consistency and maximum efficiency the entire time.   Higher voltage at equal power will always result in lower current, and current is where we get resistive losses and excess heat.   

### Describe alternatives you've considered
Alternatives are to use properly matched motors and voltages or the static output limit, but the potential effect of a dynamic output limit may be very beneficial for those wanting a more consistent experience throughout the Lipos life.

Would also suggest some level of averaging when increasing the limit.  Or only capturing voltage data when throttle is below a set threshold for the sake of applying the dynamic output limit to get a more realistic reading.

LSchwerdt, 2021-07-05:
In what way does your feature request differ from the dynamic battery sag compensation added in 4.2?
https://github.com/betaflight/betaflight/pull/9561
https://github.com/betaflight/betaflight/wiki/4.2-Tuning-Notes#Dynamic-battery-sag-compensation

When combining the battery sag compensation with a static output limit you should be able to achieve your desired behaviour, although the parametrization is more complicated than your request. And it is not 100% clear to me if the additional reduction is additive or multiplicative.

For your example, battery sag compensation produces a limit equivalent to 3.5Vpc (at default), so to get to 16.8V from 6s you would need an additional static motor limit of 16.8V/(6*3.5V) = 80% (asssuming a multiplicative combination of the output limits).

haslinghuis, 2021-07-09:
Duplicate of #9561

closed 2021-07-09 by haslinghuis