betaflight/betaflight #10819: ISSUE - "Invalid Mixer Mode Selected"

SpeedyPizzaDrones, 2021-07-02:
### Describe the bug
After switching on the Bidirectional Dshot and rebooting, the error in the attached picture appears when entering again in the "motor page". Weird that the error message talks about motor mix but it's left by default Quad X. Message appear after turning on "Bidir. Dshot" only
[image]

### To Reproduce
BETAFPVF411RX - switch on Bidir. Dshot and reboot

### Expected behavior
Normal functioning as in the 4.2.9 Bidirectional Dshot is working

### Flight controller configuration
BETAFPVF411RX (BF 4.3 #2476) with Jesc Firmware

PASTE THE OUTPUT OF 'diff' HERE
```
# 

# diff all

# version
# Betaflight / STM32F411 (S411) 4.2.9 Apr 27 2021 / 19:33:23 (e097f4ab7) MSP API: 1.43
# config: manufacturer_id: BEFH, board_name: BETAFPVF411RX, version: b500c5ae, date: 2020-04-26T06:31:02Z

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name BETAFPVF411RX
manufacturer_id BEFH
mcu_id 005100163036511036383539
signature 

# name: Barux

# resources
resource SERIAL_TX 2 NONE
resource SERIAL_RX 2 NONE
resource CAMERA_CONTROL 1 A02
resource PINIO 1 A03

# timer
timer A02 AF2
# pin A02: TIM5 CH3 (AF2)

# feature
feature -RX_SPI
feature RX_SERIAL
feature LED_STRIP

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -RX_SET
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CRASH_FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL

# beacon
beacon RX_LOST
beacon RX_SET

# serial
serial 0 64 115200 57600 0 115200

# led
led 0 0,0::CT:13
led 1 1,0::CT:13
led 2 2,0::CT:13
led 3 3,0::CT:13
led 4 4,0::CT:13
led 5 5,0::CT:13
led 6 6,0::CT:13
led 7 7,0::CT:13
led 8 8,0::CT:13
led 9 9,0::CT:13
led 10 10,0::CT:13
led 11 11,0::CT:13

# color
color 13 30,0,255

# mode_color
mode_color 7 0 6

# aux
aux 0 0 1 1875 2100 0 0
aux 1 15 0 1050 2100 0 0
aux 2 26 1 1925 2100 0 0
aux 3 35 3 1900 2100 0 0
aux 4 36 2 1950 2100 0 0
aux 5 39 0 1050 2100 0 0
aux 6 49 3 1400 1625 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 1
vtxtable powerlabels 25

# master
set gyro_lowpass2_hz = 400
set dyn_notch_width_percent = 0
set dyn_notch_q = 250
set dyn_notch_min_hz = 125
set dyn_notch_max_hz = 350
set dyn_lpf_gyro_min_hz = 320
set dyn_lpf_gyro_max_hz = 800
set acc_calibration = -259,-118,247,1
set mag_hardware = NONE
set baro_hardware = NONE
set min_check = 1010
set max_check = 2000
set rc_smoothing_derivative_type = PT1
set rc_smoothing_auto_smoothness = 7
set serialrx_provider = CRSF
set min_throttle = 1010
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set motor_poles = 12
set vbat_max_cell_voltage = 450
set vbat_min_cell_voltage = 340
set vbat_duration_for_warning = 5
set ibata_scale = 205
set yaw_motors_reversed = ON
set small_angle = 180
set deadband = 1
set yaw_deadband = 1
set ledstrip_visual_beeper = ON
set ledstrip_profile = RACE
set ledstrip_race_color = YELLOW
set ledstrip_beacon_color = RED
set osd_warn_no_gps_rescue = OFF
set osd_warn_gps_rescue_disabled = OFF
set osd_rssi_alarm = 80
set osd_vbat_pos = 2456
set osd_rssi_pos = 26
set osd_tim_2_pos = 437
set osd_throttle_pos = 2393
set osd_vtx_channel_pos = 1
set osd_current_pos = 391
set osd_mah_drawn_pos = 385
set osd_craft_name_pos = 2433
set osd_warnings_pos = 2441
set osd_avg_cell_voltage_pos = 2424
set osd_stat_max_spd = OFF
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set osd_stat_min_link_quality = ON
set debug_mode = GYRO_SCALED
set vtx_band = 5
set vtx_channel = 8
set vtx_power = 1
set vtx_freq = 5917
set gyro_1_sensor_align = CW0FLIP
set gyro_1_align_pitch = 1800
set gyro_1_align_yaw = 0
set gyro_rpm_notch_harmonics = 1
set name = Barux

profile 0

# profile 0
set dyn_lpf_dterm_min_hz = 112
set dyn_lpf_dterm_max_hz = 272
set dterm_lowpass2_hz = 240
set vbat_sag_compensation = 100
set anti_gravity_gain = 6000
set iterm_relax_cutoff = 35
set throttle_boost = 0
set p_pitch = 69
set d_pitch = 58
set f_pitch = 152
set p_roll = 63
set d_roll = 54
set f_roll = 144
set p_yaw = 68
set f_yaw = 144
set d_min_roll = 0
set d_min_pitch = 0
set launch_control_mode = PITCHONLY
set launch_angle_limit = 50
set ff_spike_limit = 70
set ff_smooth_factor = 20

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set thr_expo = 35
set roll_rc_rate = 82
set pitch_rc_rate = 83
set yaw_rc_rate = 85
set roll_srate = 60
set pitch_srate = 60
set yaw_srate = 80
set tpa_breakpoint = 1650

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
# 
```
PASTE THE OUTPUT OF 'resource show all' HERE
```
# resource
resource MOTOR 1 B08
resource MOTOR 2 A00
resource MOTOR 3 B10
resource MOTOR 4 B07
resource PPM 1 A10
resource LED_STRIP 1 B01
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 11 A08
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A03
resource SERIAL_RX 11 A08
resource LED 1 C14
resource SPI_SCK 1 A05
resource SPI_SCK 2 B13
resource SPI_SCK 3 B03
resource SPI_MISO 1 A06
resource SPI_MISO 2 B14
resource SPI_MISO 3 B04
resource SPI_MOSI 1 A07
resource SPI_MOSI 2 B15
resource SPI_MOSI 3 B05
resource ADC_BATT 1 A01
resource ADC_CURR 1 B00
resource OSD_CS 1 B12
resource RX_SPI_CS 1 A15
resource RX_SPI_EXTI 1 C13
resource RX_SPI_BIND 1 B02
resource RX_SPI_LED 1 C15
resource RX_SPI_CC2500_TX_EN 1 B09
resource RX_SPI_CC2500_LNA_EN 1 A13
resource RX_SPI_CC2500_ANT_SEL 1 A14
resource GYRO_EXTI 1 B06
resource GYRO_CS 1 A04

### Setup / Versions
 Flight controller:  F4 1S AIO Brushless Flight Controller - BF 4.3 #2476 with Jesc Firmware

haslinghuis, 2021-07-02:
This feature in configurator is just checking the number of motors having an output against the motors needed in the selected mixer.
In `4.2.9` it's ok, but in `4.3.0` outputs seem to be zero.

SpeedyPizzaDrones, 2021-07-02:
@haslinghuis this is the resource of the motors I have in both 4.2.9 and 4.3.0 

resource MOTOR 1 B08
resource MOTOR 2 A00
resource MOTOR 3 B10
resource MOTOR 4 B07

so why in 4.3 is not working? is this a bug maybe?

@joelucid

haslinghuis, 2021-07-02:
Can you please change the resources output with resource show all - so we can see timers.
Also I see configuration of a timer in the diff:
```
# timer
timer A02 AF2
# pin A02: TIM5 CH3 (AF2)
```

SpeedyPizzaDrones, 2021-07-02:
@haslinghuis  Uhm never changed those, not sure why they are in the Diff.

"resource show all" is not recognized by CLI. Got the following by "Dump all" :

# resources
resource BEEPER 1 NONE
resource MOTOR 1 B08
resource MOTOR 2 A00
resource MOTOR 3 B10
resource MOTOR 4 B07
resource MOTOR 5 NONE
resource MOTOR 6 NONE
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 A10
resource PWM 1 NONE
resource PWM 2 NONE
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource LED_STRIP 1 B01
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 NONE
resource SERIAL_TX 4 NONE
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 NONE
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 A08
resource SERIAL_TX 12 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A03
resource SERIAL_RX 3 NONE
resource SERIAL_RX 4 NONE
resource SERIAL_RX 5 NONE
resource SERIAL_RX 6 NONE
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource SERIAL_RX 11 A08
resource SERIAL_RX 12 NONE
resource INVERTER 1 NONE
resource INVERTER 2 NONE
resource INVERTER 3 NONE
resource INVERTER 4 NONE
resource INVERTER 5 NONE
resource INVERTER 6 NONE
resource INVERTER 7 NONE
resource INVERTER 8 NONE
resource INVERTER 9 NONE
resource INVERTER 10 NONE
resource INVERTER 11 NONE
resource INVERTER 12 NONE
resource I2C_SCL 1 NONE
resource I2C_SCL 2 NONE
resource I2C_SCL 3 NONE
resource I2C_SDA 1 NONE
resource I2C_SDA 2 NONE
resource I2C_SDA 3 NONE
resource LED 1 C14
resource LED 2 NONE
resource LED 3 NONE
resource RX_BIND 1 NONE
resource RX_BIND_PLUG 1 NONE
resource SPI_SCK 1 A05
resource SPI_SCK 2 B13
resource SPI_SCK 3 B03
resource SPI_MISO 1 A06
resource SPI_MISO 2 B14
resource SPI_MISO 3 B04
resource SPI_MOSI 1 A07
resource SPI_MOSI 2 B15
resource SPI_MOSI 3 B05
resource ESCSERIAL 1 NONE
resource CAMERA_CONTROL 1 NONE
resource ADC_BATT 1 A01
resource ADC_RSSI 1 NONE
resource ADC_CURR 1 B00
resource ADC_EXT 1 NONE
resource BARO_CS 1 NONE
resource BARO_EOC 1 NONE
resource BARO_XCLR 1 NONE
resource COMPASS_CS 1 NONE
resource COMPASS_EXTI 1 NONE
resource SDCARD_CS 1 NONE
resource SDCARD_DETECT 1 NONE
resource PINIO 1 NONE
resource PINIO 2 NONE
resource PINIO 3 NONE
resource PINIO 4 NONE
resource USB_MSC_PIN 1 NONE
resource FLASH_CS 1 NONE
resource OSD_CS 1 B12
resource RX_SPI_CS 1 A15
resource RX_SPI_EXTI 1 C13
resource RX_SPI_BIND 1 B02
resource RX_SPI_LED 1 C15
resource RX_SPI_CC2500_TX_EN 1 B09
resource RX_SPI_CC2500_LNA_EN 1 A13
resource RX_SPI_CC2500_ANT_SEL 1 A14
resource GYRO_EXTI 1 B06
resource GYRO_EXTI 2 NONE
resource GYRO_CS 1 A04
resource GYRO_CS 2 NONE
resource USB_DETECT 1 NONE
resource PULLUP 1 NONE
resource PULLUP 2 NONE
resource PULLUP 3 NONE
resource PULLUP 4 NONE
resource PULLDOWN 1 NONE
resource PULLDOWN 2 NONE
resource PULLDOWN 3 NONE
resource PULLDOWN 4 NONE

# timer
timer A00 AF1
# pin A00: TIM2 CH1 (AF1)
timer A08 AF1
# pin A08: TIM1 CH1 (AF1)
timer A10 AF1
# pin A10: TIM1 CH3 (AF1)
timer B01 AF2
# pin B01: TIM3 CH4 (AF2)
timer B07 AF2
# pin B07: TIM4 CH2 (AF2)
timer B08 AF2
# pin B08: TIM4 CH3 (AF2)
timer B10 AF1
# pin B10: TIM2 CH3 (AF1)

# dma
dma SPI_TX 1 NONE
dma SPI_TX 2 NONE
dma SPI_TX 3 NONE
dma SPI_RX 1 NONE
dma SPI_RX 2 NONE
dma SPI_RX 3 NONE
dma ADC 1 0
# ADC 1: DMA2 Stream 0 Channel 0
dma ADC 2 NONE
dma ADC 3 NONE
dma UART_TX 1 NONE
dma UART_TX 2 NONE
dma UART_TX 3 NONE
dma UART_TX 4 NONE
dma UART_TX 5 NONE
dma UART_TX 6 NONE
dma UART_TX 7 NONE
dma UART_TX 8 NONE
dma UART_RX 1 NONE
dma UART_RX 2 NONE
dma UART_RX 3 NONE
dma UART_RX 4 NONE
dma UART_RX 5 NONE
dma UART_RX 6 NONE
dma UART_RX 7 NONE
dma UART_RX 8 NONE
dma pin A00 0
# pin A00: DMA1 Stream 5 Channel 3
dma pin A08 0
# pin A08: DMA2 Stream 6 Channel 0
dma pin A10 NONE
dma pin B01 0
# pin B01: DMA1 Stream 2 Channel 5
dma pin B07 0
# pin B07: DMA1 Stream 3 Channel 2
dma pin B08 0
# pin B08: DMA1 Stream 7 Channel 2
dma pin B10 0
# pin B10: DMA1 Stream 1 Channel 3

haslinghuis, 2021-07-02:
`resource show all` should work. We use it all the time? Maybe save your diff, flash again and first try to setup motors and rpm filtering before pasting a diff (as everybody does for the safe parts)

Example on Matek F722Mini:
```
resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: FREE
A03: SERIAL_RX 2
A04: FREE
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: LED 2
A14: LED 1
A15: FREE
B00: MOTOR 3
B01: MOTOR 4
B02: GYRO_CS 1
B03: FREE
B04: MOTOR 1
B05: MOTOR 2
B06: FREE
B07: FREE
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: FREE
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_CURR
C02: ADC_BATT
C03: FREE
C04: GYRO_EXTI
C05: FREE
C06: FREE
C07: FREE
C08: PINIO 1
C09: PINIO 2
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: BEEPER
C14: FREE
C15: GYRO_CS 2
D00: FREE
D01: FREE
D02: FLASH_CS
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1: DSHOT_BITBANG 2
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 2
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

# 
```

SpeedyPizzaDrones, 2021-07-02:
@haslinghuis sorry, my bad on typing. Here you go hereunder. 

Yes after reflashing I do nothing if not just turning on Bidirectional Dshot and Dshot600 (with 300 is same) with default Quad X, then get that message. Anything weird hereunder?

A00: MOTOR 2
A01: ADC_BATT
A02: FREE
A03: FREE
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: SERIAL_TX 11
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: RX_SPI_CC2500_LNA_EN
A14: RX_SPI_CC2500_ANT_SEL
A15: RX_SPI_CS
B00: ADC_CURR
B01: FREE
B02: RX_SPI_BIND
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: GYRO_EXTI
B07: MOTOR 4
B08: MOTOR 1
B09: RX_SPI_CC2500_TX_EN
B10: MOTOR 3
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: FREE
C03: FREE
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: RX_SPI_EXTI
C14: LED 1
C15: LED
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE

Currently active Timers:
-----------------------
TIM1:
    CH1: SERIAL_TX 11
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC 1
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

haslinghuis, 2021-07-02:
Yes afaik RPM filter depends on TIM1 or TIM8. I don't see the latter and the first is occupied by the SPI RX. Not sure what is going wrong.

SpeedyPizzaDrones, 2021-07-02:
@haslinghuis  who should we ping here? :)

haslinghuis, 2021-07-02:
@mikeller or @SteveCEvans or @jflyper or even @hydra

SpeedyPizzaDrones, 2021-07-02:
@haslinghuis thank you! let's see

Asizon, 2021-07-03:
First, this issue should go into Configurator repo issues.
I have tried disabling Dshot BitBang and works, then this issue is related to these timers asigments.

SpeedyPizzaDrones, 2021-07-03:
@Asizon so is the configurator that assigns wrong Timers? Maybe from the "Apply default" of that specific target?

Asizon, 2021-07-03:
@SpeedyPizzaDrones not wrong timers, there are some troubleshoots with F411 boards and Bitbang dshot.

SpeedyPizzaDrones, 2021-07-03:
@Asizon got it thanks. If I have to do any other test let me know 👍🏻

Asizon, 2021-07-03:
Yes i have do another test using MatekF411Rx target and works as expected, im going to look at Betafpvf411 config now.

Asizon, 2021-07-03:
Ok definitelly the problem is here,not sure why @betafpv-engineer removed Bitbang Off change in their board config here:
https://github.com/betaflight/unified-targets/commit/715ff3cdaff4a769febbddf4dd2f42f709263fd1#diff-3223d02af2dfa489274ab57db642a069d934d57286534f38047149b50cebd4dd

Asizon, 2021-07-03:
@SpeedyPizzaDrones maybe you are using the incorrect target, what is exactly your board(link)?

SpeedyPizzaDrones, 2021-07-03:
@Asizon  this is the board link (no rx version, using Tracer) : https://betafpv.com/products/f4-1s-aio-brushless-flight-controller?_pos=1&_sid=2377bee59&_ss=r

I'm using the same target as the default, which works with Bidirectional Dshot on 4.2.9 : BETAFPVF411RX

Asizon, 2021-07-03:
This betafpv target is designed for Toothpick board that include blackbox flash, so please can you try flashing "MATEKF411" target and see if all works as expected?

SpeedyPizzaDrones, 2021-07-03:
@Asizon Tried MATEKF411 but only Motor 4 works

[image]

Asizon, 2021-07-03:
And using MATEKF411RX?

haslinghuis, 2021-07-03:
The link provided uses a STM32 unified target, is BetaFPV still working with dumps?

Configurator is just verifying the motor outputs available from firmware. As firmware is reporting 0 motors in this case it presents the dialog (we could include a message indicating enabling RPM settings could render the firmware incapable of having outputs), so it's a firmware (configuration) issue.

SpeedyPizzaDrones, 2021-07-03:
@Asizon tried MATEKF411RX and several others but nothing to do.. on 2 same fc's. 

@haslinghuis  I'd say this model of fc has issues, don't know if firmware or whatever but even for esc firmware there is something wrong: on website is claimed "O_H_5_REV16_8.HEX" but they come with S_H_70 16.7 .

At this point I have 2 fc's with identical issue whatever BF firmware is uploaded: only motor 4 is spinning

not even sure if there s a solution. Fun fact it's it was flying on day 1 with 4.2.9 (conf. 10.7.0), Jesc, and RPM ON

Asizon, 2021-07-04:
works in 4.2 because it comes from factory with an custom target, please contact to @betafpv-engineer

wenzhicode, 2021-07-05:
> @haslinghuis I'd say this model of fc has issues, don't know if firmware or whatever but even for esc firmware there is something wrong: on website is claimed "O_H_5_REV16_8.HEX" but they come with S_H_70 16.7 .

The no rx version is for 1-2s, esc firmware is S_H_70_REV16_7.HEX.

wenzhicode, 2021-07-05:
> Ok definitelly the problem is here,not sure why @betafpv-engineer removed Bitbang Off change in their board config here:
> [betaflight/unified-targets@715ff3c#diff-3223d02af2dfa489274ab57db642a069d934d57286534f38047149b50cebd4dd](https://github.com/betaflight/unified-targets/commit/715ff3cdaff4a769febbddf4dd2f42f709263fd1#diff-3223d02af2dfa489274ab57db642a069d934d57286534f38047149b50cebd4dd)

I think it is for safety.

betafpv-engineer, 2021-07-05:
> Ok definitelly the problem is here,not sure why @betafpv-engineer removed Bitbang Off change in their board config here: [...]

For safety reasons, when the ESC does not support the dshot protocol, this will result in the motors spinning up as soon as a battery is connected, potentially resulting in damage or injury.

closed 2021-07-05 by haslinghuis

betafpv-engineer, 2021-07-05:
@SpeedyPizzaDrones  I'm very sorry for causing you trouble. For the F4 1S AIO NO RX flight control, FC uses the BETAFPVF411RX target, and ESC uses the S_H target. You need to check whether these 2 firmwares are correct first.

A complete dump:
`# 

# dump

# version
# Betaflight / STM32F411 (S411) 4.2.9 Apr 27 2021 / 19:33:23 (e097f4ab7) MSP API: 1.43
# config: manufacturer_id: BEFH, board_name: BETAFPVF411RX, version: b500c5ae, date: 2020-04-26T06:31:02Z

# start the command batch
batch start

board_name BETAFPVF411RX
manufacturer_id BEFH

# name: -

# resources
resource BEEPER 1 NONE
resource MOTOR 1 B08
resource MOTOR 2 A00
resource MOTOR 3 B10
resource MOTOR 4 B07
resource MOTOR 5 NONE
resource MOTOR 6 NONE
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 A10
resource PWM 1 NONE
resource PWM 2 NONE
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource SONAR_TRIGGER 1 NONE
resource SONAR_ECHO 1 NONE
resource LED_STRIP 1 B01
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 NONE
resource SERIAL_TX 3 NONE
resource SERIAL_TX 4 NONE
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 NONE
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 A08
resource SERIAL_TX 12 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 NONE
resource SERIAL_RX 3 NONE
resource SERIAL_RX 4 NONE
resource SERIAL_RX 5 NONE
resource SERIAL_RX 6 NONE
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
[... 1039 lines cut ...]
set pitch_srate = 70
set yaw_srate = 70
set tpa_rate = 65
set tpa_breakpoint = 1350
set tpa_mode = D
set throttle_limit_type = OFF
set throttle_limit_percent = 100
set roll_rate_limit = 1998
set pitch_rate_limit = 1998
set yaw_rate_limit = 1998

# end the command batch
batch end

# 
`

If you need to turn on Bidirectional DShot(requires supported ESC firmware), you need to enter CLI:

`set dshot_idle_value = 550
set dshot_burst = AUTO
set dshot_bidir = ON
set dshot_bitbang = OFF
set dshot_bitbang_timer = AUTO
save`

betafpv-engineer, 2021-07-05:
@SpeedyPizzaDrones  I'm very sorry for causing you trouble. For the F4 1S AIO NO RX flight control, FC uses the BETAFPVF411RX target, and ESC uses the S_H target. You need to check whether these 2 firmwares are correct first.

A complete dump:
`# 

# dump

# version
# Betaflight / STM32F411 (S411) 4.2.9 Apr 27 2021 / 19:33:23 (e097f4ab7) MSP API: 1.43
# config: manufacturer_id: BEFH, board_name: BETAFPVF411RX, version: b500c5ae, date: 2020-04-26T06:31:02Z

# start the command batch
batch start

board_name BETAFPVF411RX
manufacturer_id BEFH

# name: -

# resources
resource BEEPER 1 NONE
resource MOTOR 1 B08
resource MOTOR 2 A00
resource MOTOR 3 B10
resource MOTOR 4 B07
resource MOTOR 5 NONE
resource MOTOR 6 NONE
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 A10
resource PWM 1 NONE
resource PWM 2 NONE
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource SONAR_TRIGGER 1 NONE
resource SONAR_ECHO 1 NONE
resource LED_STRIP 1 B01
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 NONE
resource SERIAL_TX 3 NONE
resource SERIAL_TX 4 NONE
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 NONE
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 A08
resource SERIAL_TX 12 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 NONE
resource SERIAL_RX 3 NONE
resource SERIAL_RX 4 NONE
resource SERIAL_RX 5 NONE
resource SERIAL_RX 6 NONE
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
[... 1038 lines cut ...]
set roll_srate = 70
set pitch_srate = 70
set yaw_srate = 70
set tpa_rate = 65
set tpa_breakpoint = 1350
set tpa_mode = D
set throttle_limit_type = OFF
set throttle_limit_percent = 100
set roll_rate_limit = 1998
set pitch_rate_limit = 1998
set yaw_rate_limit = 1998

# end the command batch
batch end

# 
`

If you need to turn on Bidirectional DShot(requires supported ESC firmware), you need to enter CLI:
set dshot_idle_value = 550
set dshot_burst = AUTO
set dshot_bidir = ON
set dshot_bitbang = OFF
set dshot_bitbang_timer = AUTO
save

betafpv-engineer, 2021-07-05:
@SpeedyPizzaDrones  For F4 1S AIO SPI Rx flight control, FC uses MATEKF411RX target, and ESC uses O_H target. You need to check whether these 2 firmwares are correct.

A complete dump:

`
# 

# dump
###WARNING: NO CUSTOM DEFAULTS FOUND###

# version
# Betaflight / STM32F411 (S411) 4.2.0 Jun 14 2020 / 03:04:43 (8f2d21460) MSP API: 1.43
###ERROR: dump: NO CONFIG FOUND###
# start the command batch
batch start

board_name MATEKF411RX
manufacturer_id MTKS

# name: Meteor65

# resources
resource BEEPER 1 C15
resource MOTOR 1 B10
resource MOTOR 2 B06
resource MOTOR 3 B07
resource MOTOR 4 B08
resource MOTOR 5 NONE
resource MOTOR 6 NONE
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 A03
resource PWM 1 A02
resource PWM 2 A09
resource PWM 3 A10
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource SONAR_TRIGGER 1 NONE
resource SONAR_ECHO 1 NONE
resource LED_STRIP 1 A00
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 NONE
resource SERIAL_TX 4 NONE
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 NONE
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 NONE
resource SERIAL_TX 12 NONE
resource SERIAL_RX 1 NONE
resource SERIAL_RX 2 A03
resource SERIAL_RX 3 NONE
resource SERIAL_RX 4 NONE
resource SERIAL_RX 5 NONE
resource SERIAL_RX 6 NONE
resource SERIAL_RX 7 NONE
[... 1041 lines cut ...]
set yaw_expo = 0
set roll_srate = 70
set pitch_srate = 70
set yaw_srate = 70
set tpa_rate = 45
set tpa_breakpoint = 1550
set tpa_mode = D
set throttle_limit_type = OFF
set throttle_limit_percent = 100
set roll_rate_limit = 1998
set pitch_rate_limit = 1998
set yaw_rate_limit = 1998

# end the command batch
batch end

# 
`
If you need to turn on Bidirectional DShot(requires supported ESC firmware), you need to enter CLI:
set dshot_idle_value = 550
set dshot_burst = AUTO
set dshot_bidir = ON
set dshot_bitbang = OFF
set dshot_bitbang_timer = AUTO
save

Asizon, 2021-07-05:
Thankyou for explain @betafpv-engineer, so definitelly the issue is related to known Bitbang issue on some F411 boards.

Asizon, 2021-07-05:
> For safety reasons, when the ESC does not support the dshot protocol, this will result in the motors spinning up as soon as a battery is connected, potentially resulting in damage or injury.

@betafpv-engineer Your are talking here about dshot protocol that yes, is better to take it "DISABLED", but it is not related to Dshot Bigbang feature that you can disable by default in your config like here:

https://github.com/betaflight/unified-targets/pull/95

Asizon, 2021-07-05:
Your comment comes from @mikeller coments her https://github.com/betaflight/unified-targets/pull/327#discussion_r464322780 , that is completly related to dshot protocol that now we recomend to use DISABLED by default. So you can disable bitbang in your target config to avoid TIM1 conflicts and user reports.

SpeedyPizzaDrones, 2021-07-05:
[image]

[image]

But even ESC have some issues, everytime I connect and disconnect it reads different things in blheli and is not possible to reflash because not able to read them all. On both Fc's

@betafpv-engineer

liyouluo, 2021-07-06:
> > > Ok definitelly the problem is here,not sure why @betafpv-engineer removed Bitbang Off change in their board config here:
> > > [betaflight/unified-targets@715ff3c#diff-3223d02af2dfa489274ab57db642a069d934d57286534f38047149b50cebd4dd](https://github.com/betaflight/unified-targets/commit/715ff3cdaff4a769febbddf4dd2f42f709263fd1#diff-3223d02af2dfa489274ab57db642a069d934d57286534f38047149b50cebd4dd)
> > 
> > 
> > For safety reasons, when the ESC does not support the dshot protocol, this will result in the motors spinning up as soon as a battery is connected, potentially resulting in damage or injury.
> 
> @betafpv-engineer Your are talking here about dshot protocol that yes, is better to take it "DISABLED", but it is not related to Dshot Bitbang feature that you can disable by default in your config like here:
> 
> [betaflight/unified-targets#95](https://github.com/betaflight/unified-targets/pull/95)

Thank you for your help, I will perfect him.

liyouluo, 2021-07-06:
> [image]
> 
> [image]
> 
> But even ESC have some issues, everytime I connect and disconnect it reads different things in blheli and is not possible to reflash because not able to read them all. On both Fc's
> 
> @betafpv-engineer
Can you contact me by mail? I will assist you to solve the problem, my email: engineer@betafpv.com.

bru-martin, 2022-02-13:
Io ho risolto così:
set dshot_bitbang = ON  (Sbagliato)
set dshot_bitbang = OFF (Giusto)