betaflight/betaflight #10822: Auto land on gps rescue mode

hapsti, 2021-07-05:
### Is your feature request related to a problem? Please describe.
I would like to be able to flick on gps recue mode and have my quad fly back to me, but not drop out of the sky when near, it would be really nice to have it auto land.

### Describe the solution you'd like
The quad should stay at its altitude, not starts decending, in case obstacles are infront of home position, and then start slowing down as it nears the home position, hove over it at the defined altitude in failsafe tab, and then decend while maintaining position

### Describe alternatives you've considered
An alternative, but auto land is much better, may be to circle around home pos for a while, letting you reconnect to rx, and if not done in a certain period of time, auto land

butterSnake, 2021-07-06:
Not enough sensors on our fcs for that. You need proper hight above ground sensors, eg ultrasound ranger or lidar. Then there is the coding and tuning of such a feature with each quad needing tuning to make it work. Inav is more focused automated features, frankly it’s not in keeping with the direction Betaflight so you can ask but I doubt your going to receive

hapsti, 2021-07-06:
> Not enough sensors on our fcs for that. You need proper hight above ground sensors, eg ultrasound ranger or lidar. Then there is the coding and tuning of such a feature with each quad needing tuning [...]

Actually inav does a very good job rth with the normal fc sensors, its just betalfights needs a bit of tuning, like not turning around when over the home point and instead trying to hold its position while descending, also not descending until above the home point would be good.

robustini, 2021-07-17:
INAV autoland perfectly, same hardware.

mituritsyn, 2021-08-10:
If one want autoland, autoreturn or anything else with auto prefix - he should use Inav or ardupilot.

mikeller, 2021-08-10:
@hapsti: Looks like you are answering your own question here: :grinning: 
> Actually inav does a very good job rth with the normal fc sensors, its just betalfights needs a bit of tuning, like not turning around when over the home point and instead trying to hold its position while descending, also not descending until above the home point would be good.

Just switch to iNav and you get a product that specialises in exactly the things you want.

closed 2021-09-18 by github-actions[bot]