betaflight/betaflight #10825: JHEMCU F405 Pro AIO "No Gyro" Error

CloslaRoche, 2021-07-05:
### Describe the bug
When flashing the JHEMCU F405 pro AIO board 45a, using the 'JHEF405' Target with any version produces a "no gyro" error, and the 2 FC boot beeps are no longer present when plugging in a battery. 

The board came flashed with 4.1.0 and was working perfectly before attempting to update to 4.2.9

### To Reproduce
Attempt to flash a JHEMCU F405 Pro AIO with betaflight using the configurator 

### Expected behavior
Gyro to be detected and working and the FC beeps to be present 

### Flight controller configuration
N/A, cant configure as cant fly

### Setup / Versions
JHEMCU F405 Pro AIO 45a. Tried flashing with 4.1.0, 4.2.0, 4.2.3 and 4.2.9

haslinghuis, 2021-07-05:
Please provide a `diff` and `resource show all` with three backticks (`) before and after the output.
You should be flashing `JHEF405`.

CloslaRoche, 2021-07-05:
```
# diff

# version
# Betaflight / STM32F405 (S405) 4.2.9 Apr 27 2021 / 19:33:01 (e097f4ab7) MSP API: 1.43
# config: manufacturer_id: JHEF, board_name: JHEF405, version: dfa09d94, date: 2021-03-06T10:55:26Z

# start the command batch
batch start

board_name JHEF405
manufacturer_id JHEF

# feature
feature -RX_PARALLEL_PWM

# master
set gyro_1_align_yaw = 900

profile 0

rateprofile 0

# end the command batch
batch end
```

```
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: FREE
A03: FREE
A04: PREINIT 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: PREINIT 2
B00: FREE
B01: FREE
B02: FREE
B03: FREE
B04: BEEPER
B05: LED 1
B06: FREE
B07: FREE
B08: PULLDOWN 1
B09: FREE
B10: FREE
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_CURR
C02: ADC_BATT
C03: FREE
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: INVERTER 2
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: ADC 1
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```
```
# resource
resource BEEPER 1 B04
resource MOTOR 1 B00
resource MOTOR 2 B01
resource MOTOR 3 C09
resource MOTOR 4 C08
resource MOTOR 5 B06
resource PPM 1 B08
resource PWM 1 B08
resource LED_STRIP 1 B06
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 B10
resource SERIAL_TX 6 C06
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A03
resource SERIAL_RX 3 B11
resource SERIAL_RX 6 C07
resource INVERTER 2 C13
resource LED 1 B05
resource SPI_SCK 1 A05
resource SPI_SCK 2 B13
resource SPI_SCK 3 C10
resource SPI_MISO 1 A06
resource SPI_MISO 2 B14
resource SPI_MISO 3 C11
resource SPI_MOSI 1 A07
resource SPI_MOSI 2 B15
resource SPI_MOSI 3 C12
resource ESCSERIAL 1 B08
resource ADC_BATT 1 C02
resource ADC_CURR 1 C01
resource FLASH_CS 1 A15
resource OSD_CS 1 B12
resource GYRO_EXTI 1 C04
resource GYRO_CS 1 A04
resource USB_DETECT 1 C05
resource PULLDOWN 1 B08
```

CloslaRoche, 2021-07-05:
> Please provide a `diff` and `resource show all` with three [backticks](https://docs.github.com/en/github/writing-on-github/working-with-advanced-formatting/creating-and-highlighting-code-blocks#fenced-code-blocks) before and after the output.
> You should be flashing `JHEF405`.

I am flashing using the JHEF405 target. 

Also, can confirm the JHEF405 target works for the non-pro version boards.

haslinghuis, 2021-07-05:
Weird nothing exotic here, gyro is somehow detected otherwise it would not put in these lines:
```
# master
set gyro_1_align_yaw = 900
```
@atgfpeyv

CloslaRoche, 2021-07-05:
Yep I applied custom defaults after on all occasions. 

Yea no gyro, no accelerometer, "onboard ADC" doesn't read voltage from USB or battery either.

CloslaRoche, 2021-07-06:
Ok JHEMCU has provided me with firmware and a CLI Resource map: 

Attached is a ZIP of the 4.2.6 hex they sent me and a txt file of the resource map

Everything seems to be working. I have Gyro, Accel, usb power and Smart Audio is working. Haven't flown it yet. 

Now how do we go about getting this as an actual firmware target? 

[Firmware (extract.me).zip](https://github.com/betaflight/betaflight/files/6772743/Firmware.extract.me.zip)

```
board_name JHE405PRO
manufacturer_id JHEM

# resources
resource BEEPER 1 C13
resource MOTOR 1 B00
resource MOTOR 2 B01
resource MOTOR 3 A03
resource MOTOR 4 A02
resource MOTOR 5 B05
resource MOTOR 6 B07
resource MOTOR 7 C09
resource MOTOR 8 C08
resource PPM 1 B08
resource LED_STRIP 1 A09
resource SERIAL_TX 1 B06
resource SERIAL_TX 2 D05
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 A00
resource SERIAL_TX 6 C06
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 D06
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 A01
resource SERIAL_RX 6 C07
resource I2C_SCL 1 B08
resource I2C_SDA 1 B09
resource LED 1 C14
resource SPI_SCK 1 A05
resource SPI_SCK 3 C10
resource SPI_MISO 1 A06
resource SPI_MISO 3 C11
resource SPI_MOSI 1 A07
resource SPI_MOSI 3 C12
resource ESCSERIAL 1 B08
resource ADC_BATT 1 C03
resource ADC_RSSI 1 C00
resource ADC_CURR 1 C02
resource FLASH_CS 1 B03
resource OSD_CS 1 B14
resource GYRO_EXTI 1 B13
resource GYRO_CS 1 B12
resource USB_DETECT 1 A08

# timer
timer B00 AF2
# pin B00: TIM3 CH3 (AF2)
timer B01 AF2
# pin B01: TIM3 CH4 (AF2)
timer A03 AF1
# pin A03: TIM2 CH4 (AF1)
timer A02 AF1
# pin A02: TIM2 CH3 (AF1)
timer B05 AF2
# pin B05: TIM3 CH2 (AF2)
timer B07 AF2
# pin B07: TIM4 CH2 (AF2)
timer C09 AF3
# pin C09: TIM8 CH4 (AF3)
timer C08 AF3
# pin C08: TIM8 CH3 (AF3)
timer A09 AF1
# pin A09: TIM1 CH2 (AF1)

# dma
dma ADC 1 0
# ADC 1: DMA2 Stream 0 Channel 0
dma pin B00 0
# pin B00: DMA1 Stream 7 Channel 5
dma pin B01 0
# pin B01: DMA1 Stream 2 Channel 5
dma pin A03 1
# pin A03: DMA1 Stream 6 Channel 3
dma pin A02 0
# pin A02: DMA1 Stream 1 Channel 3
dma pin B05 0
# pin B05: DMA1 Stream 5 Channel 5
dma pin B07 0
# pin B07: DMA1 Stream 3 Channel 2
dma pin C09 0
# pin C09: DMA2 Stream 7 Channel 7
dma pin C08 0
# pin C08: DMA2 Stream 2 Channel 0
dma pin A09 0
# pin A09: DMA2 Stream 6 Channel 0

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature MOTOR_STOP
feature RSSI_ADC
feature LED_STRIP
feature OSD

# led
led 0 4,6::CB:8
led 1 5,6::CB:8
led 2 6,6::CB:8
led 3 7,6::CB:8

# master
set mag_bustype = I2C
set mag_i2c_device = 1
set mag_hardware = NONE
set baro_bustype = I2C
set baro_i2c_device = 1
set serialrx_provider = SBUS
set blackbox_device = SPIFLASH
set dshot_burst = ON
set motor_pwm_protocol = DSHOT600
set current_meter = ADC
set battery_meter = ADC
set ibata_scale = 170
set beeper_inversion = ON
set beeper_od = OFF
set osd_vbat_pos = 14571
set osd_ah_sbar_pos = 14542
set osd_ah_pos = 14414
set osd_current_pos = 14634
set system_hse_mhz = 8
set max7456_spi_bus = 3
set dashboard_i2c_bus = 1
set pinio_box = 40,41,255,255
set flash_spi_bus = 3
set gyro_1_bustype = SPI
set gyro_1_spibus = 1
set gyro_1_sensor_align = CW180
set gyro_1_align_yaw = 1800
set gyro_2_spibus = 1
save
```

haslinghuis, 2021-07-06:
@atgfpeyv Could you please look into this and update the config in the unified targets?

CloslaRoche, 2021-07-08:
Another update: 

The config JHEMCU provided is incorrect. 

The OSD is assigned to pin B14 in the provided config. However I did not have an OSD after flashing the config and assumed the OSD chip was bad. 

But, the OSD is actually on pin A04.

haslinghuis, 2021-07-08:
@CloslaRoche I have no other contacts with JHEMCU. Maybe also contact them directly.

CloslaRoche, 2021-07-08:
@haslinghuis I have been in contact with them this whole time, I'm posting to keep a log of things in case someone else runs into the issue.

haslinghuis, 2021-07-08:
Much appreciated but it's clear there needs to be an RFC for the config issue.

CloslaRoche, 2021-07-10:
Wanted to update, still having OSD issues. Only way to get the OSD to show up is to go into the BF CLI, wait for the 2 FC boot beeps, then click the OSD tab. Otherwise the OSD does not show up.

howels, 2021-09-28:
> Another update:
> 
> The config JHEMCU provided is incorrect.
> 
> The OSD is assigned to pin B14 in the provided config. However I did not have an OSD after flashing the config and assumed the OSD chip was bad.
> 
> But, the OSD is actually on pin A04.

Awwww no. I was testing on DJI, so I didn't catch that osd issue and assumed that B14 was correct. If you are sure it's A04 I can submit a patch.

carbendi, 2021-10-02:
For me B14 is working fine on Analog and DJI OSD on betaflight 4.3

howels, 2021-10-03:
> For me B14 is working fine on Analog and DJI OSD on betaflight 4.3

thanks for testing that.

howels, 2022-04-15:
This should be closed, the creation of the new JHE F405 PRO AIO target resolved this issue.

closed 2022-04-15 by klutvott123