betaflight/betaflight #10827: Unusable SITL and High Cycle Time

ad1tyat, 2021-07-07:
### Describe the bug

Once I compile betaflight with target SITL, the cycle time is ~210, but to actually use it, the official [README](https://github.com/betaflight/betaflight/tree/master/src/main/target/SITL) it advises to  
>In configuration page:
>
>ESC/Motor: PWM, disable Motor PWM speed Sparted from PID speed
>PID loop frequency as high as it can.

Once I do this, the cycle time becomes 3000-4000 and is too slow to use. I am using a modified version of [YAMSPy](https://github.com/ad1tyat/YAMSPy/tree/usetcp) to test it.
### To Reproduce

Compile betaflight with TARGET=SITL as written [here](https://github.com/betaflight/betaflight/tree/master/src/main/target/SITL)
Connect Betaflight Configurator and select Manual Selection , enter port = `tcp://127.0.0.1:5761`  

Do the changes written [here](https://github.com/betaflight/betaflight/tree/master/src/main/target/SITL)  

```bash
git clone https://github.com/ad1tyat/YAMSPy.git
cd YAMSPy
git checkout -b usetcp
python3 simpleUI.py
```

### Expected behavior

The SITL should run faster and work with the simpleUI.py interface

### Flight controller configuration

```
diff

# version
# Betaflight / SITL (SITL) 4.3.0 Jul  7 2021 / 16:13:37 (3d5a0e585) MSP API: 1.44

# start the command batch
batch start

board_name SITL

# feature
feature -GPS
feature -DYNAMIC_FILTER

# aux
aux 0 0 255 1400 2050 0 0
aux 1 1 255 1300 1700 0 0

# master
set motor_pwm_protocol = PWM
set pid_process_denom = 16

profile 0

rateprofile 0

# end the command batch
batch end

# 
```
```
Error no such command ? 'resource show all'
```

### Setup / Versions
 
 - Flight controller: 
     - Betaflight
     - 'flightControllerVersion': '4.3.0'
     - boardIdentifier': 'SITL'
 - Other components: NA
 
 - How are the different components wired up: NA

KarateBrot, 2021-07-07:
`set pid_process_denom = 16` looks very unusual. What happens if you `set pid_process_denom = 1` ?

ad1tyat, 2021-07-07:
Thanks for your suggestion, I tried 
```
set pid_process_denom = 1
save
```
and restarted Betaflight and the configurator, however, it doesn't get saved, diff shows `set pid_process_denom = 16` again

KarateBrot, 2021-07-07:
Ok, I have no idea about this stuff but can you set the motor protocol to DSHOT600? Maybe after that you can `set pid_process_denom = 1`

ad1tyat, 2021-07-07:
I tried this, the clock speed is not better 100-200, but now I can't test- fly it using betaflight configurator (which I earlier could by going to the `Motors` tab and just lift it off )
simpleUI.py performance hasn't improved, still slow (note the `GUI cycleTime`).
The simpleUI.py screen : 
```
ARMED: False                                      armingDisableFlags: ['RXLOSS', 'ANGLE', '', '']
cpuload: 0                                        cycleTime: 152
mode: 0                                           Flight Mode: []
Battery Voltage: 0.00V  ULTRA LOW BATT!!!
Motor Values: [0, 0, 0, 0, 0, 0, 0, 0]

GUI cycleTime: 501.43ms (average 1.81Hz)

apiVersion: 1.44.0                                flightControllerIdentifier: BTFL
flightControllerVersion: 4.3.0                    boardIdentifier: SITL
boardName: SITL                                   name: 
```
>The readme states
>>disable Motor PWM speed Sparted from PID speed
>Do you know what it means? Maybe that's whats prevents you from changing pid_process_denom

I'll try to understand what this means. Thanks.

KarateBrot, 2021-07-07:
@ad1tyat 
It seems to be a typing error. It means: 
_disable Motor PWM speed **separated** from PID speed_

[image]

closed 2024-01-01 by haslinghuis