betaflight/betaflight #10831: Please add support of new imu icm42605 and icm42688.

and-sh, 2021-07-10:
### Is your feature request related to a problem? Please describe.
Progress does not stand still and tdk has released several new imu. 
The top of the list of low noise imu is attached: 
[ProductComparison.pdf](https://github.com/betaflight/betaflight/files/6795105/ProductComparison.pdf)
Unfortunately, there was no revolutionary improvement in performance. 
But all characteristics, such as stability, accuracy, linearity, zero output, both for the gyroscope and for the accelerometer, have improved. 

### Describe the solution you'd like
Please add support of new gyros icm42605 and icm42688.

### Describe alternatives you've considered
Using old gyros.

Quick-Flash, 2021-07-13:
icm42605 and icm42688-p support is already added to the code.

and-sh, 2021-07-13:
matek 722 mini se
BF 4.3.0 a week ago.
no gyro. Who am I = 42 - icm42605, 
I have icm42688 IC already, haven't tried it yet.
[image]

bf 4.2.8 from matek website, the same.

Quick-Flash, 2021-07-13:
Well it's in the code and dominic is making fc with both gyros and they both work.

and-sh, 2021-07-14:
Ok. I will check it one more time and will make bug report instead.

closed 2021-07-14 by and-sh