betaflight/betaflight #10839: ESC Temperature Displays Incorrect in Betaflight (but ok in INAV)

GaryDT, 2021-07-13:
### Describe the bug

ESC temperature displays incorrectly in DJI OSD, even after enabling UART 7, ESC in the Sensors Input dropdown, and enabling ESC_SENSOR in the configuration. The initial value prior to enabling was fixed at 32 degrees Fahrenheit (probably the expected default), which changed to about 39 degrees Fahrenheit after enabling... I vaguely remember at the field it raising to 40, maybe 41 Fahrenheit., which is obviously incorrect.

### To Reproduce

Enable telemetry and set the temperature to display in the OSD.

### Expected behavior

The temperature should display between outside temperature (about 65 Fahrenheit for UK summer) up to typical expected ESC temperatures, such as gradually increasing to 190 when having been at high throttle for an amount of time.

### Flight controller configuration

```
PASTE THE OUTPUT OF 'diff' HERE
diff

# version
# Betaflight / STM32F745 (S745) 4.2.9 Apr 27 2021 / 19:34:04 (e097f4ab7) MSP API: 1.43
# config: manufacturer_id: HBRO, board_name: KAKUTEF7HDV, version: cc8e118f, date: 2020-03-04T02:37:20Z

# start the command batch
batch start

board_name KAKUTEF7HDV
manufacturer_id HBRO

# name: Gary Quad 6 Inch

# feature
feature -AIRMODE
feature GPS
feature ESC_SENSOR

# map
map TAER1234

# serial
serial 1 2 115200 115200 0 115200
serial 6 1024 115200 115200 0 115200

# aux
aux 0 0 2 1900 2100 0 0
aux 1 1 1 1400 1600 0 0
aux 2 2 1 1900 2100 0 0
aux 3 46 0 900 1100 0 0
aux 4 13 3 900 1100 0 0
aux 5 28 3 1900 2100 0 0

# rxrange
rxrange 0 2000 1000
rxrange 2 2000 1000

# master
set acc_calibration = 60,70,-641,1
set align_mag = CW270FLIP
set mag_align_pitch = 1800
set mag_align_yaw = 2700
set mag_hardware = AUTO
set mag_calibration = 121,-64,-72
set baro_hardware = AUTO
set serialrx_provider = SRXL2
set blackbox_p_ratio = 64
set dshot_bidir = ON
set failsafe_procedure = GPS-RESCUE
set bat_capacity = 1550
set vbat_max_cell_voltage = 425
set vbat_min_cell_voltage = 315
set vbat_scale = 11
set vbat_divider = 1
set ibata_scale = 168
set small_angle = 180
set gps_provider = UBLOX
set gps_sbas_mode = AUTO
set gps_auto_baud = ON
set gps_ublox_use_galileo = ON
set gps_rescue_sanity_checks = RESCUE_SANITY_FS_ONLY
set gps_rescue_min_dth = 50
set telemetry_disabled_esc_current = OFF
set telemetry_disabled_esc_voltage = OFF
set telemetry_disabled_esc_rpm = OFF
set telemetry_disabled_esc_temperature = OFF
set osd_units = IMPERIAL
set osd_warn_rssi = ON
set osd_warn_link_quality = ON
set osd_warn_rssi_dbm = ON
set osd_warn_over_cap = ON
set osd_vbat_pos = 497
set osd_rssi_pos = 2069
set osd_link_quality_pos = 5
set osd_rssi_dbm_pos = 18
set osd_flymode_pos = 2053
set osd_current_pos = 2537
set osd_mah_drawn_pos = 2535
set osd_gps_speed_pos = 2061
set osd_gps_lon_pos = 32
set osd_gps_lat_pos = 46
set osd_gps_sats_pos = 2071
set osd_home_dir_pos = 2059
set osd_home_dist_pos = 2056
set osd_compass_bar_pos = 15
set osd_altitude_pos = 2063
set osd_power_pos = 2532
set osd_warnings_pos = 498
set osd_avg_cell_voltage_pos = 2541
set osd_disarmed_pos = 2100
set osd_esc_tmp_pos = 2546
set osd_esc_rpm_pos = 247
set osd_stat_rtc_date_time = ON
set osd_stat_tim_1 = ON
set osd_stat_max_dist = ON
set osd_stat_endbatt = ON
set osd_stat_battery = ON
set osd_stat_max_alt = ON
set osd_stat_max_g_force = ON
set osd_stat_max_esc_temp = ON
set osd_stat_max_esc_rpm = ON
set osd_stat_min_link_quality = ON
set osd_stat_flight_dist = ON
set osd_stat_max_fft = ON
set osd_stat_total_flights = ON
set osd_stat_total_time = ON
set osd_stat_total_dist = ON
set osd_stat_min_rssi_dbm = ON
set debug_mode = GYRO_SCALED
set gyro_1_align_yaw = 2700
set name = Gary Quad 6 Inch

profile 1

# profile 1
set level_limit = 75

rateprofile 0

# end the command batch
batch end

# 
```
```
PASTE THE OUTPUT OF 'resource show all' HERE
resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: LED 1
A03: FREE
A04: SDCARD_CS
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FREE
B00: MOTOR 1
B01: MOTOR 2
B02: FREE
B03: FREE
B04: FREE
B05: FREE
B06: I2C_SCL 1
B07: I2C_SDA 1
B08: FREE
B09: FREE
B10: FREE
B11: FREE
B12: PREINIT 3
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: ADC_CURR
C03: ADC_BATT
C04: FREE
C05: FREE
C06: SERIAL_TX 6
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: FREE
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: SERIAL_TX 2
D06: SERIAL_RX 2
D07: FREE
D08: SDCARD_DETECT
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: BEEPER
E00: FREE
E01: GYRO_EXTI
E02: SPI_SCK 4
E03: FREE
E04: GYRO_CS 1
E05: SPI_MISO 4
E06: SPI_MOSI 4
E07: SERIAL_RX 7
E08: FREE
E09: MOTOR 3
E10: FREE
E11: MOTOR 4
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1: DSHOT_BITBANG 2
    CH2: DSHOT_BITBANG 5
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 2
DMA2 Stream 3: DSHOT_BITBANG 5
DMA2 Stream 4: ADC
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

# 
```

### Setup / Versions

- Flight controller: Holybro Kakute F7 HDV Tekko32 F3 Metal 4in1 ESC Combo (65A). Bought from “FPV World” in the UK.

- Other components: Beitian BN-880 GPS and Compass Module, Spektrum DSMX SRXL2 Serial Micro Receiver (SPM4650), VIFLY V2 Lost Model Finder Battery Buzzer, DJI Air Unit.

- How are the different components wired up: Air Unit plugs directly into the FC. GPS Module is connected to SDA, SCL & UART 2 pads. Buzzer to buzzer pads. SPM4650 to UART 6. 

This exact same quad when using INAV does indeed provide the correct ESC temperature. Also, I have another quad with the same setup apart from that it uses the Kakute F7 HDV 45A stack, and that also displays 39 Fahrenheit. I've also attempted to resolve this issue by posting on the following forum: https://intofpv.com/t-temperature-telemetry-incorrect-betaflight

GaryDT, 2021-07-16:
Please can someone kindly have a look into this, and confirm if it's a bug or a setup issue... I guess it's a bug, but I simply have little idea, thank you.

haslinghuis, 2021-07-17:
Checked with an Aikon ESC and Radix board (sorry don't other FC connected to an ESC with telemetry.
I does show the temperature in OSD preview and on my OSD screen. It does only show the highest temperature of the motors used.

GaryDT, 2021-08-15:
**Update:** I've re-flashed using full chip erase for both INAV and Betaflight to establish the minimum steps required for each to display ESC Temperature, as below...

**INAV (minimum steps required)**

1. Flash “3.0.1 – KAKUTEF7HDV – 2021-7-4 18:14” firmware (leave as default: Full chip erase) … click “Connect” … select “Mini Quad with 3” – 7” propellers” … wait for reboot to complete
2. Mixer tab: click “Load and apply” in mixer tab… select “Mini Quad with 3” - 7” propellers” and wait for reboot to complete
3. Outputs tab: enable “Enable motor and servo output” and also select DSHOT 600 and then click save
4. Ports tab: for UART1 disable MSP and select DJI FPV VTX under “Peripherals” instead. Also select “ESC output/telemetry” for UART 7 under “Peripherals” and then click save and reboot
5. OSD tab: enable “ESC Temperature” and click save
6. Result: ESC Temperature displays correctly in DJI Goggles for both metric and imperial

**Betaflight (minimum steps required. Note: simply ignore/close the warnings)**

1. Flash “KAKUTEF7HDV – 4.2.9 – 27-04-2021 20:40” firmware (leave as default: Full chip erase) … click “Connect” … select “Apply Custom Defaults” … wait for reboot to complete
2. Ports tab: for UART7 select “ESC” under “Peripherals” and then click save and reboot
3. Configuration tab: enable “ESC SENSOR” and “Bidirectional DShot” and then click save and reboot
4. OSD tab: enable “ESC temperature” and click save
5. Result: ESC Temperature displays incorrectly in DJI Goggles for both metric and imperial… varying from 39 to 41 degrees Fahrenheit

If anybody has this FC and some DJI goggles to hand, please confirm the above sequence of testing if possible.

Thank you, Gary.

closed 2021-09-23 by github-actions[bot]

Gereon99, 2021-09-25:
I have the same problem, exactly as you described. Did you ever find a fix / workaround? I used Betaflight 4.2.9 and in the OSD the temperature is always 4°C.

GaryDT, 2021-09-27:
Unfortunately I haven't found a fix.

I also have a Matek F722-HD flight controller, which I recently put on my 6 inch quad in question, and it has this exact same issue with Betaflight (again, ok with INAV).

It must be a Betaflight BUG... it's clearly working perfectly in INAV, and it's happening on at least two different FC boards (probably all FCs... at least with our setup).

Hopefully some kind lady or gentleman will take a look at it for us.

Thank you very much for posting (Y), Cheers, Gary.

Gereon99, 2021-10-05:
I was able to fix it :)
Apparently the problem is that Betaflight sends ESC information for each motor individually, but the DJI Air Unit and Caddx Vista expect just one combined value.

If you or someone else needs ESC temperature in their OSD you can check out [this](https://github.com/Gereon99/betaflight/tree/4.2-maintenance) fork. It fixes this problem for Betaflight 4.2.X.

GaryDT, 2021-10-06:
Awesome! :) x 2

IversonWin, 2022-05-19:
Is it okay to have the bug fix in the above fork merged back to 4.3 released?

damosvil, 2022-07-26:
Same happens to me. I finally let units to farenheith instead of celsius so I get the correct temperature in celsius but printed as farenheith