betaflight/betaflight #10840: Improvements for multicopters with mixed motors

LSchwerdt, 2021-07-16:
### Is your feature request related to a problem? Please describe.
When trying out different motors or switching from 6s to 4s one may end up with incomplete sets of motors (due to some getting damaged). To avoid throwing away 1-3 perfectly usable motors (if you can not find someone to sell them to), one can mix different motors on one quadcopter. Sadly, betaflight is not setup to handle this case very well.

Example of quadcopters with mixed motors (1408, 1507, 1606 & 1804):
[image]

### Describe the solution you'd like

- Enable output limit/scaling per motor instead of globally.
- Make the number of motor poles a setting per motor instead of globally, to enable the rpm filter and dynamic idle when mixing motors with different number of poles.

It is not best practice to mix different kinds of motors, so having this accessible only through the cli should be fine in my opinion. And it would avoid inconveniencing regular users.

### Describe alternatives you've considered

- The PID controller (specifically the I-term) deals with **small** differences of the motors without issues.
- One can use a custom mixer to reduce the output of some motors, but afaik the output is not limited. A combination of throttle/roll/pitch/yaw can then lead to higher than desired motor output, potentially smoking the motor/esc. Additionally I could not find any documentation about this use case for the custom mixer.
- One can use different propellers per motor. This could help but never truly compensate for different motors. And you would need to buy more types of propellers, potentially reducing/eliminating the money saved from 'using up' old motors.
- The new dynamic notch filter in 4.3 seems to be a decent alternative to the rpm filter, but you loose the dynamic idle feature.

mikeller, 2021-08-16:
@LSchwerdt: From tests with using different / damaged props we know that the PID controller in Betaflight is pretty good at dealing with imperfect / unequal motor / prop performance. Have you tested this with the hardware setup that you describe, and found that it is not sufficient to make it flyable?

LSchwerdt, 2021-08-16:
It is "pretty good", and the quads fly ok. But it could be better ;). When using these configurations one has to live without having the ability to use rpm filtering and dynamic idle. The difference in kV is quite small in my case, so it is is similar to having an off center cog. But I think whith the suggested improvements one could fly 4s and 6s motors together with 99% of the performance compared to using only one kind of motor.

With the currently very high component prices, I think having a feature to help users save money would also be great from a marketing standpoint.

1. I think making the number of motor poles specific to each motor would probably be not that difficult, and would enable users to have the aformentioned features. As far as I can see only the variable erpmToHz would need to be changed to be an array containing a value for each motor.
2. Having the ability to set the output scaling per motor would be nice, but it seems like more [places in the code ](https://github.com/betaflight/betaflight/pull/7482/files) would have to be touched to make it work. So the cost vs. benefit is probably worse compared to 1.

closed 2021-09-24 by github-actions[bot]