betaflight/betaflight #10843: 3inch quad lost control(rolling) when RC input big movement

lipk888, 2021-07-18:
hi team:
i have a issue for my 3INCH quad.when i have big movement RC input like right rolling, then increase throttle, then quad lost control.(please see my log file 45s).
i have check log file and found motor 2 have full power output when rolling issue happen.(please see my two screen shot)
 is any mis-setting i set?

**Describe the bug**
3inch quad lost control(rolling) when RC input big movement
**To Reproduce**
yes different PID and fliter have same issue, please see btfl_001 issue.bbl.csv 2th flight 
**Flight controller configuration**
[BTFL_cli_HMM_20210718_182226.txt](https://github.com/betaflight/betaflight/files/6836035/BTFL_cli_HMM_20210718_182226.txt)
**Setup / Versions**
Power:4S 
motor: 1106 4500KV 
pop: emax 3inch
Flight controller/ESC: JHEMCU F411AIO 30A, TARGET: JHEMCU F411
Other components and how they are wired
FPORT Receiver: Frsky XM+, DJI camera and vtx
Transmitter: Taranis X9E with OpenTX 2.2.4

[image]
[image]
[btfl_002 issue.bbl.csv](https://github.com/betaflight/betaflight/files/6835967/btfl_002.issue.bbl.csv)
[btfl_001 issue.bbl.csv](https://github.com/betaflight/betaflight/files/6836012/btfl_001.issue.bbl.csv)

daleckystepan, 2021-07-18:
Not an expert and laso cannot open your logs but it looks like desync issue. Check motor mechanically, check wirings and then you can update ESC itself and also tune its settings - mainly timing and demag compensation

lipk888, 2021-07-18:
[issue log.zip](https://github.com/betaflight/betaflight/files/6836278/issue.log.zip)
@ daleckystepan 
thanks for updating i have upload bbl file in the loop

closed 2021-09-01 by github-actions[bot]