betaflight/betaflight #10849: "FPV Angle Mix" mode doesn't take FF values for Yaw axis into the account for "mixed" yaw moves

haplm, 2021-07-22:
### Describe the bug
This bug only demonstrates when you are flying in "FPV Angle Mix" mode. Looking at the BBE, it is clear that for any "mixed" or "virtual" Yaw move, Yaw FF value is completely ignored (0% during hard yaw move caused by rolling the quad):
[image]

This is now the situation looks like in its entirety:
[image]

During normal yaw moves (yaw stick physically deflected) FF acts as expected, no ugly overshoots:
[image]

### To Reproduce
Enable "FPV angle mix" mode, set some FF on yaw, do a roll, and see what happens. 

### Expected behavior
In "FPV angle mix", the "virtual" yaw stick movement needs to trigger FF on yaw axis.

### Flight controller configuration

```
# 

# diff

# version
# Betaflight / STM32F7X2 (S7X2) 4.2.9 Apr 27 2021 / 19:34:29 (e097f4ab7) MSP API: 1.43
# config: manufacturer_id: CLRA, board_name: CLRACINGF7, version: 294f73cd, date: 2020-11-05T19:01:48Z

# start the command batch
batch start

board_name CLRACINGF7
manufacturer_id CLRA

# name: SmoothOperator

# resources
resource MOTOR 1 B08
resource MOTOR 2 B06
resource MOTOR 3 B09
resource MOTOR 4 B07

# feature
feature GPS
feature TELEMETRY
feature ESC_SENSOR

# beeper
beeper -ON_USB

# serial
serial 1 2 115200 57600 0 115200
serial 2 2048 115200 57600 0 115200
serial 3 1024 115200 57600 0 115200

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 1 1700 2100 0 0
aux 2 2 1 1300 1700 0 0
aux 3 46 3 1700 2100 0 0
aux 4 13 2 1700 2100 0 0
aux 5 26 3 1300 2100 0 0
aux 6 30 0 1300 2100 0 0

# adjrange
adjrange 0 0 4 900 2100 12 4 0 0
adjrange 3 0 6 900 2100 29 6 0 0

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665    0 5885 5905    0    0
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 IMD6     I CUSTOM  5732 5765 5828 5840 5866 5740    0    0
vtxtable powerlevels 3
vtxtable powervalues 0 1 2
vtxtable powerlabels 25 200 500

# master
set gyro_lowpass_hz = 0
set gyro_lowpass2_hz = 0
set gyro_to_use = FIRST
set dyn_notch_width_percent = 5
set dyn_notch_min_hz = 100
set dyn_notch_max_hz = 500
set dyn_lpf_gyro_min_hz = 0
set dyn_lpf_gyro_max_hz = 650
set acc_calibration = -44,21,29,1
set mag_hardware = NONE
set rc_smoothing_auto_smoothness = 20
set fpv_mix_degrees = 25
set serialrx_provider = CRSF
set blackbox_p_ratio = 64
set dshot_idle_value = 500
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set failsafe_procedure = GPS-RESCUE
set align_board_roll = 180
set bat_capacity = 1300
set ibata_scale = 121
set gps_provider = UBLOX
set gps_sbas_mode = AUTO
set gps_auto_baud = ON
set gps_ublox_use_galileo = ON
set gps_set_home_point_once = ON
set gps_rescue_angle = 35
set gps_rescue_descent_dist = 150
set gps_rescue_min_sats = 7
set pid_process_denom = 2
set osd_warn_link_quality = ON
set osd_rssi_alarm = 0
set osd_cap_alarm = 1300
set osd_alt_alarm = 150
set osd_esc_temp_alarm = 70
set osd_rssi_pos = 6241
set osd_link_quality_pos = 14509
set osd_rssi_dbm_pos = 6273
set osd_tim_2_pos = 6529
set osd_remaining_time_estimate_pos = 450
set osd_throttle_pos = 2487
set osd_vtx_channel_pos = 2539
set osd_crosshairs_pos = 14574
set osd_ah_pos = 6255
set osd_current_pos = 6592
set osd_mah_drawn_pos = 6562
set osd_gps_speed_pos = 6295
set osd_gps_lon_pos = 2121
set osd_gps_lat_pos = 2089
set osd_gps_sats_pos = 6199
set osd_home_dir_pos = 6204
set osd_home_dist_pos = 6327
set osd_flight_dist_pos = 6263
set osd_altitude_pos = 6231
set osd_pidrate_profile_pos = 161
set osd_warnings_pos = 14666
set osd_avg_cell_voltage_pos = 14637
set osd_pit_ang_pos = 2519
set osd_battery_usage_pos = 6600
set osd_disarmed_pos = 14763
set osd_esc_tmp_pos = 2113
set osd_esc_rpm_pos = 439
set osd_core_temp_pos = 2081
set osd_log_status_pos = 2529
set osd_rate_profile_name_pos = 6647
set osd_efficiency_pos = 2262
set osd_stat_max_dist = ON
set osd_stat_battery = ON
set osd_stat_min_rssi = OFF
set osd_stat_bb_no = OFF
set osd_stat_min_link_quality = ON
set osd_stat_flight_dist = ON
set osd_stat_total_dist = ON
set osd_stat_min_rssi_dbm = ON
set debug_mode = GYRO_SCALED
set vtx_band = 5
set vtx_channel = 3
set vtx_power = 3
set vtx_low_power_disarm = ON
set vtx_freq = 5732
set gyro_rpm_notch_q = 800
set name = SmoothOperator

profile 1

# profile 1
set dyn_lpf_dterm_min_hz = 0
set dterm_lowpass_type = BIQUAD
set dterm_lowpass_hz = 100
set dterm_lowpass2_hz = 0
set iterm_relax_cutoff = 10
set p_pitch = 60
set i_pitch = 118
set d_pitch = 58
set f_pitch = 275
set p_roll = 47
set i_roll = 96
set d_roll = 53
set f_roll = 260
set p_yaw = 77
set i_yaw = 109
set f_yaw = 260
set d_min_roll = 0
set d_min_pitch = 0
set idle_min_rpm = 14

rateprofile 0

# rateprofile 0
set thr_mid = 22
set thr_expo = 60
set rates_type = ACTUAL
set roll_rc_rate = 10
set pitch_rc_rate = 10
set yaw_rc_rate = 10
set roll_expo = 60
set pitch_expo = 60
set yaw_expo = 60
set yaw_srate = 60
set tpa_rate = 60
set tpa_breakpoint = 1300

# end the command batch
batch end
```
PASTE THE OUTPUT OF 'resource show all' HERE
```
resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: SERIAL_RX 4
A02: SERIAL_TX 2
A03: SERIAL_RX 2
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: PINIO 1
A15: OSD_CS
B00: LED 1
B01: FREE
B02: FREE
B03: CAMERA_CONTROL
B04: BEEPER
B05: SPI_MOSI 3
B06: MOTOR 2
B07: MOTOR 4
B08: MOTOR 1
B09: MOTOR 3
B10: SERIAL_TX 3
B11: FREE
B12: FLASH_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_CURR
C02: ADC_BATT
C03: FREE
C04: GYRO_EXTI
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: MCO 2
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SERIAL_TX 5
C13: PREINIT 2
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: SERIAL_RX 5
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2:
    CH2: CAMERA_CONTROL
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1: DSHOT_BITBANG 2
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 2
DMA2 Stream 3: FREE
DMA2 Stream 4: ADC
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

# 
```

### Setup / Versions
 
 - Flight controller: CLRACINGF7
 
 - Other components: doesn't matter
 
 - How are the different components wired up: doesn't matter

haslinghuis, 2021-07-22:
Afaik I think it's not a bug but a feature. ATM Betaflight doesn't allow FF in Angle Mode.

There is a proposed change here https://github.com/betaflight/betaflight/pull/10778

haplm, 2021-07-23:
@haslinghuis - "FPV Angle mix" has nothing to do with angle mode ;-). It just coincidentally uses "angle" in its name...

haslinghuis, 2021-07-23:
@ctzsnooze This is your department

ctzsnooze, 2021-07-26:
I've never heard of 'FPV Angle Mix' mode ha ha

ctzsnooze, 2021-07-26:
I see... you mean the `fpv_mix_degrees` setting in the CLI, which activates the `scaleSetpointToFpvCamAngle` function, which mixes roll and yaw to change their respective set points according to camera angle.

I was unsure how best to handle this when recently revising rc.c

The cause of the issue is, perhaps, that feedforward is calculated from `feedforwardApply` which uses `getRawSetpoint`, and the camera angle correction code doesn't change rawSetpoint.

The camera control adjustments to setpoint are performed on setpointRate, which is the smoothed output value.

So feedforward is calculated on the non-camera-corrected setpoint values at the moment, ie on the stock values.

I'll have to figure out how to rearrange the code so that the camera correction changes to the setpoint happen earlier.

haplm, 2021-07-26:
Thank you, @ctzsnooze !

PR #10860 (2021-08-03): fix fpv camera angle code error
Fixes #10849.

As noted in issue #10849, the changes recently made in rc.c have had the unintended result that feedforward isn't being calculated on the new camera-angle modified set points.  

This would happen when `fpv_mix_degrees` is set to a non-zero value, activating the FPV camera angle correction code, which crosses roll and yaw set points over to compensate for FPV camera angle.

This PR is a bug-fix that performs the camera angle corrections on the 'rawSetpoint' steps, which are the values used for feedforward calculations.  The result should be feedfo [... cut]
closed 2021-08-03 by mikeller