betaflight/betaflight #10850: Request: better iTerm management for winged quads and / or wing aircraft

ResinSmoker, 2021-07-23:
### Is your feature request related to a problem? Please describe.

### Describe the solution you'd like

### Describe alternatives you've considered

mituritsyn, 2021-08-12:
It is hard to investigate without log file. Do you have I-term relax enabled on RPY? After enabling, try to adjust cutoff

ResinSmoker, 2021-08-12:
@mituritsyn 

Problem with most builds using 1S (18650) is that there is really nothing available that has Black Box. Installing a 3rd party option such as OpenLagger is problematic due to the limited number of ports available for what amounts to a Whoop FC. (one for the FrySky RX and one for TX control) 

To answer your question: iTerm relax has been previously tested as you've suggested, with the cutoff set at a range of multiple values. At best all this does is change the frequency of the windup and its intensity. Granted, lower cutoff values perform better but the copter is only barely manageable in turns.

In other news: I'm trying new wings for both upper and lower mounts. Just testing to see if the dissimilar wing sizes could be creating a situation where the wings are fighting against the copters attempts to Yaw (Roll in fwd flight). If this is true then it means that the wings are self-stabilizing Yaw (Roll) as the larger wing set orients itself above the copters CG. (See previously attached image to get an idea of what I mean with regard to the wing sizes) Beyond all of this I'll be resetting BF to the PID defaults as this appeared to work the ok'ish for the initial flights.

ResinSmoker, 2021-08-12:
Today I flew with two set of wings.... Long and Stubby. (They came out really nice too)

Note: iterm_windup = 50 and iterm_relax = 10 (PRY)

Here is what the long (170 mm tip to tip, 48 mm wide) looks like...
[image]
Video link https://youtu.be/1njnwBWP6jY
Seams like the wing length (tip to tip) was too much and causing the Yaw (Roll in winged mode) to fight against the wings. Once the copter Pitches / Rolls to prepare for a turn the windup releases.

Here is the Stubby (Approx. 98mm tip to tip, 48 mm wide)...
[image]
Video link https://youtu.be/K7VVIE-FNQQ
Overall this flight was very smooth and even tried a few tight turns towards the end of the flight.

closed 2021-09-20 by github-actions[bot]