betaflight/betaflight #10858: GPS Rescue sometimes turns off the motors for a short time ---> crashed

aLDime, 2021-07-24:
### Is your feature request related to a problem? Please describe.

GPS Rescue sometimes turns off the motors for a short time causing a crash

### Describe the solution you'd like

Return to home in case of loss of RC signal

### Describe alternatives you've considered

In total for a flight of 4 RTH, If you watch the video at first, twice from the switch, everything is fine, the first RXLOSS - everything is fine, last time = crash :
https://youtu.be/n0xShL13hQY?t=327

[BTFL_BLACKBOX_LOG_Mark4_7inch_20210724_213223.zip](https://github.com/betaflight/betaflight/files/6872858/BTFL_BLACKBOX_LOG_Mark4_7inch_20210724_213223.zip)

[BTFL_cli_Mark4_7inch_20210724_125104.txt](https://github.com/betaflight/betaflight/files/6872867/BTFL_cli_Mark4_7inch_20210724_125104.txt)

[image]

aLDime, 2021-07-25:
I tried crutches like:
1. MULTISHOT
2. MOTOR_STOP - OFF
3. RC FailSafe CH3 = -10 (which is about 40% throttle)
3. BF FailSafe Throttle [T] - HOLD

it's not a feature - it's a bug!
at the moment of activating Rescue clockwise everything is OK, when turning counterclockwise the BF turns off the motors:
https://youtu.be/07EYYvoQKSY
[BTFL_cli_Mark4_7inch_20210725_131500(OSD_ED).txt](https://github.com/betaflight/betaflight/files/6873648/BTFL_cli_Mark4_7inch_20210725_131500.OSD_ED.txt)
[BTFL_BLACKBOX_LOG_Mark4_7inch_20210725_152936.zip](https://github.com/betaflight/betaflight/files/6873649/BTFL_BLACKBOX_LOG_Mark4_7inch_20210725_152936.zip)

I'm deleting this stuff from all my copters, and many people will find out about it .. but you can continue to rest .. ODIOS

mituritsyn, 2021-08-10:
Looks like motor desync. Either 2/4 motor is damaged or all motors too weak to deal with such rapid setpoint change with 7" props. You can try this: https://www.youtube.com/watch?v=jWdkl0EHkyU

mikeller, 2021-08-10:
@aLDime: Please stop posting duplicates of your support requests.
If nobody wants to help you, then maybe this could be because of the attitude you are showing, so maybe have a think about this being the right way to interact with something that other people provide to you for absolutely free.

pfeerick, 2021-08-12:
Definitely have a look at the desync video... if you turn on the Motor PID values view in Blackbox Explorer, you can see quite clearly motor 2 is being told to run up to 100%, and this is usually how you spot a desync. 

[image]

closed 2021-09-20 by github-actions[bot]