betaflight/betaflight #10862: BLHeli 32.8 regression that may impact dynamic notch filtering

jacqu, 2021-07-26:
Hi,

I just made the following bug report on the BLHeli git page. 32.8 BLHeli ESCs may yield important disturbances that could impact dynamic notch filtering.

Jacques

---
Hi,

Our lab is developing a Simulink block to access Beltaflight in hardware-in-the-loop simulations:
[image]
We modified betaflight to be able to send a custom MSP command at a high frequency. The command returns, among others, the ERPM velocity of the motors acquired with bidirectional DSHOT.
We noticed a regression between 32.7 and 32.8: in the same conditions (same Simulink, same throttle value, same setup) with the same ESC configuration (Low PWM frequ = High PWM freq = PWM freq of 32.7=48kHz), the ERPM measurement is far more noisy with 32.8 with huge spikes as shown in the following Simulink plots. All data have been acquired at 200Hz, the ERPM signal is converted to RPM with the following code in the modified betaflight firmware (the betaflight function `getDshotTelemetry` has not been altered):
`(uint16_t)lrint( (float)getDshotTelemetry(i) * 100.0 * 2.0 / motorConfig()->motorPoleCount )`
[image]
[image]
[image]

The same behavior occurs with various ESCs. Here we are using the Flywoo F745 AOI.

Thanks for looking into this since it may yield important disturbances for the dynamic notch filtering in betaflight.

Jacques

KarateBrot, 2021-07-26:
I think this is more significant for RPM filtering than for the dynamic notch.

But still, if this means the RPM filter is not able to filter as well on 32.8 and some motor noise is leaking, this also impacts the dynamic notch filter slightly because it will try to clean up residual motor noise.

jacqu, 2021-07-27:
Hello,

My bug report on the BLHeli git has picked some interest. One developer is kindly willing to dig into this. I accepted to help him make some tests on my board. All the updates are on this thread:

[Link to issue on BLHeli git](https://github.com/bitdump/BLHeli/issues/546)

Best,
Jacques

closed 2021-09-03 by github-actions[bot]